mobizt/Firebase-ESP8266 #156: RTDB Stream not getting all data

BorjanKalinoski, 2020-10-26:
Hello,

The stream callback is not always triggered.
Using the same example from your library, whenever I change/add data to the path provided to the callback, 
the callback is not always triggered, i.e the data is not forwarded to the ESP8266. 

What is the reason behind this? Is there any way to be absolutely sure that the callback will get triggered when there is a change in the path provided?

As you can see, the data that the callback function got is not up to date with the data in firebase.

[image]

mobizt, 2020-10-26:
It's the server event of http stream which depending on network status and your blocking code or other code execution time.

Should not use any delay in your loop as it's a single core mcu.

mobizt, 2020-10-26:
You should let the readStream run freely even you set the stream callback, the readStream will run in the esp schedule task which will be block with the long running user task too.

closed 2020-10-26 by mobizt

mobizt, 2020-10-26:
If the data on the database is unchanged, no server event push to the client.
When the network delay or lost, all server event will be missed even the stream was resumed by the library.

The library design to auto resume every 40 seconds that the keep-alive event missing from server,

No connected and disconnected event available on http stream unless other platform that using the Firebase SDK which run on Google own web socket technology.

If you want to keep track for all change with the data guarantee, MQTT is the good choice.

BorjanKalinoski, 2020-10-27:
Okay, I want to thank you for your effort on this library, it has helped me and many more a lot. The dedication and the fast responses is something i truly respect and admire. Instead of opening an new issue i will ask my question here.

I ditched the idea of stream callback since there is data loss. Instead i use the .getJSON() method every X seconds to read  data(not ideal, but it works in my case). But every once in a while i get an timeout error, both when using .getJSON() and .set().
Maybe this is similar to the issue you described above, but i wanted to ask what is causing this, are there any ways to avoid it, and how often will this occur?

The idea is I send a command to RTDB from an app, ESP8266 reads that command, does something and then sends a response back to RTDB and the app accordingly. I avoided MQTT since there is a need to maintain a server, but if there is no other way i will look into Firebase functions and integrate it using MQTT that way.

Thanks in advance, best regards!

BorjanKalinoski, 2020-10-27:
I notice when i use only the set() funciton, there is no timeout (happens less atleast), when i use get() & set() the timeout occur more often.It is probably random , but  I am just asking for  what is causing theese timeouts and what is the best way to avoid (if possible) them.

mobizt, 2020-10-27:
If you use the same Firebase Data object for stream and other Firebase calls (I often describe this many times in many issues), the stream need to be stop and open the new connection for other get/set/push operation.

To maintain the stream to stay connected to server, you need to use another Firebase Data object to work with stream only, don't share it to other task.

And you need to use the stable and reliable network  too.

mobizt, 2020-10-28:
If you follow [this example](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Stream_callback/Stream_callback.ino), you'll see that it never touch the firebaseData1 which works for stream and only retrieve the data from its stream event.

The stream connected in the special condition of http in stream mode that the server and client need to keep the connection open, the client need to wait for server event data (one directional data transfer from server to client). If client is not exist or the connection was closed due to some interruption of internet or the client closes the connection itself as it works in other http connections, the stream event data will be missed without any notice unless the Firebase SDK client which works under web socket.

While the connection keep open in stream mode, the server will send the payload JSON data for keep-alive acknowledgement for client.  The interval of server sending the keep-alive event data may vary from 30 seconds to 1 minutes. This library use this data to notify user that the stream was timed out if the keep-alive event can't not get in time and resume (start new) http stream connection.