mobizt/Firebase-ESP8266 #160: Enhancement Request: support email/password authentication

goneall, 2020-11-01:
I'm looking for an alternative to the database secret authentication for Firebase on the ESP8266 and EPS32.  I've tried out this library and a couple other Firebase libraries for the ESP8266 and this library has worked great - compact and no issue with my use case.

There are two motivations for the different authentication:

- The database secret is deprecated
- I would like to use the Firebase rules to restrict which data the ESP device can update

In searching through past issues in this repo, I saw some references to OAUTH2 support as a possible alternative, but issues with memory/supported crypto libraries were mentioned.

Would it be possible / desirable to use the [Firebase Auth REST API](https://firebase.google.com/docs/reference/rest/auth) to authenticate?

This could support email/password as well as custom token authentication by using a REST interface.

I am currently using the email/password authentication for a Raspberry Pi implementation which is being ported to the ESP platform.  For reference, the Python library used on the Pi is [Pyrebase](https://github.com/thisbejim/Pyrebase) and uses the [Auth function](https://github.com/thisbejim/Pyrebase#authentication) to login using username/email.

If this looks like it is a promising approach, I can do a bit more research and possibly a PR (although C++ isn't my native language).

mobizt, 2020-11-02:
Thanks for address this.

It's interesting for email and password signing in alternative option which the signed custom token need the signing function that is not available in BearSSL ESP8266 SSL library unless ESP32 can do with mbedTLS.

I will look into this and testing.

amitrao17, 2020-11-09:
I have the same issue; the implementation works great (thank you) but used a deprecated method for authentication.

Looking at the pyrebase code, there are two parts 1) login and 2) refresh.

based on this I was able to make 2 curl commands that give me a usable idToken, which i can just pass to Firebase.begin(FIREBASE_HOST, idToken) in my Arduino code.

curl -H "Content-Type: application/json; charset=UTF-8" \
       -X POST \
       -d '{"email":"INSERT_EMAIL_HERE","password":"INSERT_PASSWORD_HERE","returnSecureToken":"True"}' \
       https://www.googleapis.com/identitytoolkit/v3/relyingparty/verifyPassword?key=INSERT_KEY_HERE

and

curl -H "Content-Type: application/json; charset=UTF-8" \
       -X POST \
       -d '{"grantType": "refresh_token", "refreshToken": "INSERT_REFRESH_TOKEN_HERE"}' \
       https://securetoken.googleapis.com/v1/token?key=INSERT_KEY_HERE

so...if these two functions were incorporated, it would be possible to use email/password authentication. the refresh call would be done automatically in the background maybe?

mobizt, 2020-12-15:
Now I successfully add more authentication options to the library and currently testing the library stabilities.

The authentication features are

* [x] Sign in as user or guess with id token using Email and password for RTDB and storage grant access.

* [x]  Sign in as admin with custom token using Service Account credentials for RTDB and storage grant access.

* [x]  Sign in as admin with OAuth2 access token using Service Account credentials for all Firebase 
         products and services grant access.

* [x] Sign in with legacy token using database secret for only RTDB grant access. 

The service account credentials can be read directly from json file or hard coded.

The custom and access tokens above required RS256 signed JWT token for  token exchange.

The JWT token generation was done using the additional ported Arduino BearSSL and compiled ESSP8266 WiFiClientSecure's BearSSL because the WiFiClientSecure's compiled BearSSL version does not include all functions required for RSA signing. 

The signature which consisted in tail of JWT token, generated using the BearSSL library which leads to the software watch dog timed out in such a ESP8266 mcu because the calculation time used in overall process is relatively large i.e. approx. 10 seconds to 3 seconds which depending on the algorithm selected. Additionally the faster algorithm may required more heaps and stacks.

From this  authentication  achievement, the ESP8266 and ESP32 libraries can be merged into one library, and supports for Firebase storage, user sign up and management will be added in the further updates which the library APIs may be changes and may not compatible with the older ESP8266 or ESP32 version.

The next update for ESP8266 and ESP32 libraries will include this new authentication methods only.

EgHubs, 2020-12-16:
Can't wait for this update 😃

amitrao17, 2020-12-16:
Thank you thank you Mobizt!

goneall, 2020-12-16:
@mobizt Thanks for the update!  Looking forward to it.

mobizt, 2020-12-23:
Hi all,

The library is updated and more authentication methods added.

Please update and check [this examples](https://github.com/mobizt/Firebase-ESP8266/tree/master/examples/Authentications).

goneall, 2020-12-25:
Thanks @mobizt !  The examples looks pretty clear - I'll try it out in my weather station project in the next week or so.

closed 2020-12-27 by mobizt

goneall, 2020-12-27:
Just FYI - I've been running with the email authentication for the last 2 days and it is working well.

I ran into some errors early on in testing where it was reporting an error of not ready, but after putting in a delay the problem went away.

mobizt, 2020-12-27:
This is due to low cellulation power in esp8266 compare to esp32 to calculate the signature.