mobizt/Firebase-ESP8266 #164: HELP, Is it possible, that I set values together with timestamp in same FirebaseJson Object before pushJSON?

jatu-studiobox, 2020-11-05:
Is it possible, that I set values together with timestamp in same FirebaseJson Object before pushJSON?

Now, 
1. I use 'set' function for setting FirebaseJson values first. 
2. Then I use 'pushJSON' function for saving FirebaseJson Object into Realtime Database.
3. I gather 'Id' from 'pushName()' function for set timestamp at Inserted FirebaseJson Object node. 
3. Then I use ''setTimestamp' function by using 'Id' at previous step for saving 'timestamp'.

Has anyway for set timestamp together the other values before push FirebaseJson?

mobizt, 2020-11-05:
No, you can't set the timestamp with the data that is being sent.

The Firebase timestamp requires its  unique  JSON (key/value) payload in single http request which treated internally in the library and  accept no data  included in that payload otherwise the operation failed due to server rejected.

If you don't mind about server's timestamp for time synchronization in all clients, use the time from NTP server as timestamp is recommended.

closed 2020-11-05 by mobizt

jatu-studiobox, 2020-11-05:
Thank you very much for your help 👍 

If I use NTP Server time, the timestamp is 'unsigned log' variable. 

I must cast 'unsigned long' to 'double' for setting value to FirebaseJson Object, right?

mobizt, 2020-11-05:
Yes, the double size is 8 bytes that can handle 4 bytes of the long type data.

jatu-studiobox, 2020-11-06:
Excuse me. I have a question about 'set' double variable to FirebaseJson object.

I had converted 'unsigned long' to 'double' for value to FirebaseJson object.

My 'unsigned long' variable has value '1604669040'. After I cast '1604669040' to 'double' and set to FIrebaseJson object.

Then I pushJson to Realtime Database and check insert value. Its value is '1604669040.0000002'.

Could you please advise me. How can I remove digits '.0000002' from 'double' variable.

Below code is my code for casting 'unsigned long' variable to 'double' variable. Then I set it to FirebaseJson object.

`firebaseJsonDto.set("log_datetime", (double)currentEpochTime);`

mobizt, 2020-11-06:
The problem in **dtostrf**,  the C++ function which convert the float numner to string before sending the data.

Please use this to ignore all decimal places
```C++
Firebase.setDoubleDigits(0);
```

And you need to set it back to original or other decimal place count after used
```C++
Firebase.setDoubleDigits(9); 
```
Which 9 is the default decimal places defined in the library which can be 1,2,3 or any, depending on the precision required.

jatu-studiobox, 2020-11-08:
Thank you very much for your advise. 

I try to use 

`Firebase.setDoubleDigits(0);`

before FirebaseJson.set(...) for double variable, but the digits of inserted data had remain.

Below is my code.

[image]

No. 1 -> call Firebase.setDoubleDigits(0);
No. 2 -> set double variable to FirebaseJson
No. 3 -> Use pushJSON for saving data into Firebase Realtime database.

mobizt, 2020-11-08:
You can cast your timestamp to int instead which its max positive value is 2147483647 or can be store the timestamp (seconds) until Jan 19, 2038.

The setDoubleDigits function is for the database reading.

BorjanKalinoski, 2020-11-08:
@jatu-studiobox You can write a firebase generated timestamp this way:

[image]