mobizt/Firebase-ESP8266 #165: BUG

phoenixxd, 2020-11-05:
**Describe the bug**
streaming data from firebase causes a software crash. 
This started happening after I updated the library to the latest version.

**To Reproduce**
I tried to reproduce the issue using the example provided, and its not working
Use the example you provided for stream_callback. I can provide the firebase auth and host I'm using if required (its a hobby database)

The stack trace I got is as follows
`>>>stack>>>

ctx: cont
sp: 3ffffac0 end: 3fffffc0 offset: 01a0
3ffffc60:  00004c98 0000020d 0000020e 40203576  
3ffffc70:  3ffef2f4 ffffffff 00000000 4020227e  
3ffffc80:  00000024 3ffef2f4 00000020 00000000  
3ffffc90:  4026be15 3ffef2f4 3ffef2f4 402035e1  
3ffffca0:  3fff4bdc 3ffef2f4 3ffffce0 40227f39  
3ffffcb0:  4026be94 3fff2b2c a7ae147a 40227f24  
3ffffcc0:  3ffef2f4 0000020d 3ffffd30 3fff47bc  
3ffffcd0:  00000024 3ffef2f4 3ffffd30 402039ec  
3ffffce0:  0000020e 3fff055c 0000020d 40227f39  
3ffffcf0:  3fff4bdc 00000000 00000000 40227f24  
3ffffd00:  3ffef2f4 0000020e 0000020d 3ffeed68  
3ffffd10:  3ffef2f4 0000020d 0000020d 40206ccf  
3ffffd20:  3fff2a6c 00000000 3fff2c24 4022e164  
3ffffd30:  000000c8 000001e8 ffffffff 00000000  
3ffffd40:  00000024 3fffc600 00000000 00000000  
3ffffd50:  00000000 00000000 00000000 00000000  
3ffffd60:  00000000 01000001 3f000000 3ffefa00  
3ffffd70:  3fff2aa0 3fff29d8 3fff2a78 3fff2a30  
3ffffd80:  3fff49b8 3ffefa00 3ffefa00 3ffefa00  
3ffffd90:  3ffefa00 00000000 0000000d 0000199b  
3ffffda0:  00000000 000001bb 00000064 00000010  
3ffffdb0:  001d9b50 00000000 00000064 4021561a  
3ffffdc0:  000003be 00000001 3fff055c 00000000  
3ffffdd0:  00000000 00000000 3fff055c 4021388e  
3ffffde0:  3fff4bdc 00000000 3fff055c 0000010a  
3ffffdf0:  0000020d 00000001 0000020d 3fff45a4  
3ffffe00:  00000000 00000000 00000009 00000200  
3ffffe10:  3fff45a4 00000000 00000001 0000199b  
3ffffe20:  000019a4 00000000 3ffeef68 3ffeef78  
3ffffe30:  001d9b59 00000000 0000020e 4021e38d  
3ffffe40:  00003008 3ffeed68 3fff055c 00000000  
3ffffe50:  3ffef2f4 3ffeed68 3ffeee68 4020799b  
3ffffe60:  3ffef2f4 00000001 3ffef2f4 4020569c  
3ffffe70:  3fff29cc 3fff2684 3ffffeb0 40227f39  
3ffffe80:  3ffef268 00000001 3ffeeec8 40201f47  
3ffffe90:  3ffef268 00000001 00000000 3ffef38c  
3ffffea0:  3ffef2f4 00000000 00000000 40208b6f  
3ffffeb0:  3fff2b2c 00000601 40201f10 4021e158  
3ffffec0:  00003008 3fff2b2c 3fffff10 4020101c  
3ffffed0:  40201d50 3ffef2f4 00000020 40100acf  
3ffffee0:  40201d50 3ffef2f4 402088a4 00000000  
3ffffef0:  00002b78 0000056f 0000056f 40100844  
3fffff00:  0000063b 0000002d 00000000 00000001  
3fffff10:  3ffef434 3ffef438 0000002f 40100acf  
3fffff20:  00000001 3ffef2f4 3ffeed68 40207dd4  
3fffff30:  3fff2b2c 3ffef3f4 3fff2670 00000001  
3fffff40:  3ffef434 3ffef438 00000001 4021e177  
3fffff50:  3ffef434 3ffef438 00000001 40214ad4  
3fffff60:  007a1200 755bba80 3ffef300 402011ef  
3fffff70:  3fff2684 3fff266c 00000020 3fff2600  
3fffff80:  00000000 00000000 00000001 3ffef49c  
3fffff90:  3fffdad0 00000000 3ffef45c 402156cd  
3fffffa0:  3fffdad0 00000000 3ffef45c 40215703  
3fffffb0:  feefeffe feefeffe 3ffe8538 40100d91  
<<<stack<<<`

Decoding the Exception I found the following
`Decoding stack results
0x40203576: FirebaseESP8266::strpos(char const*, char const*, int) at C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 4809
0x4020227e: FirebaseESP8266::getPGMString(char const*) at C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 4727
0x402035e1: FirebaseESP8266::getHeader(char const*, char const*, char const*, int&, int) at C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2777
0x40227f39: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x40227f24: operator delete[](void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_opv.cc line 33
0x402039ec: FirebaseESP8266::parseRespPayload(char const*, server_response_data_t&, bool) at C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2663
0x40227f39: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x40227f24: operator delete[](void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_opv.cc line 33
0x40206ccf: FirebaseESP8266::handleResponse(FirebaseData&) at C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3192
0x4022e164: tcp_send_empty_ack at core/tcp_out.c line 2057
0x4021561a: __yield() at C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 100
0x4021388e: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 453
0x4021e38d: FirebaseESP8266HTTPClient::connected() at C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266HTTPClient.cpp line 61
0x4020799b: FirebaseESP8266::handleStreamRead(FirebaseData&) at C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3723
0x4020569c: FirebaseESP8266::set_scheduled_callback(std::function ) at C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/FirebaseESP8266.h line 2639
0x40227f39: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x40201f47: std::_Function_base::_Base_manager    (FirebaseESP8266*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\admin\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 1954
0x40208b6f: FirebaseESP8266::runStreamTask() at C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 1952
0x40201f10: std::_Function_base::_Base_manager    (FirebaseESP8266*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\admin\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x4021e158: std::_Function_handler    (FirebaseESP8266*)> >::_M_invoke(std::_Any_data const&) at c:\users\admin\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x4020101c: streamTimeoutCallback(bool) at C:\Users\ADMIN\Documents\Arduino\Stream_callback_ESP8266/Stream_callback_ESP8266.ino line 50
0x40201d50: streamCallback(StreamData) at C:\Users\ADMIN\Documents\Arduino\Stream_callback_ESP8266/Stream_callback_ESP8266.ino line 37
0x40100acf: free(void*) at C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\umm_malloc\umm_malloc.cpp line 362
0x40201d50: streamCallback(StreamData) at C:\Users\ADMIN\Documents\Arduino\Stream_callback_ESP8266/Stream_callback_ESP8266.ino line 37
0x402088a4: FirebaseESP8266::runStreamTask() at C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2335
0x40100844: umm_free_core(void*) at C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\umm_malloc\umm_malloc.cpp line 316
0x40100acf: free(void*) at C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\umm_malloc\umm_malloc.cpp line 362
0x40207dd4: FirebaseESP8266::setStreamCallback(FirebaseData&, void (*)(StreamData), void (*)(bool)) at C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 4590
0x4021e177: std::_Function_handler    (FirebaseESP8266*)> >::_M_invoke(std::_Any_data const&) at c:\users\admin\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40214ad4: run_scheduled_functions() at C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/interrupts.h line 25
0x402011ef: setup() at C:\Users\ADMIN\Documents\Arduino\Stream_callback_ESP8266/Stream_callback_ESP8266.ino line 104
0x402156cd: __loop_end() at C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 167
0x40215703: loop_wrapper() at C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 181`

**IDE and its version:**
 - Arduino via VS Code
 - Arduino Version 1.8.13

**ESP8266 Arduino Core SDK version**
 - Version 2.6.3

**Additional context**
Add any other context about the problem here.

mobizt, 2020-11-05:
I did not found bugs as you descried with your code, the latest ESP8266 Core and the latest library.

It may depending on the amount of payload from your stream path which too large for ESP8266 to handle, you should reduce and redesign your database.

No issue from the update of the library.

closed 2020-11-05 by mobizt

phoenixxd, 2020-11-06:
Hi @mobizt 
Thanks a ton for the quick reply.

The payload when it connects for the first time would 350Bytes.
For every change in state, the payload should be around 10Bytes (assuming one character consumes a byte)

And this much data I was already receiving without any issues with earlier version of the library. After the update it started crashing.

Attaching the image of the what the datalooks like. S1-S7 keys are only there
[image]

This much data should be parseable right?

mobizt, 2020-11-06:
Can you export the json under the stream path and post it here.

mobizt, 2020-11-06:
Ok it's enough info for me.

Because of large data of stream which when the stream establish or reconnected after timeout, the payload pushes from server is too large which is a whole data inside the stream path.

The library need to compare the changes in the new server push data and the old data when the stream resumed, this process changes from the old version you used and may cause the memory allocation error because it needs to reserved memory for new large data too.

Redesign your database to get the only small, instance and important stream data instead of stream your large log data that being glow bigger.

phoenixxd, 2020-11-06:
The payload for full fetch even is small I think. 350Bytes - 400Bytes shouldn't cause an issue for ESP8266. If you could confirm this is something that's an error in the library and not an issue in how I'm running your code, I could look into it for a fix and give a PR.  

When you say to redesign the database to get small important data, how small the payload should be? 

There are 7pieces of information I need in real-time so should I try with Multipath stream ?

mobizt, 2020-11-06:
I don't think that. and absolutely sure that the problem is rather in your management.

The library version 2.7.4 is 10 months ago and the 49 commits (49 versions updated) behind the latest which the ESP8266 WiFiClient core library and many features changes so much since that day of that outdated version you used.

Don't ask something like how much payload that can handle with which actually the library offers the options to limit the memory usage, one for the  reserved received payload size and one for BearSSL rx and tx buffer.

Check [this](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Basic/Basic.ino#L52-L55).

You should check the examples and changes the code to comply with the latest version library

mobizt, 2020-11-06:
Don't PR for just your wrong implement the library.

mobizt, 2020-11-06:
Don't use Multipath stream because it works internally the same as normal stream.

Redesign your database which your should separate the instant status data, config data and log or history data.

The instant status data should keep small and suit for stream as you use in MQTT concept.

The bigger data for config and history data should not contain in the stream path.

phoenixxd, 2020-11-06:
Thanks a lot @mobizt 
I will look into the examples and will try to find an apt fit. 

Really appreciate your quick replies!! ☺️
Thanks again for this amazing library.

mobizt, 2020-11-11:
The library updated with the bugs fixed.
Please try to download via GitHub.

phoenixxd, 2020-11-14:
Thanks a lot. @mobizt 
It's working now