mobizt/Firebase-ESP8266 #168: Stack error after update from 2.9.6 to 3.0.2

alexandrdiachenko, 2020-11-10:
**Describe the bug**
I receive stack error after few seconds after start.
My stream data is not big.
[siswood-menu-export.txt](https://github.com/mobizt/Firebase-ESP8266/files/5517685/siswood-menu-export.txt)
This problem is typical for the code from the Examples folder.
After 10-20 error reboots, data can be received successfully.
Changing the buffer size doesn't help.

mobizt, 2020-11-10:
์No details and code. Why you think it's a bug?

closed 2020-11-10 by mobizt

alexandrdiachenko, 2020-11-10:
Because in past version all works very well. 
I will try give you full information as i can. 
I don't know how to decode stack trace in platformio, sorry.  
I recorded my screen. Link : [https://fex.net/s/dvfzypn](url)
When I downgrade to 2.9.6 problem is gone.

Test code `

#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>
#include <./secret.h>

//Define FirebaseESP8266 data object
FirebaseData firebaseData;

unsigned long sendDataPrevMillis = 0;

String path = "/devices/11:11:11:11:11:11/menu";

uint16_t count = 0;

void printResult(FirebaseData &data);

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
  firebaseData.setBSSLBufferSize(2048, 2048);

  //Set the size of HTTP response buffers in the case where we want to work with large data.
  firebaseData.setResponseSize(2048);

  if (!Firebase.beginStream(firebaseData, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}

void loop()
{
  
  if (millis() - sendDataPrevMillis > 15000)
  {
    sendDataPrevMillis = millis();
    count++;

  }

  if (!Firebase.readStream(firebaseData))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't read stream data...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  if (firebaseData.streamTimeout())
  {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }

  if (firebaseData.streamAvailable())
  {
    Serial.println("------------------------------------");
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData.streamPath());
    Serial.println("EVENT PATH: " + firebaseData.dataPath());
    Serial.println("DATA TYPE: " + firebaseData.dataType());
    Serial.println("EVENT TYPE: " + firebaseData.eventType());
    Serial.print("VALUE: OK");
    Serial.println("------------------------------------");
    Serial.println();
  }
}
`

Stack Trace 

`--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (29):
epc1=0x402225f5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb80 end: 3fffffc0 offset: 0190
3ffffd10:  c026859c 00000000 0000020e 40202b6c  
3ffffd20:  00005648 0000020d 0000020e 402040da  
3ffffd30:  3ffeeca8 ffffffff 00000000 40202dca  
3ffffd40:  00000024 3ffeeca8 00000020 00000000  
3ffffd50:  4026859d 3ffeeca8 3ffeeca8 40204145  
3ffffd60:  3fff4d04 3ffeeca8 3ffffda0 402245f1  
3ffffd70:  4026861c 3fff239c 4bc6a7f0 402245dc  
3ffffd80:  3ffeeca8 0000020d 3ffffdf0 3fff4aec  
3ffffd90:  00000024 3ffeeca8 3ffffdf0 4020454c  
3ffffda0:  0000020e 3fff0574 0000020d 402245f1  
3ffffdb0:  3fff4d04 00000000 00000000 402245dc  
3ffffdc0:  00000000 0000020e 0000020d 3ffee700  
3ffffdd0:  3ffeeca8 0000020d 0000020d 40206fd3  
3ffffde0:  4023475d 3fff3fb1 3fff3fac 3fff2904  
3ffffdf0:  000000c8 000001e8 ffffffff 00000000  
3ffffe00:  00000024 3fffc600 00000000 00000000  
3ffffe10:  00000000 00000000 00000000 00000000  
3ffffe20:  00000000 01000001 3f000000 3ffeefb0  
3ffffe30:  3fff4ac0 3fff23f8 3fff4a60 3fff4a48  
3ffffe40:  3fff4f28 3ffeefb0 3ffeefb0 3ffeefb0  
3ffffe50:  3ffeefb0 4bc6a7f0 0000000d 00000010  
3ffffe60:  00000000 00000000 4bc6a7f0 00000000  
3ffffe70:  00000000 4bc6a7f0 00001519 00000010  
3ffffe80:  00000000 00000000 00000000 00000000  
3ffffe90:  00000000 00000001 40100256 1e76c8b4  
3ffffea0:  3fff4d04 00000000 3fff0574 0000010a  
3ffffeb0:  0000020d 00000001 0000020d 3fff27d4  
3ffffec0:  00000000 00000000 00000009 00000200  
3ffffed0:  3fff4824 00000000 00000001 00000000  
3ffffee0:  00001527 00000000 3ffee900 3ffee910  
3ffffef0:  00000000 00000000 0000020e 210624dd  
3fffff00:  00000000 3ffee700 3fff0574 00000000  
3fffff10:  3ffeeca8 3ffee700 3ffee800 40207ac9  
3fffff20:  00000000 4bc6a7f0 00001519 3ffeedec  
3fffff30:  00000000 4bc6a7f0 00001519 3ffeedec  
3fffff40:  3ffeec00 00000001 3ffee860 00000000  
3fffff50:  00000000 3fffff70 40100256 3ffeedec  
3fffff60:  3fffdad0 3ffee700 3ffee6f8 40207af8  
3fffff70:  3fffdad0 00000000 3ffeedac 4020120c  
3fffff80:  00000000 00000000 00000001 401001b4  
3fffff90:  3fffdad0 00000000 3ffeedac 401001d5  
3fffffa0:  3fffdad0 00000000 3ffeedac 4020c40c  
3fffffb0:  feefeffe feefeffe 3ffe8538 40100141  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------`

mobizt, 2020-11-11:
run **pio update** in the console of platfornIO

mobizt, 2020-11-11:
It's not because the update of library with something basic like this. You should provide the full code instead of just paste the some part of code.

Try with all examples.

Check what third party libraries you included in your sketch and what's the conflicts.

Move your code to run in Arduino IDE and decode the stack there.

alexandrdiachenko, 2020-11-11:
I tried all examples (

Fully new Arduino IDE

Full code 
[testing.ino.txt](https://github.com/mobizt/Firebase-ESP8266/files/5522877/testing.ino.txt)

Decode Stack trace
`
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x40224829
EXCVADDR: 0x00000000

Decoding stack results
0x40201a9c: FirebaseESP8266::newS(unsigned int) at /Users/pc/Documents/Arduino/libraries/Firebase_ESP8266_Client/src/FirebaseESP8266.cpp line 2315
0x4020302a: FirebaseESP8266::strpos(char const*, char const*, int) at /Users/pc/Documents/Arduino/libraries/Firebase_ESP8266_Client/src/FirebaseESP8266.cpp line 4809
0x40201d2e: FirebaseESP8266::getPGMString(char const*) at /Users/pc/Documents/Arduino/libraries/Firebase_ESP8266_Client/src/FirebaseESP8266.cpp line 4727
0x40203095: FirebaseESP8266::getHeader(char const*, char const*, char const*, int&, int) at /Users/pc/Documents/Arduino/libraries/Firebase_ESP8266_Client/src/FirebaseESP8266.cpp line 2777
0x402276cd: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x402276b8: operator delete[](void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_opv.cc line 33
0x4020349c: FirebaseESP8266::parseRespPayload(char const*, server_response_data_t&, bool) at /Users/pc/Documents/Arduino/libraries/Firebase_ESP8266_Client/src/FirebaseESP8266.cpp line 2663
0x402276cd: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x402276b8: operator delete[](void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_opv.cc line 33
0x40206227: FirebaseESP8266::handleResponse(FirebaseData&) at /Users/pc/Documents/Arduino/libraries/Firebase_ESP8266_Client/src/FirebaseESP8266.cpp line 3192
0x40100246: millis() at /Users/pc/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 188
0x4021281a: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at /Users/pc/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 453
0x40206ef8: FirebaseESP8266::handleStreamRead(FirebaseData&) at /Users/pc/Documents/Arduino/libraries/Firebase_ESP8266_Client/src/FirebaseESP8266.cpp line 3723
0x40100246: millis() at /Users/pc/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 188
0x40206f24: FirebaseESP8266::readStream(FirebaseData&) at /Users/pc/Documents/Arduino/libraries/Firebase_ESP8266_Client/src/FirebaseESP8266.cpp line 1953
0x402017ed: loop() at /Users/pc/Documents/Arduino/testing/testing.ino line 88
0x40213cb1: String::invalidate() at /Users/pc/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/WString.cpp line 140
0x4020113f: setup() at /Users/pc/Documents/Arduino/testing/testing.ino line 48
0x401001a4: ets_post(uint8, ETSSignal, ETSParam) at /Users/pc/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4021457c: loop_wrapper() at /Users/pc/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
`

mobizt, 2020-11-11:
What is about the result from trying all examples?

alexandrdiachenko, 2020-11-11:
Get and post requests working well with any size of data.
Stream functions has a problems like in yesterday video.

mobizt, 2020-11-11:
Video?

alexandrdiachenko, 2020-11-11:
I recorded my screen. Link : https://fex.net/s/dvfzypn

mobizt, 2020-11-11:
I test your code and did not found anything wrong and nothing is related to the update of library.

The reset cause 2 and stack trace printed like this is related to your hardware failure, most of them is the power at your device is not enough which caused by long and bad USB cable.

Please try to use large capacitor connected across 3v3 and ground on your esp8266 board.

mobizt, 2020-11-11:
As you will see, the device reset immediately after boot which it fails to fetch any user program from flash memory. It reset prior to the serial initialization and WiFi connection.

Another reason is your device uses some ports (GPIO 0, 2 and 15) to connect to flash memory at boot and that ports are not available (pull high or low). For this case, please google how to avoid the boot failure.

mobizt, 2020-11-11:
I delete the post about boot failure due to the WiFi is already connected but it's not enough power due to high inrush current during WiFi data transmission.

alexandrdiachenko, 2020-11-11:
I used several new ESPs for the test. My main board is powered independently by 5v 2a with a 2.2k capacitor.
I think that's not the point.
Please check this video, with two version of lib: https://fex.net/s/ol3lomk

alexandrdiachenko, 2020-11-11:
In my main code stream start after few seconds.

mobizt, 2020-11-11:
The result of your code on Arduino IDE v1.8.13 ESp8266 Core SDK 2.7.4
The library was developed in VSCode with platformIO extension, and nothing failed from long intensive test in many devices

Connected with IP: 192.168.43.26

------------------------------------
Stream Data available...
STREAM PATH: /devices/11:11:11:11:11:11/menu
EVENT PATH: /
DATA TYPE: null
EVENT TYPE: put
VALUE: null
------------------------------------

------------------------------------
Set string...
PASSED
PATH: /devices/11:11:11:11:11:11/menu/String
TYPE: string
VALUE: Hello World! 1
------------------------------------

Stream timeout, resume streaming...

------------------------------------
Stream Data available...
STREAM PATH: /devices/11:11:11:11:11:11/menu
EVENT PATH: /
DATA TYPE: json
EVENT TYPE: put
VALUE: 
Pretty printed JSON data:
{
    "String": "Hello World! 1"
}

Iterate JSON data:

0, Type: object, Key: String, Value: Hello World! 1
------------------------------------

Stream timeout, resume streaming...

------------------------------------
Stream Data available...
STREAM PATH: /devices/11:11:11:11:11:11/menu
EVENT PATH: /
DATA TYPE: json
EVENT TYPE: put
VALUE: 
Pretty printed JSON data:
{
    "String": "Hello World! 1"
}

Iterate JSON data:

0, Type: object, Key: String, Value: Hello World! 1
------------------------------------

------------------------------------
Stream Data available...
STREAM PATH: /devices/11:11:11:11:11:11/menu
EVENT PATH: /
DATA TYPE: json
EVENT TYPE: put
VALUE: 
Pretty printed JSON data:
{
    "String": "Hello World! 1"
[... 8517 lines cut ...]

0, Type: object, Key: String, Value: Hello World! 135
------------------------------------

------------------------------------
Stream Data available...
STREAM PATH: /devices/11:11:11:11:11:11/menu
EVENT PATH: /
DATA TYPE: json
EVENT TYPE: put
VALUE: 
Pretty printed JSON data:
{
    "String": "Hello World! 135"
}

Iterate JSON data:

0, Type: object, Key: String, Value: Hello World! 135
------------------------------------

Stream timeout, resume streaming...

------------------------------------
Set string...

mobizt, 2020-11-11:
You may check my post in this [similar issue](https://github.com/mobizt/Firebase-ESP8266/issues/165)

alexandrdiachenko, 2020-11-11:
When the stream exceeds 30 data, the new version crashes.
Yes, i read all post after 2.9.6 version, and decide write to you, because the library not working with a stream that is smaller than most JSON API objects. And increasing the buffer size doesn't affect anything.

mobizt, 2020-11-11:
What is this mean **exceeds 30 data**

mobizt, 2020-11-11:
Did you try to reduce the data under the stream path by follow my recommendation?

mobizt, 2020-11-11:
In the new version, no stream payload size limit, user needs to manage it to suit to the available memory. Only normal requests (get/set/push/update) payload size is limited by the function **setResponseSize**.

In older version, the stream payload and normal request payload size is limited by setting **setResponseSize**.

alexandrdiachenko, 2020-11-11:
Sorry, google translate ))
Json fields. 
If nothing has changed, then why did everything stop working on the same stream :(
Yes, I removed part of stream, and all work well, but 30 json data fields it is not good. 

The language barrier does not allow for fluent communication. If all the videos talk about the normal operation of the new version, then I just have to adjust.
Thank you for your time.

mobizt, 2020-11-11:
please read my above post again

mobizt, 2020-11-11:
> In the new version, no stream payload size limit, user needs to manage it to suit to the available memory. Only normal requests (get/set/push/update) payload size is limited by the function **setResponseSize**.
> 
> In older version, the stream payload and normal request payload size is limited by setting **setResponseSize**.
> 
> When the stream first connected, server will push all data under the stream path to the device, if this payload data is too big (in the new version) for device to handle with the dynamic memory allocation, your device may reset due to out of memory.
> 
> This is the changes in new version because I add the File and Blob supports in the stream then no payload size is limit anymore. It will be the user risk to take care of the stream data size.

mobizt, 2020-11-11:
I will fix this error and inform you soon.

mobizt, 2020-11-11:
The library updated with the bugs fixed.
Please try to download via GitHub.

alexandrdiachenko, 2020-11-13:
Perfect! Thank you!