mobizt/Firebase-ESP8266 #169: Randomly get Exception cause: 28

UltimatPronin, 2020-11-11:
Arduino IDE 1.8.13;
ESP8266 Arduino Core SDK version 2.7.4;
NodeMCU V3;

Randomly get Exception cause: 28.
Sometimes after 5-10 minutes, sometimes after 0.5-1.5 hours, sometimes after >12-22 hours.

Your great library is used in big home security sketch, I can't paste it here - it is so big.
Firebase used for monitoring (update status data -  used 1st object, not stream) & sending commands (used 2nd - stream object). Available heap all time is about 9500-11000 bytes with fragmentation about 2-4%.
Now I can't direct access to serial, but exceptions causes & stacks is saved in FS. 

Here it is:
```
Restart reason: 2
Exception cause: 28
epc1=0x402085ee epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
>>>stack>>>
3ffffd10: 3ffef7a4 3ffe8d51 3fff3fc4 402085e8 
3ffffd20: 4bc6a7f0 93b645a1 00000000 00000030 
3ffffd30: ffffffff ffffffff ffffffff 00000000 
3ffffd40: 00000000 40100b00 00000000 00000000 
3ffffd50: 00000000 00000000 00000000 00000000 
3ffffd60: 00000000 00000000 00000000 3fff05ec 
3ffffd70: 3fff05ec 3fff05ec 3fff05ec 3fff05ec 
3ffffd80: 3fff05ec 3fff05ec 3fff05ec 3fff05ec 
3ffffd90: 3fff05ec 4bc6a7f0 00c49ba5 00000000 
3ffffda0: 00000000 00000000 4bc6a7f0 00000000 
3ffffdb0: 00000000 00000001 40100d8c 00006c47 
3ffffdc0: 00000000 4bc6a7f0 3fff457c 4024ba3a 
3ffffdd0: 00000000 00000001 3fff457c 402163f3 
3ffffde0: 00000000 00000000 00000000 3fff4024 
3ffffdf0: 00006c47 4bc6a7f0 05604189 00000000 
3ffffe00: 00000000 00000000 4bc6a7f0 00000000 
3ffffe10: 00000000 00006c46 40100d8c 00006c47 
3ffffe20: 00000000 3fff3ec4 3fff457c 00000000 
3ffffe30: 007d321e 00000000 ffffffff 40216595 
3ffffe40: 3fffdad0 3fff3ec4 3fff457c 00000000 
3ffffe50: 3ffef7a4 3fff3ec4 3fff3fc4 402097e0 
3ffffe60: 3ffef7a4 00000001 3ffef7a4 40207420 
3ffffe70: 3fff8f24 3fff8a38 3ffffeb0 4023b52d 
3ffffe80: 3fff43c4 00000001 3fff4024 40204347 
3ffffe90: 3fff43c4 00000001 00000000 3ffef834 
3ffffea0: 3ffef7a4 00000000 00000000 4020a2ef 
3ffffeb0: 3fff8f14 00000050 40204310 402304a8 
3ffffec0: 00000000 00000000 00000000 fffffffe 
3ffffed0: ffffffff 3fffc6fc 00000001 3ffe8560 
3ffffee0: 00000000 3fffdad0 4020a024 00000000 
3ffffef0: 000087f8 000010ff 000010ff 40101493 
3fffff00: 00000000 3fffdad0 3fff0024 00000001 
3fffff10: 3ffeff08 3ffeff0c 0000002f 40101717 
3fffff20: 00000001 00000050 80ffff60 00000000 
3fffff30: 00000000 3ffef3d0 3fff4d98 00000001 
3fffff40: 3ffeff08 3ffeff0c 00000001 402304c7 
3fffff50: 3ffeff08 3ffeff0c 00000001 40225634 
3fffff60: 007a1200 827d69c0 00000000 00000000 
3fffff70: 3fff4d7c 3fff4d94 00000020 00000000 
3fffff80: 00000000 00000000 00000001 3fff0024 
3fffff90: 3fffdad0 00000000 3ffeffe4 40227581 
3fffffa0: feefeffe feefeffe 3ffeffe4 402275b7 
3fffffb0: feefeffe feefeffe 3ffe8560 401019d1 
<<<stack<<<
```

Decoding stack results:
```
0x402085e8: FirebaseESP8266::handleResponse(FirebaseData&) at D:\Profile\Ultimat\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2976
0x40100b00: ets_intr_unlock() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 163
0x40100d8c: millis() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 188
0x4024ba3a: br_ssl_engine_flush at src/ssl/ssl_engine.c line 1304
0x402163f3: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 554
0x40100d8c: millis() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40216595: BearSSL::WiFiClientSecure::available() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 386
0x402097e0: FirebaseESP8266::handleStreamRead(FirebaseData&) at D:\Profile\Ultimat\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3723
0x40207420: FirebaseESP8266::set_scheduled_callback(std::function ) at D:\Profile\Ultimat\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/FirebaseESP8266.h line 2639
0x4023b52d: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x40204347: std::_Function_base::_Base_manager    (FirebaseESP8266*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\ultimat\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 1954
0x4020a2ef: FirebaseESP8266::runStreamTask() at D:\Profile\Ultimat\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 1952
0x40204310: std::_Function_base::_Base_manager    (FirebaseESP8266*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\ultimat\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x402304a8: std::_Function_handler    (FirebaseESP8266*)> >::_M_invoke(std::_Any_data const&) at c:\users\ultimat\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x4020a024: FirebaseESP8266::runStreamTask() at D:\Profile\Ultimat\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2335
0x40101493: umm_free_core(void*) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 351
0x40101717: free(void*) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 398
0x402304c7: std::_Function_handler    (FirebaseESP8266*)> >::_M_invoke(std::_Any_data const&) at c:\users\ultimat\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40225634: run_scheduled_functions() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/interrupts.h line 25
0x40227581: __loop_end() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 184
0x402275b7: loop_wrapper() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 198

```

Firebase data updates about 1 time per 5-30 seconds.
Please, take a look - is problem in library or in used board.

mobizt, 2020-11-11:
If you use stream, be careful for the latest version that the memory allocation for the stream payload is unlimited and when the stream data push to the device is too big to device to handle, it may cause out of memory and reset.

For another reason of out of memory, you need to check the free heap for very firebase calls.

If your data at the stream path is too large, reduce it or redesign your database.

The data under the stream path should keep small and should not contains the large log or history data.

closed 2020-11-11 by mobizt

UltimatPronin, 2020-11-11:
Thank you, I will check.
For now, stream is listening the empty patch (not exists for test), but in normal - only very small json is used for stream.

UltimatPronin, 2020-11-11:
After 16 Hours 51 Minutes 17 Seconds I get Exception cause: 28 again...
Decoding stack results:
```
0x40209719: FirebaseESP8266::handleStreamRead(FirebaseData&) at D:\Profile\Ultimat\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 3685
0x40207420: FirebaseESP8266::set_scheduled_callback(std::function ) at D:\Profile\Ultimat\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/FirebaseESP8266.h line 2639
0x4023b52d: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x40204347: std::_Function_base::_Base_manager    (FirebaseESP8266*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\ultimat\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 1954
0x4020a2ef: FirebaseESP8266::runStreamTask() at D:\Profile\Ultimat\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 1952
0x40204310: std::_Function_base::_Base_manager    (FirebaseESP8266*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\ultimat\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x402304a8: std::_Function_handler    (FirebaseESP8266*)> >::_M_invoke(std::_Any_data const&) at c:\users\ultimat\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x4020a024: FirebaseESP8266::runStreamTask() at D:\Profile\Ultimat\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2335
0x40101493: umm_free_core(void*) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 351
0x40101717: free(void*) at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 398
0x402304c7: std::_Function_handler    (FirebaseESP8266*)> >::_M_invoke(std::_Any_data const&) at c:\users\ultimat\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40225634: run_scheduled_functions() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/interrupts.h line 25
0x40227581: __loop_end() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 184
0x402275b7: loop_wrapper() at C:\Users\Ultimat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 198
```

Can you advice, how to avoid it. For now I don't have stream data (for test), but stream is configured for listening patch. Normally I need listen the patch for small json data with command & status for all time.

mobizt, 2020-11-11:
Ok I found the bugs in the string utility functions in the library and trying to fix it and will inform you soon.

UltimatPronin, 2020-11-11:
Thank you very much!

mobizt, 2020-11-11:
The library updated with the bugs fixed.
Please try to download via GitHub.

UltimatPronin, 2020-11-11:
Thank you for very fast reply! I will try updates now.