mobizt/Firebase-ESP8266 #172: HELP Resolve no member clear error

Abbad-Zafar, 2020-12-03:
\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseJson.cpp:3314:26: error: 'class String' has no member named 'clear'

         data.stringValue.clear();

                          ^
i am getting multiple errors on this . Can anybody help ? using example provided https://medium.com/@vibrologic/serverless-iots-with-firebase-realtime-database-and-esp8266-9937d98d1ae0

mobizt, 2020-12-03:
You should update the Firebase library to the latest. 

If you ever install the FirebaseJson library (check from the libraries folder), remove it if not used or update to the latest version.

closed 2020-12-03 by mobizt

Abbad-Zafar, 2020-12-03:
Firebasejson version is 2.3.9 and FirebaseESP8266 client is 3.0.3 it updated to latest

mobizt, 2020-12-03:
Remove FirebaseJson if you don't use it.

Abbad-Zafar, 2020-12-03:
remove it still issue not resolved

mobizt, 2020-12-03:
No issue unless you installed the third party libs which the outdated FirebaseJson library was use internally.

mobizt, 2020-12-03:
Search the FirebaseJson.cpp or FirebaseJson.h in the libraries folder or past the full error here.

Abbad-Zafar, 2020-12-03:
Get this error when i remove FirebaseJson folder.

Arduino: 1.8.11 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Warning: Board Breadboard Arduino:avr:atmega328bb doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA328BB
In file included from C:\Users\Dr. Muhammad Irfan\Downloads\esp8266-sim800l-master\esp8266-sim800l-master\gprs\gprs.ino:2:0:

C:\Users\Dr. Muhammad Irfan\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/FirebaseESP8266.h:56:26: fatal error: FirebaseJson.h: No such file or directory

 #include "FirebaseJson.h"

                          ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mobizt, 2020-12-03:
> Remove FirebaseJson if you don't use it.
> 
> I mean if you installed it, remove it, don't delete the FirebaseJson files in the Firebase library folder.

Abbad-Zafar, 2020-12-03:
What do you mean by remove it and dont delete the folder ?

Abbad-Zafar, 2020-12-03:
How to remove it without deleting FirebaseJson folder?

mobizt, 2020-12-03:
Don't conclude something, let me explain. 

I mean if you ever installed FirebaseJson library, remove it.
If you never install it, search it.
If you found FirebaseJson.h and FirebaseJson.cpp in other places except for the file in the Firebase  library folder.

To make sure  temporary remove all libraries from libraries folder and install new Firebase and third party libs you used.

mobizt, 2020-12-03:
The full error before you delete FirebaseJson give you the reason of error if you can interpret it and you need to learn to understand it.