mobizt/Firebase-ESP8266 #173: streamAvailable not read changes

cezar-pereira, 2020-12-04:
Hello Dude, I have a problem with streamAvailable (), because sometimes it is recognized and sometimes not a change in firebase.

I'm leaving a video on YT to better express my problem.

https://youtu.be/tx65G-tifaA

LOOP:
if (firebaseData.streamAvailable())
    {
        Serial.println("STREAM PATH: " + firebaseData.streamPath());
        Serial.println("EVENT PATH: " + firebaseData.dataPath());
        Serial.println("DATA TYPE: " + firebaseData.dataType());
        Serial.println("EVENT TYPE: " + firebaseData.eventType());

        String eventPath = firebaseData.dataPath();
        Serial.println("EVENT PATH: " + firebaseData.dataPath());

        if (eventPath == "/CheckTemperature")
            disableCheckTest();
}

mobizt, 2020-12-05:
The stream data reading  in loop may be interrupted by your other codes e.g. delay and some hidden delay used in third party libs or the problem with internet access that's why you'll see the stream time out message.

mobizt, 2020-12-05:
After the stream was resumed, you'll see the stream data with event path "/" which return from the new stream connection.

closed 2020-12-05 by mobizt

cezar-pereira, 2020-12-06:
> Depois que o fluxo foi retomado, você verá os dados do fluxo com o caminho do evento "/" que retornam da nova conexão de fluxo.

Yes.
Problem is when he does not recognize the new change, it is not a problem with internet and delay (), because I did a test using only his lib and wifi.

new video
https://youtu.be/-M_GlnHc-JM

cezar-pereira, 2020-12-06:
part my cod teste
`
class Credentials
{
public:
    //Wifi
    const String WIFI_SSID = "....";
    const String WIFI_PASSWORD = "....";

    //Firebase
    const String FIREBASE_HOST = "....";
    const String FIREBASE_AUTH = "....";
};

class Paths
{
public:
    const String PATH = "/sensor01/";
    const String CHECK_TEMPERATURE = PATH + "/CheckTemperature/value";
};

#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>

Paths paths;
Credentials credentials;
FirebaseData firebaseData;

void setup() {
  Serial.begin(115200);

    wifiConnect();

    Firebase.begin(credentials.FIREBASE_HOST, credentials.FIREBASE_AUTH);

    if (!Firebase.beginStream(firebaseData, paths.PATH))
    {
        Serial.println("------------------------------------");
        Serial.println("Can't begin stream connection...");
        Serial.println("REASON: " + firebaseData.errorReason());
        Serial.println("------------------------------------");
        Serial.println();
    }

    Firebase.setBool(firebaseData, paths.CHECK_TEMPERATURE, false);

}

void loop()
{
    if (!WiFi.isConnected())
    {
        Serial.println("Desconectado, reconectando...");
        wifiConnect();
    }

    if (!Firebase.readStream(firebaseData))
    {
        Serial.println("------------------------------------");
        Serial.println("Can't read stream data...");
        Serial.println("REASON: " + firebaseData.errorReason());
        Serial.println("------------------------------------");
        Serial.println();
        return;
    }

    if (firebaseData.streamTimeout())
    {
        Serial.println("Stream timeout, resume streaming...");
        Serial.println();
    }

    if (firebaseData.streamAvailable())
    {
        Serial.println("STREAM PATH: " + firebaseData.streamPath());
        Serial.println("EVENT PATH: " + firebaseData.dataPath());
        Serial.println("DATA TYPE: " + firebaseData.dataType());
        Serial.println("EVENT TYPE: " + firebaseData.eventType());

        String eventPath = firebaseData.dataPath();
        Serial.println("EVENT PATH: " + firebaseData.dataPath());

        if (eventPath == "/CheckTemperature")
            disableCheckTest();
    }
}
void wifiConnect()
{
    WiFi.begin(credentials.WIFI_SSID, credentials.WIFI_PASSWORD);
    Serial.print("\nconectando");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(500);
    }
    Serial.print("\nconectado: ");
    Serial.println(WiFi.localIP());
}

void disableCheckTest()
{
    Firebase.setBool(firebaseData, paths.CHECK_TEMPERATURE, false);

    Serial.println("CHECKPATH" + paths.CHECK_TEMPERATURE);
}`

mobizt, 2020-12-06:
Remove this code from the loop, it can't help you to reconnect as just call WiFi begin again. 

```C++
   if (!WiFi.isConnected()){
        Serial.println("Desconectado, reconectando...");
        wifiConnect();
    }
```

You need to disconnect and let the WiFi library auto reconnect for you.  Which this reconnect feature already done by the library since you use **Firebase.reconnectWiFi(true)** and above lines of code is not necessary.

I see the code that you use one Firebase Data object for stream and normal Firebase calls e.g. get/set/push/update.

This kind of usage will interrupt the stream when you call setBool. The stream connection will be close and open the new connection to set the boolean and then resume the stream again.

If you want to stream operate continuously, you need to use other Firebase Data object to do other works.

Please follow the stream examples.

cezar-pereira, 2020-12-06:
> Firebase.reconnectWiFi(true)

THankss, 

seems to have improved, but I still suffer from double clicks or even single clicks on the APP.

Now I use a new instance: 

`FirebaseData firebaseDataSet;`

```
void disableCheckTest (){
     Firebase.setBool (firebaseDataSet, paths.CHECK_TEMPERATURE, false);
     Serial.println ("CHECKPATH" + paths.CHECK_TEMPERATURE);
}
```

video:
https://youtu.be/ARWuQH5f09w

mobizt, 2020-12-06:
Your algorithm is the problem.

You make a recursive operation like a echo which you should avoid when you use the result of the change to trigger it change.

Please redesign your algorithm and the database.

mobizt, 2020-12-06:
I lock the conversation because its too much and not a library issue.
You can reply the Email if you want me for further help or open the new issue for bugs, questions and improvement.