mobizt/Firebase-ESP8266 #174: HELP with query

NiKola-it, 2020-12-04:
Hi, great project but i have big problem on understanding how query works ( probably also my little knowledge on firebase is not helping ... )

i have a firebase like

`"Data" : {
    "0" : {
      "categoria" : "categoria 4",
      "codice" : "111",
      "dataodierna" : "gio, nov 26, 2020",
      "date1" : "N/A",
      "date2" : "N/A",
      "email" : "iosdfs@nfdsla.it",
    },
    "1" : {
      "categoria" : "categoria 4",
      "codice" : "222",
      "dataodierna" : "gio, nov 26, 2020",
      "date1" : "N/A",
      "date2" : "N/A",
      "email" : "ifdso@nikfdsa.it",
    },
    "2" : {
      "categoria" : "categoria 3",
      "codice" : "333",
      "dataodierna" : "gio, nov 26, 2020",
      "date1" : "N/A",
      "date2" : "N/A",
      "email" : "modfgb@wgdfa.it",

    }`

i have to find codice = 222 and then read categoria and get the entry number ( in this case "2" ) to see if is ok with the rest of the code.
readme and code example is not very clear to me, tested many solutions but nothing help. Can you? 
thank you
   NiKo - Italy

mobizt, 2020-12-05:
Let test with the following example code. You will get the idea.

Please read the Firebase document which related to REST API.

The library is only the tool which provides the function that followed the API provided by Google.

When you want to use some features that you don't know how it works, you need to know the concept and usage first instead of selecting the tool or function to do your work.

You may need to work with Javascript or Android clients to better understanding of Firebase.

```C++
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

#define FIREBASE_HOST "esp32-firebase-project.firebaseio.com"
#define FIREBASE_AUTH "kHsmYBYY2Dv6vZKNWQbJDwKyQOeiNwZHvZwhM2LR"

#define WIFI_SSID "."
#define WIFI_PASSWORD "39013145"

FirebaseData firebaseData;

//prototype function to set the index to some node in the database rules to ready for query
void setIndex(const String path, const String value);

//The node (paret) that we want to query data from its children
String queryNode = "/example/Data";
//The node that we want the value from it
String indexNode = "codice";

void setIndex();

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
  firebaseData.setBSSLBufferSize(2048, 1024);

  //Set the size of HTTP response buffers in the case where we want to work with large data.
  firebaseData.setResponseSize(2048);

  //Set the index (.indexOn) to the node "codice" which is under the parent node "/example/Data" in database rule
  //whether it is the top level chid or not.
  //Check your database rules to see the changes
  setIndex(queryNode, indexNode);

  FirebaseJson json;

  json.set("/0/categoria", "categoria 1");
  json.set("/0/codice", "111");
  json.set("/0/dataodierna", "gio, nov 26, 2020");
  json.set("/0/date1", "N/A");
  json.set("/0/date2", "N/A");
  json.set("/0/email", "iosdfs@nfdsla.it");

  json.set("/1/categoria", "categoria 2");
  json.set("/1/codice", "333");
  json.set("/1/dataodierna", "gio, nov 26, 2020");
  json.set("/1/date1", "N/A");
  json.set("/1/date2", "N/A");
  json.set("/1/email", "iosdfs@nfdsla.it");

  json.set("/2/categoria", "categoria 3");
  json.set("/2/codice", "333");
  json.set("/2/dataodierna", "gio, nov 26, 2020");
  json.set("/2/date1", "N/A");
  json.set("/2/date2", "N/A");
  json.set("/2/email", "iosdfs@nfdsla.it");

  json.set("/3/categoria", "categoria 4");
  json.set("/3/codice", "444");
  json.set("/3/dataodierna", "gio, nov 26, 2020");
  json.set("/3/date1", "N/A");
  json.set("/3/date2", "N/A");
  json.set("/3/email", "iosdfs@nfdsla.it");

  json.set("/4/categoria", "categoria 5");
  json.set("/4/codice", "333");
  json.set("/4/dataodierna", "gio, nov 26, 2020");
  json.set("/4/date1", "N/A");
  json.set("/4/date2", "N/A");
  json.set("/4/email", "iosdfs@nfdsla.it");

  json.set("/5/categoria", "categoria 6");
  json.set("/5/codice", "666");
  json.set("/5/dataodierna", "gio, nov 26, 2020");
  json.set("/5/date1", "N/A");
  json.set("/5/date2", "N/A");
  json.set("/5/email", "iosdfs@nfdsla.it");

  Firebase.set(firebaseData, queryNode, json);

  QueryFilter query;

  //get the query result only one (1) last node of codice which its value equals "333"
  //if you don't use limitToLast or limitToFirst, you wil get more nodes in the result

  //query.orderBy("codice").equalTo("333");
  //query.orderBy("codice").equalTo("333").limitToFirst(1);
  query.orderBy("codice").equalTo("333").limitToLast(1);
 

  if (Firebase.getJSON(firebaseData, queryNode, query))
  {
    if (firebaseData.dataType() == "json")
    {
      //parse the query result
      json = firebaseData.jsonObject();
      //Print all object data
      Serial.println("Pretty printed JSON data:");
      String jsonStr;
      json.toString(jsonStr, true);
      Serial.println(jsonStr);
      Serial.println();
      Serial.println("Iterate JSON data:");
      Serial.println();
      size_t len = json.iteratorBegin();
      String key, value = "";
      int type = 0;
      for (size_t i = 0; i < len; i++)
      {
        json.iteratorGet(i, type, key, value);
        Serial.print(i);
        Serial.print(", ");
        Serial.print("Type: ");
        Serial.print(type == FirebaseJson::JSON_OBJECT ? "object" : "array");
        if (type == FirebaseJson::JSON_OBJECT)
        {
          Serial.print(", Key: ");
          Serial.print(key);
        }
        Serial.print(", Value: ");
        Serial.println(value);
      }
      json.iteratorEnd();
    }
  }
}

//function to set the index to some node in the database rules to ready for query
void setIndex(const String path, const String value)
{
  if (Firebase.getRules(firebaseData))
  {
    FirebaseJsonData jdata;
    FirebaseJson &json = firebaseData.jsonObject();
    bool ruleExisted = false;
    String rulePath = "rules" + path + "/.indexOn";

    if (json.get(jdata, rulePath))
    {
      if (jdata.stringValue == value)
        ruleExisted = true;
    }

    if (!ruleExisted)
    {
      json.set(rulePath, value);
      String rules;
      json.toString(rules, true);

      if (!Firebase.setRules(firebaseData, rules))
      {
        Serial.println(firebaseData.errorReason());
      }
    }

    json.clear();
  }
}

void loop()
{
}

```

closed 2020-12-05 by mobizt

mobizt, 2020-12-05:
Let test with the following example code. You will get the idea.

Please read the Firebase document which related to REST API.

The library is only the tool which provides the functions that followed the API provided by Google.

When you want to use some features that you don't know how it works, you need to know the concept and usage first instead of selecting the tool or function to do your work.

You may need to work with Javascript or Android clients to better understanding of Firebase.

```C++
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

#define FIREBASE_HOST "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

#define WIFI_SSID "xxxxxxxxxxxxxx"
#define WIFI_PASSWORD "xxxxxxxxxxxxxxxxx"

FirebaseData firebaseData;

//prototype function to set the index to some node in the database rules to ready for query
void setIndex(const String path, const String value);

//The node (paret) that we want to query data from its children
String queryNode = "/example/Data";
//The node that we want the value from it
String indexNode = "codice";

void setIndex();

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
  firebaseData.setBSSLBufferSize(2048, 1024);

  //Set the size of HTTP response buffers in the case where we want to work with large data.
  firebaseData.setResponseSize(2048);

  //Set the index (.indexOn) to the node "codice" which is under the parent node "/example/Data" in database rule
  //whether it is the top level chid or not.
  //Check your database rules to see the changes
  setIndex(queryNode, indexNode);
  
  //below is for create the demo data
  FirebaseJson json;

  json.set("/0/categoria", "categoria 1");
  json.set("/0/codice", "111");
  json.set("/0/dataodierna", "gio, nov 26, 2020");
  json.set("/0/date1", "N/A");
  json.set("/0/date2", "N/A");
  json.set("/0/email", "iosdfs@nfdsla.it");

  json.set("/1/categoria", "categoria 2");
  json.set("/1/codice", "333");
  json.set("/1/dataodierna", "gio, nov 26, 2020");
  json.set("/1/date1", "N/A");
  json.set("/1/date2", "N/A");
  json.set("/1/email", "iosdfs@nfdsla.it");

  json.set("/2/categoria", "categoria 3");
  json.set("/2/codice", "333");
  json.set("/2/dataodierna", "gio, nov 26, 2020");
  json.set("/2/date1", "N/A");
  json.set("/2/date2", "N/A");
  json.set("/2/email", "iosdfs@nfdsla.it");

  json.set("/3/categoria", "categoria 4");
  json.set("/3/codice", "444");
  json.set("/3/dataodierna", "gio, nov 26, 2020");
  json.set("/3/date1", "N/A");
  json.set("/3/date2", "N/A");
  json.set("/3/email", "iosdfs@nfdsla.it");

  json.set("/4/categoria", "categoria 5");
  json.set("/4/codice", "333");
  json.set("/4/dataodierna", "gio, nov 26, 2020");
  json.set("/4/date1", "N/A");
  json.set("/4/date2", "N/A");
  json.set("/4/email", "iosdfs@nfdsla.it");

  json.set("/5/categoria", "categoria 6");
  json.set("/5/codice", "666");
  json.set("/5/dataodierna", "gio, nov 26, 2020");
  json.set("/5/date1", "N/A");
  json.set("/5/date2", "N/A");
  json.set("/5/email", "iosdfs@nfdsla.it");

  Firebase.set(firebaseData, queryNode, json);

  QueryFilter query;

  //get the query result only one (1) last node of codice which its value equals "333"
  //if you don't use limitToLast or limitToFirst, you wil get more nodes in the result

  //query.orderBy("codice").equalTo("333");
  //query.orderBy("codice").equalTo("333").limitToFirst(1);
  query.orderBy("codice").equalTo("333").limitToLast(1);
 

  if (Firebase.getJSON(firebaseData, queryNode, query))
  {
    if (firebaseData.dataType() == "json")
    {
      //parse the query result
      json = firebaseData.jsonObject();
      //Print all object data
      Serial.println("Pretty printed JSON data:");
      String jsonStr;
      json.toString(jsonStr, true);
      Serial.println(jsonStr);
      Serial.println();
      Serial.println("Iterate JSON data:");
      Serial.println();
      size_t len = json.iteratorBegin();
      String key, value = "";
      int type = 0;
      for (size_t i = 0; i < len; i++)
      {
        json.iteratorGet(i, type, key, value);
        Serial.print(i);
        Serial.print(", ");
        Serial.print("Type: ");
        Serial.print(type == FirebaseJson::JSON_OBJECT ? "object" : "array");
        if (type == FirebaseJson::JSON_OBJECT)
        {
          Serial.print(", Key: ");
          Serial.print(key);
        }
        Serial.print(", Value: ");
        Serial.println(value);
      }
      json.iteratorEnd();
    }
  }
}

//function to set the index to some node in the database rules to ready for query
void setIndex(const String path, const String value)
{
  if (Firebase.getRules(firebaseData))
  {
    FirebaseJsonData jdata;
    FirebaseJson &json = firebaseData.jsonObject();
    bool ruleExisted = false;
    String rulePath = "rules" + path + "/.indexOn";

    if (json.get(jdata, rulePath))
    {
      if (jdata.stringValue == value)
        ruleExisted = true;
    }

    if (!ruleExisted)
    {
      json.set(rulePath, value);
      String rules;
      json.toString(rules, true);

      if (!Firebase.setRules(firebaseData, rules))
      {
        Serial.println(firebaseData.errorReason());
      }
    }

    json.clear();
  }
}

void loop()
{
}

```

NiKola-it, 2020-12-05:
thank you for your help! i'm still studing firebase, but not much time to work on it so i go a little bit with...let say improvising :) 
equalTo accept string or int? can i pass a variable to it? i need to put the query in a function so i can reuse it

mobizt, 2020-12-05:
The library allows to string and integer as param for equalTo.

No matter you use string or number, the query result depends on the type of data at the node you assign in orderBy.

The query parameters will be sent as url parameters to the server then avoid the space in the function parameter value and if you use the space in the value you should encode it prior to use (search how to url encode in google).

Ex. **equalTo("sensor 1")** should be **equalTo("sensor+1")**

NiKola-it, 2020-12-05:
ok, because i had some crash of the esp caused probably from memory problem on using string in the search function i created with your code. But no space used. I will double check it.
thanks!

mobizt, 2020-12-05:
Crash? It's not because you pass the wrong data type to function, because it can't compile.
The possible device reset error are the error in memory usage and out of memory.

NiKola-it, 2020-12-05:
decoder said
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store

i just put your query in a function and called it 2-3 times.

NiKola-it, 2020-12-06:
ok, i give up and ask again for help. If i just take your code and put in a function, let's say
void search(String Value)
and call it, it get executed and after it esp crash and reboot. what i'm doing wrong?

mobizt, 2020-12-07:
You should learn to debug first as it your program design.
You should be careful about the memory allocation in the embedded device.

Nobody can help unless you provide the code.

NiKola-it, 2020-12-07:
i'm not new to using arduino, even if my coding style is not very correct it normally works.
i have done many test in the last two days and i can't do two query to firebase without crashing the esp8266. 
have you ever tested a function called in main loop() more times with query?
for ex, a simple mod of your code from your first post:

`#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

#define FIREBASE_HOST "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

#define WIFI_SSID "xxxxxxxxxxxxxx"
#define WIFI_PASSWORD "xxxxxxxxxxxxxxxxx"
FirebaseData firebaseData;

//prototype function to set the index to some node in the database rules to ready for query
void setIndex(const String path, const String value);

void SearchQuery();

//The node (paret) that we want to query data from its children
String queryNode = "/example/Data";
//The node that we want the value from it
String indexNode = "codice";

void setIndex();

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
  firebaseData.setBSSLBufferSize(2048, 1024);

  //Set the size of HTTP response buffers in the case where we want to work with large data.
  firebaseData.setResponseSize(2048);

  //Set the index (.indexOn) to the node "codice" which is under the parent node "/example/Data" in database rule
  //whether it is the top level chid or not.
  //Check your database rules to see the changes
  setIndex(queryNode, indexNode);
  
  //below is for create the demo data
  FirebaseJson jsona;

  jsona.set("/0/categoria", "categoria 1");
  jsona.set("/0/codice", "111");
  jsona.set("/0/dataodierna", "gio, nov 26, 2020");
  jsona.set("/0/date1", "N/A");
  jsona.set("/0/date2", "N/A");
  jsona.set("/0/email", "iosdfs@nfdsla.it");

  jsona.set("/1/categoria", "categoria 2");
  jsona.set("/1/codice", "333");
[... 204 lines cut ...]
3fffff10:  feef0100 00000000 000000e5 000000e5  
3fffff20:  000000eb 0000000f 00000003 00000000  
3fffff30:  3ffef50c 65646e00 00000000 00000000  
3fffff40:  00000000 00000000 3ffef50c 3ffef50c  
3fffff50:  3ffef50c 3ffef50c 3ffef50c 3ffef50c  
3fffff60:  3fff6200 0010008f 800007d0 3fff2b00  
3fffff70:  0005000f 80fe0061 3fff6500 00d700df  
3fffff80:  80213741 00000001 3ffeef9c 40213736  
3fffff90:  3fffdad0 00000000 3ffeef5c 40201eb6  
3fffffa0:  3fffdad0 00000000 3ffeef5c 40213254  
3fffffb0:  feefeffe feefeffe 3ffe8538 40100c4d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld`

and it goes on and on. Any idea? thank you!

mobizt, 2020-12-08:
Sorry for error in the usage of the FirbaseJson object  I edit and please try the following code.

```C++
void SearchQuery()
{
  
  QueryFilter query;

  //get the query result only one (1) last node of codice which its value equals "333"
  //if you don't use limitToLast or limitToFirst, you wil get more nodes in the result

  //query.orderBy("codice").equalTo("333");
  //query.orderBy("codice").equalTo("333").limitToFirst(1);
  query.orderBy("codice").equalTo("333").limitToLast(1);

  if (Firebase.getJSON(firebaseData, queryNode, query))
  {
    if (firebaseData.dataType() == "json")
    {
      FirebaseJson &json = firebaseData.jsonObject(); 

      //Print all object data
      Serial.println("Pretty printed JSON data:");
      String jsonStr;
      json.toString(jsonStr, true);
      Serial.println(jsonStr);
      Serial.println();

      Serial.println("Iterate JSON data:");
      Serial.println();
      size_t len = json.iteratorBegin();
      String key, value = "";
      int type = 0;
      for (size_t i = 0; i < len; i++)
      {
        json.iteratorGet(i, type, key, value);
        Serial.print(i);
        Serial.print(", ");
        Serial.print("Type: ");
        Serial.print(type == FirebaseJson::JSON_OBJECT ? "object" : "array");
        if (type == FirebaseJson::JSON_OBJECT)
        {
          Serial.print(", Key: ");
          Serial.print(key);
        }
        Serial.print(", Value: ");
        Serial.println(value);
      }
      json.iteratorEnd();
    }
  }
}

```

NiKola-it, 2020-12-08:
working perfetly! thank you! 
why use the pointer?

mobizt, 2020-12-08:
To use the internal FirebaseJson object to reduce the memory usage.