mobizt/Firebase-ESP8266 #175: Firebase europe-based datacenter location not supported due to domain ending in .app

eligt, 2020-12-07:
Firebase recently added a second location for the real-time database datacenter storage and has now 2 locations, [more details here](https://firebase.google.com/docs/projects/locations#rtdb-locations).

This second location is not working right now - I've looked into the library code and it's due to the domain ending in `.app`, which breaks the URL parsing routine `FirebaseESP8266::getUrlInfo`.

mobizt, 2020-12-07:
I will add support for this changes. Nothing is bug.

closed 2020-12-07 by mobizt

eligt, 2020-12-07:
Well, the URL parsing routine is problematic as it won't parse URLs correctly. I'd recommend parsing the end of the domain as `.anything` using some character class matching rather than checking the individual extensions `.com`, `.edu`, etc. as if Firebase adds any new locations with differing extensions the routine will fail again.

mobizt, 2020-12-07:
Thanks for suggestion.  I think Firebase does not use variety of domains than they just added.