mobizt/Firebase-ESP8266 #176: BUG, Gives me Connection refused when connecting with Wifimanger library

EgHubs, 2020-12-10:
**Describe the bug**
Gives me Connection refused when connecting with Wifimanger library, but it's all okay when connecting with only 
`  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
`
------
Here is my code:
```
#include <FS.h>          // this needs to be first, or it all crashes and burns...
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson

#define FIREBASE_HOST "****************" //Without http:// or https:// schemes
#define FIREBASE_AUTH "**********************"

FirebaseData firebaseData;
FirebaseJson json;
char api_token[32] = "YOUR_API_TOKEN";
String  userIDString;
String path;
//default custom static IP
char static_ip[16] = "10.0.1.56";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";
bool shouldSaveConfig = false;
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}
void setup() {
 

  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
  Serial.begin(115200);  //TEST TO REMOVE
  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);

#ifdef ARDUINOJSON_VERSION_MAJOR >= 6
        DynamicJsonDocument json(1024);
        DeserializationError deserializeError = deserializeJson(json, buf.get());
        serializeJson(json, Serial);
        if (!deserializeError) {
#else
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
#endif
          Serial.println("\nparsed json");
          if (json["ip"]) {
            Serial.println("setting custom ip from config");
            strcpy(static_ip, json["ip"]);
            strcpy(static_gw, json["gateway"]);
            strcpy(static_sn, json["subnet"]);
            Serial.println(static_ip);
          } else {
            Serial.println("no custom ip in config");
          }
[... 221 lines cut ...]
   Serial.println("Gas");
       json.set("SmartHome2","fire");
    Firebase.updateNode(firebaseData, path , json);
      delay(5000);
  }
  else {
        Firebase.deleteNode(firebaseData, path );

    }
}
```
The ONLY DIFFERENCE is that I connected using

`  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
`

**IDE and its version:**
 - Arduino, PlatformIO
 - Version 1.8.13

**ESP8266 Arduino Core SDK version**
 -  Latest version 2.7.4

**Library version**
 - Latest version 3.0.3

mobizt, 2020-12-10:
If you control the Wifi reconnection by third party lib, you should comment this out
 **Firebase.reconnectWiFi(true);**
Because the Firebase library tries to reconnect by stop and start the WiFi at some interval (every 10 seconds) after the connection lost (after 2 seconds) which allow mcu system task to handle WiFi connection for a while before close and reconnect if the connection status still disconnected. This help you don't need to restart the device when WiFi can't resume as many people do.

I recommend to allow the library to do this work for you instead of WiFiManager to polling out because it can't probe the internet connectivity and server connection status and it may delay and not sync with this library and the idle time for system WiFi task may not awailable in WiFiManager.

closed 2020-12-10 by mobizt

mobizt, 2020-12-10:
Did you know, the autoconnect of WiFiManager will switch the device to be AP when the connection to WiFi failed.

EgHubs, 2020-12-11:
I removed that line and still have the same problem, I even set it to false and also the same.

> SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
> mounted file system
> reading config file
> opened config file
> {"api_token":"eJUyTR0hvnVv0Vj7ahwtMERjfMn1","ip":"192.168.1.203","gateway":"192.168.1.1","subnet":"255.255.255.0"}
> parsed json
> setting custom ip from config
> 192.168.1.203
> Path ssIS
> /usersData/eJUyTR0hvnVv0Vj7ahwtMERjfMn1/FireSystem
> *WM: [3] allocating params bytes: 20
> *WM: [2] Added Parameter: api
> *WM: [1] AutoConnect 
> *WM: [2] Connecting as wifi client... 
> *WM: [3] STA static IP: 192.168.1.203
> *WM: [2] Custom static IP/GW/Subnet/DNS 
> *WM: [2] Custom STA IP/GW/Subnet 
> *WM: [1] STA IP set: 192.168.1.203
> *WM: [1] Connecting to SAVED AP: Shalaby
> *WM: [3] Using Password: 01008711248.Sh.
> *WM: [3] WiFi station enable 
> *WM: [3] enableSTA PERSISTENT ON 
> scandone
> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
> scandone
> state: 0 -> 2 (b0)
> state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 4
> cnt 
> 
> connected with Shalaby, channel 8
> ip:192.168.1.203,mask:255.255.255.0,gw:192.168.1.1
> ip:192.168.1.203,mask:255.255.255.0,gw:192.168.1.1
> *WM: [2] Connection result: WL_CONNECTED
> *WM: [3] lastconxresult: WL_CONNECTED
> *WM: [1] AutoConnect: SUCCESS 
> *WM: [1] STA IP Address: 192.168.1.203
> connected...yeey :)
> custom_api_token hereeeeeeeeeeeeeeeee
> 
> local ip
> 192.168.1.203
> 192.168.1.1
> 255.255.255.0
> *WM: [3] freeing allocated params! 
> *WM: [3] unloading 
> Gas
> FAILED
> REASON: connection refused
> ------------------------------------
> 
> 
-----------

> This help you don't need to restart the device when WiFi can't be resumed as many people do.
Will using the Wifimanger alone manages this thing alone, I mean if I turned my router on and off I don't need to reset the ESP it justs tries to reconnect by itself.

mobizt, 2020-12-11:
Did you understand this?

> Did you know, the autoconnect of WiFiManager will switch the device to be AP when the connection to WiFi failed.
> 
> You should check the WiFi mode that is in STA , AP_STA or AP before allow other code that requires internet connection to be executed.

If you don't know what the WiFiManager functions do, learn about it. 
Your algorithm is the problem because you never check the WiFI mode and it's state before calling the Firebase functions.
I don't know what is your expectation to use WiFiManager which it is going to be a captive portal (AP + Web server) to accept the user WiFi setting.

When WiFi is in AP mode, the device can't connect to internet unless in AP_STA mode which acts as AP and also can access to the internet (if the WIFI router provide the internet access).

mobizt, 2020-12-11:
Did you undertand what you set?

char static_ip[16] = "10.0.1.56";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";

Remove this if you don't know what it is.