mobizt/Firebase-ESP8266 #177: Firebase events are missed or not detected

arunanil03, 2020-12-13:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**IDE and its version:**
 - Arduino, PlatformIO
 - Version [e.g. 1.8.9]

**ESP8266 Arduino Core SDK version**
 - Version [e.g. 2.5.1]

**Additional context**
Add any other context about the problem here.

mobizt, 2020-12-13:
When using the shared Firebase Data object for stream and other usagges i.e. normal operation to read and store data,
the stream will be interrupted to use in other tassks, the stream will be resumed (reconnection) after that normal usage was finished.

For the above case, you need to provide the free time for stream to listen to the server event data. The changes on the server at the streaming node path during the stream interruption will be missed.

To avoid this sitation, don't share the usagge of stream's Firebase Data object, use other Firebase Data object instead.

In addition, delay function used in the same loop block of readStream() will interrupt the stream operation, the server data changes may be missed.

More use of Firebase Data object at the same scope i.e more than 2 can leead to out of memory error as the most memory used in Firebase Data object is due to SSL client.

closed 2020-12-13 by mobizt

arunanil03, 2020-12-13:
I am using the stream example provided with the library

mobizt, 2020-12-13:
Even you change the few tens child nodes data simultaneously , the client will only  get few of them as this library uses the REST API  (http request/response) which not the tcp socket connection which can provide the callback handler to handle all multiple  the socket data.

arunanil03, 2020-12-13:
Is there any solution for my requirement ?

mobizt, 2020-12-13:
That's a limit on the mythology used, no solution. 
Google does not provide the socket connection for Firebase as in other client which relies on its SDK.

mobizt, 2020-12-13:
Known the limitation and redesign your database.

arunanil03, 2020-12-13:
when it comes to multiple field update, if i can implement a delay for each filed update  in my application, it can solve the problem right ?

mobizt, 2020-12-13:
You should save the local changes (store it to json), and push theses changes as a JSON data at once.

arunanil03, 2020-12-13:
OK ! will try that , thanks  ! @mobizt

arunanil03, 2020-12-13:
i dont have any problem in receiving events in version 2.0.8 of the library
, i started the project with that version and now when i tried switching
it back to 2.0.8 from the latest version , now all the events are received
one after the other . and no event is missed

>

mobizt, 2020-12-13:
I don't know what you said 2.0.8. There is no version 2.0.8.