mobizt/Firebase-ESP8266 #178: Realtime database server located in EU

opp34, 2020-12-21:
Is the library working with database located on european server? Has anyone tried that so far? Some time ago I noticed that when you create a new firebase project, you may select server's location. In default it gets path **PROJECT_ID.firebaseio.com**; when you pick up european server, it's **PROJECT_ID-default_rtdb.europe-west1.firebasedatabase.app**. When i change hostname accordingly (and update auth) in ESP8266 project, cpu can't connect to database. Is there anything more to be done? Is it supported by the library (firebase site clearly says that this is beta version but on android app's side it works)?

mobizt, 2020-12-21:
You need to update the library due to the host name issue.

opp34, 2020-12-22:
First, before I wrote this topic, I tested with 2.9.x version of library. Yesterday I did an update to 3.0.3 and it doesn't work anyway. I will test that again but at this point it seems that update doesn't fix the issue for me.

mobizt, 2020-12-22:
Sorry for my fault, I think this is already fix. 
This fix is actually ready but waiting for other updates to be completed.

mobizt, 2020-12-22:
Please be patient, the next update will be available soon.

mobizt, 2020-12-23:
The library update is available. 
This issue fixed. Please update

closed 2020-12-23 by mobizt

opp34, 2020-12-25:
Thank you for the update. I can cofirm, it works properly - I can connect with european server by changing only host name and auth. On ESP8266 side it is all that has to be done to switch server from US to EU. 

As a small tip, in Android Studio, as tutorial says, you have download **google-serveces.json** from firebase console and then copy it to  project's build gradle file. What wasn't clear for me, you have to also use the newest dependencies in your project. Your application will not connect to firebase if you don't update your libraries in Android Studio. You can check what's outdated by Tools -> Firebase -> Firebase Assistant -> Realtime Database -> Save and retrieve data -> Check status in point 2. Add the Realtime Database to your app. When you do that, pop-up opens which tells you which libraries have to be renewed.

mobizt, 2020-12-25:
I don't see this paragraph anywhere 

> Your application will not connect to firebase if you don't update your libraries in Android Studio 
You wrote by your own or already states in Firebase tutorial website.

opp34, 2020-12-25:
In tutorial posted on official firebase site there is information about Google service plugin that has to be updated.  So I updated that in project's gradle file, synced it but application didn't connect with database. Then I checked firebase assistant in Android Studio. In realtime database tab there was information that application is connected to database (1st point in assistant) but database is not added to app (2nd point). There was an option to show a pop-up that informs which libraries have to be up to date (in my case there was more than one mentioned in official tutorial). When I did what pop-up stated, status was changed to "dependencies set up correctly" and application could connect with firebase again.  

I propably wouldn't have those issues when I followed option 2 in tutorial but anyway, I wrote that in case someone has troubles with switching project in Android Studio.