mobizt/Firebase-ESP8266 #179: HELP - Design Choice Question

tanishqmanuja, 2020-12-23:
I want to create an IOT device using NodeMCU, which will control 6 appliances. Out of 6 devices 4 devices are high priority. So i want to reduce response time. What would be the best code style according to your library ?

1) Use 6 get method inside loop()
2) Combine the devices in firebase RTDB and use multipath stream for all
3) Use stream for 4 devices and get method inside loop() for other 2

I want to further use Actions-on-Google Smart home api fulfillment to link my devices to google home. So combining the device in RTDB under single node would surely increase code complexity inside my fulfillment. 

What are your suggestions keeping these considerations in mind ?

mobizt, 2020-12-23:
I recommend stream but there are some limitations.

The location or node path in multipath stream, must be under the same parent. This can make the payload that server send to client can be large if the children nodes contains some moderate data. 

The ssl client rx buffer reserved size should be enough for the large payload  and this will reduce your free heap all the time.

The library checked the stream connection and if no heartbeat data (keep-alive event data) sent from server within 40 second, the connection is set to timed out and begin reconnect.

If  your accepted interval to get all 6 nodes data in the loop is not too frequent e.g. every 2-3 minutes and the free memory is concerned, so you can get these data in loop in different times by using schedule task or use different millis time out.

tanishqmanuja, 2020-12-23:
Sorry dev, but i am still little confused. The data is actually pretty small 4 booleans and 2 strings. My main concern in is the response time ... The data will change only once per hour or so on the RTDB ... But the question is when its changed what will provide the best speed for nodeMCU to read that data, Stream or getData ?

mobizt, 2020-12-23:
In stream mode, after connected, client is only waiting for server data without sending any request to server, it faster because there is no need to send the http request to get the response every time as in get function.

mobizt, 2020-12-23:
Stream is design to listen to the change while with get function, you can get the unnecessary old and unchanged data.

tanishqmanuja, 2020-12-23:
Thanks dev for the info and also for the blazing fast reply. Stream is the way to go then :)
You can mark it closed then.

closed 2020-12-23 by mobizt