mobizt/Firebase-ESP8266 #180: data type mismatch with Email_Password example

EgHubs, 2020-12-24:
First of all, i need to thank you for your great library and especially your latest update :)
**Describe the bug**
I tried using the Email_Password example and what I get is a data type mismatch.
what I only changed with the example code is that I commented the following 3 lines
 ```
 // String var = "$user";
  // String val = "(auth.uid === $user)";
  //  prepareDatabaseRules(base_path.c_str(), var.c_str(), val.c_str(), val.c_str());
```
Also, I put my email, password, api_key, FIREBASE_HOST, and changed base_path to my path.+
Also, I enabled the Identity provider as you described.
I am sure that the library did a sign up with the email provided as I printed the path and I get my userID that is assigned to that email.
this is what I get at the serial monitor

```
Token info: type = id token, status = ready

------------------------------------
Set int test...
/usersData/eJUyTR0hvnVv0Vj7ahwtMERjfMn1    // here is   Serial.println(path);
FAILED
REASON: data type mismatch

```
also tried to pass a json instead of count++ and still the same
My FireBase rules are : 

```
{
    "rules": {
        "userData": {
            "$user_id": {
                ".write": "auth.uid != null &&$user_id === auth.uid",
                 //".read": "auth.uid != null &&$user_id === auth.uid"
            }
        },
    }
}
```
and they are working with the rules playground.

**IDE and its version:**
 - Arduino, PlatformIO
 - Version [e.g. 1.8.13]

**ESP8266 Arduino Core SDK version**
 - Version [e.g. 2.7.4]

mobizt, 2020-12-24:
Your data at that uid node may be large and the ssl client buffer is not enough and the payload sent to client can't be recognized. Please try to increase this

```cpp

 //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
  fbdo.setBSSLBufferSize(1024, 1024); <-- rx/tx

  //Set the size of HTTP response buffers in the case where we want to work with large data.
  fbdo.setResponseSize(1024);

```

EgHubs, 2020-12-25:
Will i tried 
```
Firebase.reconnectWiFi(true);
fbdo.setBSSLBufferSize(4096, 1024); //and also tried (4096, 4096)
fbdo.setResponseSize(4096);
```
also
```
  Firebase.reconnectWiFi(true);
fbdo.setBSSLBufferSize(1024, 1024);
fbdo.setResponseSize(1024);
```
and the same problem exists,
the data i'am parsing to the cloud is "count++" which is the default integer variable given in the example.
`    if (Firebase.set(fbdo, path + "/int", count++))
`
and here is my full example code which i didn't change anything else in it:

```
/**
   Created by K. Suwatchai (Mobizt)

   Email: k_suwatchai@hotmail.com

   Github: https://github.com/mobizt

   Copyright (c) 2020 mobizt

*/

/** This example will show how to authenticate as a user with Email and password.

   You need to enable Email/Password provider.
   In Firebase console, select Authentication, select Sign-in method tab,
   under the Sign-in providers list, enable Email/Password provider.

   From this example, the user will be granted to access the specific location that matches
   the user uid.

   This example will modify the database rules to set up the security rule which need to
   guard the unauthorized access with the user Email.
*/

#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "*******"
#define WIFI_PASSWORD "*******"

/* 2. Define the Firebase project host name (required) */
#define FIREBASE_HOST "***********"

/** 3. Define the API key

   The API key (required) can be obtained since you created the project and set up
   the Authentication in Firebase conssole. Then you will get the API key from
   Firebase project Web API key in Project settings, on General tab should show the
   Web API Key.

   You may need to enable the Identity provider at https://console.cloud.google.com/customer-identity/providers
   Select your project, click at ENABLE IDENTITY PLATFORM button.
   The API key also available by click at the link APPLICATION SETUP DETAILS.

*/
#define API_KEY "*************************"

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "new@test.com"
#define USER_PASSWORD "********"
[... 377 lines cut ...]
    case token_status_on_refresh:
      return "on refreshing";

    case token_status_ready:
      return "ready";

    case token_status_error:
      return "error";

    default:
      break;
  }
  return "uninitialized";
}

/* The helper function to get the token error string */
String getTokenError(struct token_info_t info)
{
  String s = "code: ";
  s += String(info.error.code);
  s += ", message: ";
  s += info.error.message.c_str();
  return s;
}
```

EgHubs, 2020-12-25:
i tried to set
`fbdo.setwriteSizeLimit(fbdo, "tiny");
`
but it says that
`class FirebaseData' has no member named 'setwriteSizeLimit.`
although its still in the [documentation](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/README.md#set-the-size-limit-of-payload-data-that-will-write-to-the-database-for-each-request)

mobizt, 2020-12-25:
Please read carefully, it belong to Firebase object.

**Firebase.setwriteSizeLimit**

mobizt, 2020-12-25:
Change your database rule to the default
```

{
  "rules": {
    ".read": "auth.uid != null",
    ".write": "auth.uid != null"
  }
}
```

If you still get the same error, check the FIREBASE_HOST. 
Make sure that you provide the correct host name.

mobizt, 2020-12-25:
The server returns something that look like not not function.
Did your project is already in Blaze plan?

EgHubs, 2020-12-25:
> Make sure that you provide the correct hostname.

I was providing

`*****.firebaseapp.com 
`

instead of 

` https://*****.firebaseio.com/
`
Sorry, my fault :).
 I am really thankful for what you have done, no doubt I will be a part of the people who are sponsoring this great library and especially the latest update :)

closed 2020-12-25 by EgHubs

mobizt, 2020-12-25:
No problem, I also try with this url (firebaseapp) like you before and get the huge of full web page content response which I try to reject this but may not properly catch the error causes that server may response something differently that I can't be checked them all the case at this time.