mobizt/Firebase-ESP8266 #181: How to connect esp8266 with Wi-Fi

Paweena1997, 2021-01-09:
[image]

mobizt, 2021-01-09:
Please clarify your question.

I don't know what do you mean.

This is Firebase library.

closed 2021-01-09 by mobizt

Paweena1997, 2021-01-10:
what caused this problem

mobizt, 2021-01-10:
I don't know what's your problem.

mobizt, 2021-01-10:
Your question is **How to connect esp8266 with Wi-Fi** and nothing related to the library.

Paweena1997, 2021-01-10:
l can not connect esp8266 with Wi-Fi

mobizt, 2021-01-10:
It's not related to the library. 
You can search google how to connect the WiFi instead of open the issue here.

Paweena1997, 2021-01-10:
l using your library but can not connect

mobizt, 2021-01-10:
Your question is not clear.

Post your code here and paste all error message here.

mobizt, 2021-01-10:
You must provide more info when asking someone help because nobody know your code.

Paweena1997, 2021-01-10:
will say  cry .
you have faccebook.

Paweena1997, 2021-01-10:
Don't know how to tell

mobizt, 2021-01-10:
I see. Are you from India or Thailand?

You can send your code and error message to Email k_suwatchai@hotmail.com.

Actually I'm not convenient to use Email to support anyone.

Paweena1997, 2021-01-10:
l am connecteed error

Paweena1997, 2021-01-10:
คือสาเหตุมันเป็นแบบนี้คือไม่สามารถเชื่อมต่อกับFirebaseได้คืออยากทราบว่าเป็นที่อะไร

mobizt, 2021-01-11:
You need to post your code and error message that printed on the serial here.
I can't help without your code.