mobizt/Firebase-ESP8266 #184: Help

lchvsanilkumar, 2021-01-26:
Hello sir, I want to pause firebase while internet is down because of I need to run another loop and when the internet is available ,continue the firebas e operation....(wifi is always available)....please help sir

mobizt, 2021-01-26:
You can use [pauseFirebase](https://github.com/mobizt/Firebase-ESP8266/tree/master/src#pauseunpause-wificlient-from-all-firebase-operations) function of the FirebaseData object.

Example, if your FirebaseData object dedined as fbdo.

```cpp
//To pause,
fbdo.pauseFirebase(true);

//To unpause,
fbdo.pauseFirebase(false);
```

After you pause the Firebase data, the http session will be terminated and the memory used by the internal SSL client will be released.

closed 2021-01-26 by mobizt

lchvsanilkumar, 2021-01-26:
Thank you ,I will try