mobizt/Firebase-ESP8266 #187: HELP

Rujengelal, 2021-02-14:
Cant parse json . says no matching function for FirebaseJson::get(FirebaseData&, const char [4])

ARDUINO IDE LOG:
C:\Users\Rujen\Documents\BEX III-II minor project\esp.ino\esp.ino.ino: In lambda function:

esp.ino:185:34: error: no matching function for call to 'FirebaseJson::get(FirebaseData&, const char [4])'

       json.get(firebaseData,"co2");

                                  ^

C:\Users\Rujen\Documents\BEX III-II minor project\esp.ino\esp.ino.ino:185:34: note: candidate is:

In file included from C:\Users\Rujen\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/FirebaseESP8266.h:55:0,

                 from C:\Users\Rujen\Documents\BEX III-II minor project\esp.ino\esp.ino.ino:2:

C:\Users\Rujen\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/FirebaseJson.h:472:10: note: bool FirebaseJson::get(FirebaseJsonData&, const String&, bool)

     bool get(FirebaseJsonData &jsonData, const String &path, bool prettify = false);

          ^

mobizt, 2021-02-14:
Read [this](https://github.com/mobizt/Firebase-ESP8266/tree/master/src#get-the-value-from-the-specified-node-path-in-firebasejson-object).

See the example [this](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/FirebaseJson/examples/basic/basic.ino#L25) and [this](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/FirebaseJson/examples/basic/basic.ino#L57).

closed 2021-02-14 by mobizt

Rujengelal, 2021-02-14:
Im sorry if i make a noob mistake but i still dont under stand whats worng

My code is:
```
Firebase.getJSON(firebaseData, "/set/"+server.arg("nodeId"));
 FirebaseJson json1;
    FirebaseData firebaseData1;
      json1=firebaseData.jsonObject();

      json1.get(firebaseData1,"co2");

```
Output of firebaseData.jsonString() is {"co2":450,"humidity":50,"light":30,"moisture":90,"temperature":25}

I referred to [this ](https://github.com/mobizt/Firebase-ESP8266/issues/83)solution provided by you

mobizt, 2021-02-14:
```cpp
String path = "/set/"+server.arg("nodeId");
Firebase.getJSON(firebaseData, path);

FirebaseJson  &json1 = firebaseData.jsonObject();
FirebaseJsonData  jsonData;

 json1.get(jsonData,"co2");

if (jsonData.typeNum == FirebaseJson::JSON_BOOL)
   Serial.println(jsonData.boolValue ? "true" : "false");
else if (jsonData.typeNum == FirebaseJson::JSON_INT)
   Serial.println(jsonData.intValue);
else if (jsonData.typeNum == FirebaseJson::JSON_FLOAT)
   Serial.println(jsonData.floatValue);
else if (jsonData.typeNum == FirebaseJson::JSON_DOUBLE)
   printf("%.9lf\n", jsonData.doubleValue);
else if (jsonData.typeNum == FirebaseJson::JSON_STRING ||  jsonData.typeNum == FirebaseJson::JSON_NULL || 
 jsonData.typeNum == FirebaseJson::JSON_OBJECT ||  jsonData.typeNum == FirebaseJson::JSON_ARRAY)
   Serial.println(jsonData.stringValue);

```