mobizt/Firebase-ESP8266 #191: HELP

danmov26, 2021-03-10:
I want to make a Wi-Fi switch, you can also turn on / off the light with a button. Each time I press the button, I need to change the value in Firebase RTDB - /Status - true/false and add events to /Events. 

[image]

I use Firebase.setBool and setInt. But each such operation is delayed, and then I cannot change the status faster. You need to wait more than 3 seconds before the operation is completed and you can press the button again. Tell me if it's possible somehow to do it faster. Is it possible to do this in another thread with ESP8266?

mobizt, 2021-03-10:
No such that delay in Firebase.

You can try with example.

Check your code what is the blocking operation inside your code then remove it e.g. delay  and use millis instead.

closed 2021-03-10 by mobizt

danmov26, 2021-03-12:
I tried your example. The speed of adding a value to the database is not as fast (0.5-2s) as I would like, as, for example, in an Android device (there it is done instantly). Looks like ESP8266 is just weak. Am I thinking correctly?

danmov26, 2021-03-12:
Сan't I somehow do it instantly? Multithreading or something else?

mobizt, 2021-03-12:
Not correct and there's something that need to clarify your understanding.

Post your code or example you've tried and tell me where is the line of code that uses much time as you said.

mobizt, 2021-03-12:
Look at this [usage example](https://medium.com/@vibrologic/serverless-iots-with-firebase-realtime-database-and-esp32-def049181b57).

danmov26, 2021-03-12:
At this speed, ESP8266 works (example - Basic)

https://user-images.githubusercontent.com/66782790/110973054-c7b2f480-836d-11eb-93e1-3cc71751a0d9.mp4

At this speed, Android Galaxy M20 works

https://user-images.githubusercontent.com/66782790/110973195-fb8e1a00-836d-11eb-90d8-1c840f72135c.mp4

mobizt, 2021-03-12:
Are you kidding.

Is it fair to compare low resource device and unlimited resource device?

And between Firebase SDK or Admin SDK which relies on Google TCP socket technology and REST APIs via http request.

The esp8266 does not have the large mem buffer for sending large data at once, it need to send as chunks.  In http protocol, the request contains header included the large payload of key or token than other protocols.  PUT, POST, GET used more time than the TCP or UDP that the socket is already open between client and server and ready to transfer.

Please learn the more about the web socket, MQTT and REST APIs.

mobizt, 2021-03-12:
Google does not provide such web socket method for other client unless that client can be install the Firebase SDK or Firebase Admin SDK.