mobizt/Firebase-ESP8266 #192: Connection refused WifiNinna

henryhd98, 2021-03-23:
this was working perfectly for 2 months but now suddenly stopped working! 

#include "Firebase_Arduino_WiFiNINA.h"
#include <WiFiNINA.h>
#include <SPI.h>
#define LED1 13
#define LED2 12
#define LED3 11
#define LED4 10
#define LED5 9
#define LED6 8
#define LED7 7

#define FIREBASE_HOST "xxxxxx-default-rtdb.europe-west1.firebasedatabase.app/"
//database api url
#define FIREBASE_AUTH "xxxxxxxxxxxx"
//database secret
#define WIFI_SSID "xxxxxx"
#define WIFI_PASSWORD "xxxxxxxx"

char ssid[] = WIFI_SSID;
char pass[] = WIFI_PASSWORD;
char FBHost[] = FIREBASE_HOST;
char FBAuth[] = FIREBASE_AUTH;

//Define Firebase data object
FirebaseData firebaseData;

String path = "/LEDS";

void setup() {
  Serial.begin(9600);
  pinMode(LED1, OUTPUT);
  pinMode(LED2, OUTPUT);
  pinMode(LED3, OUTPUT);
  pinMode(LED4, OUTPUT);
  pinMode(LED5, OUTPUT);
  pinMode(LED6, OUTPUT);
  pinMode(LED7, OUTPUT);

  int status = WL_IDLE_STATUS;
  while (status != WL_CONNECTED)
  {
    status = WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.print(".");
    delay(300);
    //Wait until connected
    delay(100);
    digitalWrite(LED1, HIGH);    // turn on LED1
    delay(100);                  // wait for 200ms
    digitalWrite(LED4, HIGH);    // turn on LED2
    delay(100);                  // wait for 200ms
    digitalWrite(LED7, HIGH);    // turn on LED3
    delay(100);                  // wait for 200ms
    digitalWrite(LED1, LOW);     // turn off LED3
    delay(200);                  // wait for 300ms before running program all over again
    digitalWrite(LED4, LOW);     // turn off LED3
    delay(200);                  // wait for 300ms before running program all over again
    digitalWrite(LED7, LOW);     // turn off LED3
    delay(200);                  // wait for 300ms before running program all over again
    //Blink till wifi connected
    Serial.print(" Waiting for connection.. ");
  }

  Serial.println("");
  Serial.println("WiFi Connected");
  digitalWrite(LED1, HIGH);
  delay(1000);
  digitalWrite(LED1, LOW);
  delay(1000);
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.begin(9600);
  delay(1000);
  Serial.println();

  Firebase.begin(FBHost, FBAuth, ssid, pass);    //Provide the autntication data & connect to real time DB

  Firebase.reconnectWiFi(true);
  delay(1000);

 Serial.println("REASON: " + firebaseData.errorReason());
}

void loop() {
  int val = 0;

  if (Firebase.getInt(firebaseData, "/LEDS")) {

    //Success, then read the payload value

    //Make sure payload value returned from server is integer
    //This prevent you to get garbage data
    if (firebaseData.dataType() == "int") {
      val = firebaseData.intData();
      Serial.println(val);
    }

  } else {
    //Failed, then print out the error detail
    Serial.println(firebaseData.errorReason());
  }
}

the serial monitor is giving me connection refused, I have a project presentation tomorrow and this is giving me a problem :(
any ideas why?

mobizt, 2021-03-24:
You post in the wrong repos, please go to the  repos for WiFiNINA and open isue there.

closed 2021-03-24 by mobizt

mobizt, 2021-03-24:
The library works as normally.

I test with Arduino MKR WiFi 1010.

Please try to renew the database secret.

henryhd98, 2021-03-25:
Thanks that worked!