mobizt/Firebase-ESP8266 #193: HELP How to use JWT generated by other programs

professor-ti, 2021-04-12:
Is it possible to authenticate with JWT generated by other programs? I generated a JWT in javascript with Service Account and would like to use it in esp8266.

I tried to use the Legacy_Token code by changing the code snippets below, but I was not successful:
```
#define FIREBASE_AUTH "<JWT TOKEN>"
...
//config.signer.tokens.legacy_token = FIREBASE_AUTH;
config.signer.tokens.jwt = FIREBASE_AUTH;
config.signer.tokens.token_type = token_type_custom_token;
config.signer.tokens.status = token_status_ready;
...
```

mobizt, 2021-04-12:
I don't know what you're doing.

The JWT token is not the auth token used in Google services. 

JWT token in Google authentication is for id token and OAuth2.0 tokens exchanging request.

After sending the token exchanging request, the Google auth server will return the id or auth token that client can be used in Google services authentication.

mobizt, 2021-04-12:
The library provides many authentication methods.
Did you ever look into [this](https://github.com/mobizt/Firebase-ESP8266/tree/master/examples/Authentications)?

closed 2021-04-12 by mobizt

mobizt, 2021-04-12:
And can read service account json file directly as[ this example](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Authentications/Access_Token_from_File/Access_Token_from_File.ino#L86).

mobizt, 2021-04-12:
Don change the **config.signer.tokens** data because it is used for internal token generation process only which data stored in the user defined config.

professor-ti, 2021-04-13:
Hey dude. First of all, thanks for the library. It will help a lot.

> I don't know what you're doing.
> 
> The JWT token is not the auth token used in Google services.
> 
> JWT token in Google authentication is for id token and OAuth2.0 tokens exchanging request.
> 
> After sending the token exchanging request, the Google auth server will return the id or access token that client can be used in Google services authentication.

I would like to generate the JWT on my secure server and not on Esp, according to the official Firebase video and page:
[Firecasts - Firebase](https://www.youtube.com/watch?v=WtYzHTXHBp0) and [Firebase docs/auth](https://firebase.google.com/docs/auth/admin/create-custom-tokens#create_custom_tokens_using_the_firebase_admin_sdk). This way I don't need to share Firebase's private key, just the generated JWT strings.

> The library provides many authentication methods.
> 
> Did you ever look into [this](https://github.com/mobizt/Firebase-ESP8266/tree/master/examples/Authentications)?
> 
> The library has the built-in token generation using the service account and you don't need any third party program to create.

Yes, I checked and tested the other methods before posting the question.

Anyway, JWT expires if not used, so I will modify the algorithm and the web server to use authentication with login and password instead of Tokens.

mobizt, 2021-04-13:
Again signed JWT itself is nothing to do with Google authentication. 

You need to exchange it with the id token (generated with custom claims JWT token) or access token (OAuth2.0).

The library does not provide the way to interfere this signing process with external token because I don't want to do extra work to check the validity and expiration of that token which is not control by the library.

You need to know that the device that use custom token (actually id token) and access token (Auth2.0) for sign-in is the  server admin right access.  

You should used the proper sign-in method for different kind of users.

If you want to distribute the device to normal user, use Email/Password sign-in method instead. Use the id or access token sign-in with the specific device that own by you or trusted user.

mobizt, 2021-04-13:
It's not good to let normal user controls your cloud storage and access the admin rights Google services no matter you hind or expose the sign-in credentials to that user.

professor-ti, 2021-04-13:
Ok, man. Thanks!