mobizt/Firebase-ESP8266 #194: HELP - DHT22 sensor

RagingNovaOP, 2021-04-16:
Hello, I have been trying to use a few libraries and fixes from some old threads on the Firebase master. 
However nothing has worked for me. I have been trying your library out and I was hoping to get some help in regards to sending temperature readings to the Firebase database. Below is the code I currently have, I just removed the connection details.

`#include <common.h>
#include <FirebaseESP8266.h>
#include <FirebaseFS.h>
#include <Utils.h>
#include <ESP8266WiFi.h>
#include "DHT.h"

#ifndef FIREBASE_HOST
#define FIREBASE_HOST "*"
#define FIREBASE_PASS "*"
#endif

#ifndef STASSID
#define STASSID "*" // Your WiFi SSID
#define STAPSK  "*" //Your WiFi password
#endif

#define DHTP22 12
#define DHTTYPE DHT22
DHT dht(DHTP22, DHTTYPE);

FirebaseData fbdo;
FirebaseConfig config;

const char* ssid = STASSID;
const char* password = STAPSK;

const char* fHost = FIREBASE_HOST;
const char* fPass = FIREBASE_PASS;

void setup() {
  dht.begin();
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");
  //Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
   }
  if (WiFi.status() == WL_CONNECTED){
      uint8_t macAddr[6];
      WiFi.macAddress(macAddr);
      Serial.printf("Connected, mac address: %02x:%02x:%02x:%02x:%02x:%02x\n", macAddr[0], macAddr[1], macAddr[2], macAddr[3], macAddr[4], macAddr[5]);
   }  
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  Firebase.begin(fHost, fPass);
  Firebase.reconnectWiFi(true);
  fbdo.setBSSLBufferSize(1024, 1024);
  fbdo.setResponseSize(1024);
}
  //Firebase.begin(FIREBASE_HOST);

}

void loop() {
  float t = dht.readTemperature();
  if(isnan(t)){
   //Check for sensor input
    Serial.println("No readings from sensor");
    return;
  }
  Serial.print("Temperature: ");
  Serial.print(t);
  Serial.print("°C ");
  String fTemp = String(t) + String("°C ");
  //Sending the temperature to the database
  Firebase.pushString("/DHT11/Temperature", fTemp);
  //If there is an issue with the connection an error is displayed.
  if(Firebase.failed()){
    Serial.print("Could not connect to database");
    Serial.println(Firebase.error());
    return;
  }
  //Firebase.setFloat("Temp", t);
  delay(3000);
}`
Currently I have been getting readings from my sensor, however, the "Could not connect to database" is printed every time. 
Any help would be appreciated!
Much thanks.

mobizt, 2021-04-16:
The syntax is not valid for this library.

I think you are confused between Firebase Arduino library and this Firebase ESP8266 Client which totally different in API and functions.

If you are new to this library, please read the [doc](https://github.com/mobizt/Firebase-ESP8266#firebase-realtime-database-arduino-library-for-esp8266) and test with ready to run [examples](https://github.com/mobizt/Firebase-ESP8266/tree/master/examples).

The examples also are in the installed folder of library.

Spend your time to test and learn how to use this library.

closed 2021-04-16 by mobizt

mobizt, 2021-04-16:
This is the [basic example](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Basic/Basic.ino) that ready to run.

Just edit  WIFI_SSID and WIFI_PASSWORD with your WiFi credentials. 
Edit FIREBASE_HOST with your database host which looks similar to xxxxx.firebaseio.com or xxxxxx.firebaseio.app
and also edit FIREBASE_AUTH with database secret which you get it from the Firebase console.

[image]

[image]

RagingNovaOP, 2021-04-16:
Thank you so much. 
The basic example connects successfully and populates the database! With any other library I could not connect at all. 
Is it still possible to use this library for my purposes however? I have a DHT22 connected to an ESP8266, this should send the temperature data to Firebase. Then I should be able to retrieve the data on an app I am still working on. 

So would using `Firebase.push("/Temperature", fTemp);` work to push the data to the database?
Because I've been going through the code provided in the Basic example and I cannot find any `push` apart from in the `if` statements. I do apologize I am extremely new to this and have been trying many things so my mind has been very muddled. 

Thanks again for the help, I am so happy that I can finally connect!

mobizt, 2021-04-16:
It's already in the basic example
https://github.com/mobizt/Firebase-ESP8266/blob/754c3d3d0a69431d607f008ada05b6511fbedad7/examples/Basic/Basic.ino#L136

The functions details are [here](https://github.com/mobizt/Firebase-ESP8266/tree/master/src#append-new-integer-value-to-the-defined-database-path).

I don't get your first paragraph but the Firebase ESP32 library is [here](https://github.com/mobizt/Firebase-ESP32).

RagingNovaOP, 2021-04-16:
I got working!!!!!! 
I figured out a bit more of the code, after realizing how the `if` statements work I just swapped out some of my code. 
It's working perfectly, thanks so much. 
Just need to add a lot more polish to it and should have something better than recording all temperatures. 
Anyhow, thank you for working on this, my project would be impossible without it. I will reference you of course :) 
Have a great weekend!