mobizt/Firebase-ESP8266 #195: HELP Question about HTTPS and HTTP requests

EgHubs, 2021-04-18:
Please excuse my noob question, but I just saw these two videos ([1](https://www.youtube.com/watch?v=hMZdM0Wlt54
) and [2](https://www.youtube.com/watch?v=Wm1xKj4bKsY
)) and they are talking about how to make an HTTPS request from ESP8266  instead of the normal nonsafe HTTP request.
my question here, I am sure that your great library is based on an HTTP request to the FIreBase site, I know that the FireBase site is protected with HTTPS, but these videos got me confused, do I need to worry about my data or about my request even if I am trying to call a secured site like FireBase, also if that what it is, I need to know what is the worst-case scenario, please.

mobizt, 2021-04-18:
No matter you or someone called HTTP request or HTTPS request, in the application layer (client or library), both have the same data (header + payload), only port is different, most HTTP go to port 80 and HTTPS go to 443.

The http and https are the protocols that the data sending between server and client through the networks is in the plain text (no encryption) or in encrypted data with SSL encryption.  There are many [resources](https://www.guru99.com/difference-http-vs-https.html) about this on the internet.

Google services included Firebase accept only secure connection on port 443 which the library works with SSL client to send the encrypt and decrypt the receive data via secure port 443 and no matter someone called http or http request.

closed 2021-04-18 by mobizt

mobizt, 2021-04-18:
You should not ask this kind of basic question as you are programmer or dev that involve with communication related application or project which you need to understand this well.

mobizt, 2021-04-18:
And also HTTP response and HTTPS response are no different at the application end as it decrypted by the SSL client already.