mobizt/Firebase-ESP8266 #197: BUG

RitvikSardana, 2021-05-11:
Multiple libraries were found for "SD.h"
In file included from C:\Users\ritvi\Documents\Arduino\libraries\Firebase-ESP8266-master\src/common.h:45:0,
 Used: D:\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\SD
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\SD
                 from C:\Users\ritvi\Documents\Arduino\libraries\Firebase-ESP8266-master\src/Utils.h:37,
                 from C:\Users\ritvi\Documents\Arduino\libraries\Firebase-ESP8266-master\src/signer/Signer.h:37,
                 from C:\Users\ritvi\Documents\Arduino\libraries\Firebase-ESP8266-master\src/FirebaseESP8266.h:53,
                 from C:\Users\ritvi\Documents\Arduino\libraries\Firebase-ESP8266-master\examples\Stream\Stream.ino:17:
C:\Users\ritvi\Documents\Arduino\libraries\Firebase-ESP8266-master\src/wcs/esp8266/FB_HTTPClient.h:56:22: fatal error: LittleFS.h: No such file or directory
 #include <LittleFS.h>
                      ^
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

P.S  -  I have installed the LittleFS library

mobizt, 2021-05-11:
Don't install the SD library from third party. It's conflicted with the ESP8266 SD Core library which used by this library. and make the ambiguous naming conflicts for C++ compiler.

Remove this folder C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\SD

RitvikSardana, 2021-05-11:
There is no such folder in my pc
@mobizt

mobizt, 2021-05-11:
It's not necessary to show your file as it is hidden and it's your installation problem which you should post at the forum.

This is not related to the library. There are many issues related to third party SD library installation in this repos which you can search.

Don't install the Arduino IDE app from Windows Store because it has the path issue. Go to Arduino website and download the installer file or zip file.

RitvikSardana, 2021-05-11:
Which library are you telling me to install?

mobizt, 2021-05-11:
Re install the Arduino IDE from Arduino website.

You currently have Arduino App installed as it shows here C:\Program Files\WindowsApp.

It's your PC and the error tells the true what is the problem. 

Delete Arduino from this folder C:\Program Files\WindowsApp or delete this WindowsApp folder.

orewayaya, 2021-05-17:
how about this problem sir..... even if i deleted the SD file at that location, the error messages just become shorter, still cannot run.

Multiple libraries were found for "SD.h"
In file included from C:\Users\yaya\Documents\Arduino\libraries\Firebase-ESP8266-master\src/common.h:45:0,
 Used: C:\Users\yaya\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
                 from C:\Users\yaya\Documents\Arduino\libraries\Firebase-ESP8266-master\src/Utils.h:37,
                 from C:\Users\yaya\Documents\Arduino\libraries\Firebase-ESP8266-master\src/signer/Signer.h:37,
                 from C:\Users\yaya\Documents\Arduino\libraries\Firebase-ESP8266-master\src/FirebaseESP8266.h:53,
                 from C:\Users\yaya\AppData\Local\Temp\arduino_modified_sketch_214887\Basic.ino:18:
C:\Users\yaya\Documents\Arduino\libraries\Firebase-ESP8266-master\src/wcs/esp8266/FB_HTTPClient.h:67:22: fatal error: LittleFS.h: No such file or directory
 #include <LittleFS.h>
                      ^
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

mobizt, 2021-05-17:
@orewayaya  I already reply in your opened issue.

Please install the ESP8266 Arduino Core SDK from Board Manager.

Or update it to v2.7.4.

vienbu, 2021-07-16:
> @orewayaya I already reply in your opened issue.
> 
> Please install the ESP8266 Arduino Core SDK from Board Manager.
> 
> Or update it to v2.7.4.

Hi, I try to updated ESP8266 v2.7.4 but it have this error, can u help me 
"C:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\vienp\\AppData\\Local\\Temp\\arduino_build_978173/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DNEW_OOM_ABORT -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DDEBUG_ESP_SSL -DDEBUG_ESP_HTTP_CLIENT -DARDUINO=10813 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\cores\\esp8266" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\variants\\nodemcu" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\ESP8266WiFi\\src" "-IC:\\Users\\vienp\\OneDrive\\Documents\\Arduino\\libraries\\Firebase-ESP8266-master\\src" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\SPI" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\SD\\src" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\SDFS\\src" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\ESP8266SdFat\\src" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\LittleFS\\src" "C:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\ESP8266WiFi\\src\\ESP8266WiFi.cpp" -o "C:\\Users\\vienp\\AppData\\Local\\Temp\\arduino_build_978173\\libraries\\ESP8266WiFi\\ESP8266WiFi.cpp.o"
"C:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\vienp\\AppData\\Local\\Temp\\arduino_build_978173/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DNEW_OOM_ABORT -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DDEBUG_ESP_SSL -DDEBUG_ESP_HTTP_CLIENT -DARDUINO=10813 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\cores\\esp8266" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\variants\\nodemcu" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\ESP8266WiFi\\src" "-IC:\\Users\\vienp\\OneDrive\\Documents\\Arduino\\libraries\\Firebase-ESP8266-master\\src" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\SPI" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\SD\\src" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\SDFS\\src" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\ESP8266SdFat\\src" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\LittleFS\\src" "C:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\ESP8266WiFi\\src\\ESP8266WiFiAP.cpp" -o "C:\\Users\\vienp\\AppData\\Local\\Temp\\arduino_build_978173\\libraries\\ESP8266WiFi\\ESP8266WiFiAP.cpp.o"
"C:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\vienp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\vienp\\AppData\\Local\\Temp\\arduino_build_978173/core" -c -w -Os -g -mlo [... cut ...]
Using library SDFS at version 0.1.0 in folder: C:\Users\vienp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SDFS 
Using library ESP8266SdFat at version 1.1.0 in folder: C:\Users\vienp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266SdFat 
Using library LittleFS at version 0.1.0 in folder: C:\Users\vienp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\LittleFS 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

mobizt, 2021-07-16:
@vienbu 
Uninstall third party SD library by delete this folder **"D:\Arduino\libraries\SD"**

vienbu, 2021-07-16:
> @vienbu
> Uninstall third party SD library by delete this folder **"D:\Arduino\libraries\SD"**

it still not work with the same errror
C:\Users\vienp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp: In static member function 'static BearSSL::CertStore::CertInfo BearSSL::CertStore::_preprocessCert(uint32_t, uint32_t, const void*)':
C:\Users\vienp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp:25:31: error: 'DEBUG_ESP_PORT' was not declared in this scope
 #define DEBUG_BSSL(fmt, ...)  DEBUG_ESP_PORT.printf_P((PGM_P)PSTR( "BSSL:" fmt), ## __VA_ARGS__)
                               ^
C:\Users\vienp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp:56:5: note: in expansion of macro 'DEBUG_BSSL'
     DEBUG_BSSL("CertStore::_preprocessCert: OOM\n");
     ^
C:\Users\vienp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp: In static member function 'static const br_x509_trust_anchor* BearSSL::CertStore::findHashedTA(void*, void*, size_t)':
C:\Users\vienp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp:25:31: error: 'DEBUG_ESP_PORT' was not declared in this scope
 #define DEBUG_BSSL(fmt, ...)  DEBUG_ESP_PORT.printf_P((PGM_P)PSTR( "BSSL:" fmt), ## __VA_ARGS__)
                               ^
C:\Users\vienp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp:208:9: note: in expansion of macro 'DEBUG_BSSL'
         DEBUG_BSSL("CertStore::findHashedTA: OOM\n");

mobizt, 2021-07-16:
@vienbu
Don't quote the post because you post on other one issue.
You should use @ followed by the name to not disturb other people.

The compile error is about improper debug settings in your Arduino IDE which is not related to library.
You set the Arduino IDE debug to OOM without setting debug port.

Change the debug port option to your serial port.

vienbu, 2021-07-16:
@mobizt thanks you verry much, it worked.

rwlakin, 2022-05-31:
Mobizt
How did SD.h  get installed?  Was it thru Arduino Manage Libraries?

mobizt, 2022-05-31:
@[rwlakin](https://github.com/rwlakin)

The SD library is already installed which is one of the core sdk libraries.