mobizt/Firebase-ESP8266 #203: HELP Hostname for firebase

fionala-code, 2021-05-22:
Hello, I'm trying to find the host name on my firebase project so i can add it in the initialization steps outlined in the Read.me file. However, I'm not exactly sure where I can find the host name. Some guidance would be appreciated!

mobizt, 2021-05-22:
https://github.com/mobizt/Firebase-ESP8266#authentication

[image]

closed 2021-05-22 by mobizt

mobizt, 2021-05-22:
The FIREBASE_HOST is no more existed in all examples, It replaced by FIREBASE_URL
The FIRBASE_AUTH also replaced by DATABASE_SECRET

All new examples will uses Email/Password for authentication for more secure.

https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Basic/Basic.ino