mobizt/Firebase-ESP8266 #205: HELP - Permission Denied Even In Basic Example

AlperSal, 2021-06-01:
Code;
```
/**
 * Created by K. Suwatchai (Mobizt)
 * 
 * Email: k_suwatchai@hotmail.com
 * 
 * Github: https://github.com/mobizt
 * 
 * Copyright (c) 2021 mobizt
 *
*/

#if defined(ESP32)
#include <WiFi.h>
#include <FirebaseESP32.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>
#endif

//Provide the token generation process info.
#include "addons/TokenHelper.h"
//Provide the RTDB payload printing info and other helper functions.
#include "addons/RTDBHelper.h"

/* 1. Define the WiFi credentials */
#define WIFI_SSID "Filled in my code"
#define WIFI_PASSWORD "Filled in my code"

/* 2. Define the API Key */
#define API_KEY "AIzaSyBilbEaam89o***************"

/* 3. Define the RTDB URL */
#define DATABASE_URL "********.firebaseio.com" //<databaseName>.firebaseio.com or <databaseName>.<region>.firebasedatabase.app

/* 4. Define the user Email and password that already registered or added in your project */
#define USER_EMAIL "alpersal235@gmail.com"
#define USER_PASSWORD "also I set email and password in Firebase console"

//Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;

int count = 0;

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  /* Assign the api key (required) */
  config.api_key = API_KEY;

  /* Assign the user sign in credentials */
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;

  /* Assign the RTDB URL (required) */
  config.database_url = DATABASE_URL;

  /* Assign the callback function for the long running token generation task */
  config.token_status_callback = tokenStatusCallback; //see addons/TokenHelper.h

  Firebase.begin(&config, &auth);

  //Or use legacy authenticate method
  //Firebase.begin(DATABASE_URL, "<database secret>");
}

void loop()
{
  if (Firebase.ready() && (millis() - sendDataPrevMillis > 15000 || sendDataPrevMillis == 0))
  {
    sendDataPrevMillis = millis();

    Serial.printf("Set int... %s\n", Firebase.setInt(fbdo, "/test/int", count) ? "ok" : fbdo.errorReason().c_str());

    Serial.printf("Get int... %s\n", Firebase.getInt(fbdo, "/test/int") ? String(fbdo.intData()).c_str() : fbdo.errorReason().c_str());

    FirebaseJson json;
    json.add("value", count);

    Serial.printf("Push json... %s\n", Firebase.pushJSON(fbdo, "/test/push", json) ? "ok" : fbdo.errorReason().c_str());

    json.set("value", count + 100);
    Serial.printf("Update json... %s\n\n", Firebase.updateNode(fbdo, String("/test/push/" + fbdo.pushName()), json) ? "ok" : fbdo.errorReason().c_str());

    count++;
  }
}

```

Serial Output - Error;
```
Connected with IP: 192.168.1.115

Firebase Client v3.2.2

Token info: type = id token, status = on request
Token info: type = id token, status = ready
Set int... Permission denied
Get int... Permission denied
Push json... Permission denied
Update json... Permission denied

Set int... Permission denied
Get int... Permission denied
Push json... Permission denied
Update json... Permission denied

Set int... Permission denied
Get int... Permission denied
Push json... Permission denied
Update json... Permission denied
```
It repeats permission denied part forever. I only changed definings in the code.
My firebase rules;
```
{
  "rules": {
    ".read": "now < 1893445200",
    ".write": "now < 1893445200",
  }
}
```

I can't think anything causes that. If you can help I would aprreciate it.

mobizt, 2021-06-01:
Timestamp in the rules should be in milliseconds, not seconds e.g. 1893445200000 instead of 1893445200.

At Firebase console, on the Rules setup, you should click at Rules Playground button to test the rules.

closed 2021-06-01 by mobizt

parulll123, 2024-11-23:
Thanks for your help sir