mobizt/Firebase-ESP8266 #206: Firebase keep disappearing every a few days

floatAsNeeded, 2021-06-17:
I'm experiencing a really weird issue that I never experienced before with one of the people testing my project. Basically real time database suddenly becomes completely blank and all data disappear and all the devices using the library can't anymore access and write and read data. It's very weird because it really never happened before to me. Anyone knows what it is?

mobizt, 2021-06-18:
Update your security rules for secure.
With recently new terms of services, Google will set the project to inactive if it has insecure rules and never update within some period.

Contact Google if it is a issue.

closed 2021-06-18 by mobizt

floatAsNeeded, 2021-06-18:
Thanks for the reply.
It seems also to be something related to Firewall https://stackoverflow.com/questions/50811695/which-ports-have-to-be-open-to-communicate-with-firebase-database-not-fcm-messa

Because I access his data from my side. Read and write, but it can't see even the database using is network. I asked if it could see it using other devices, but still waiting the answer. 

These are his rules
 
[image]

mobizt, 2021-06-18:
You should ask in community forum.

The router router is not the issue here too as you post about blank data which is not related to the library.

floatAsNeeded, 2021-06-18:
Thank you.

I was just thinking if you could have some ideas that's why I posted here. For sure not a problem with the library, because in my side there are no problems at all with the library or Google. Must be something related to his network

mobizt, 2021-06-18:
You should use keywords in your question like  "can't access", "connection failed" or something that shows the communication error as this will be showed from calling **fbdo.errorReason()**.

floatAsNeeded, 2021-06-18:
Yes. I think that line of the code could explain why it happened putting that in a Serial print. Unfortunately, I didn't enable OTA, so I can't actually modify the code from here. I didn't implement that line in the code. But I believe it must be something related to some network issue

floatAsNeeded, 2021-06-18:
Eventually we found out that it is something related to the Firewall. Because it can see the database using his mobile network on the phone but on his LAN with Wi-Fi connection. Not just it can't see the database but also the devices using this network they don't work properly. It worked fine for 5 days, so I'm wondering why then happen suddenly this issue with the Firewall if firewall was already set. I was wondering, what's the security level on reading and writing with this library? How much it is safe in running the ESP before the firewall?

mobizt, 2021-06-18:
If you are new to Firebase or Google Cloud Platform, please read the its related document how to use the services included API endpoints and authentication.

This library uses SSL client that connected to SSL port 443 on Google server.   The Google root CA can be assigned to the library for handshake of trust server. The data was SSL encrypted before transmit as other SSL client devices using mbedTLS library.

floatAsNeeded, 2021-06-18:
Thank you very much for the infos. I should have a look the documentation as well. How you assign the CA to the library?

mobizt, 2021-06-18:
https://github.com/mobizt/Firebase-ESP8266/tree/master/examples/BasicCert