mobizt/Firebase-ESP8266 #209: Exception

cezar-pereira, 2021-06-19:
Hi, i have a problem.

> User exception (panic/abort/assert)
    --------------- CUT HERE FOR EXCEPTION DECODER ---------------
>    Unhandled C++ exception: OOM
>stack
>ctx: cont
>sp: 3ffff870 end: 3fffffc0 offset: 0000
>...

>Unhandled C++ exception: OOM
>stack
>ctx: cont
>sp: 3ffff870 end: 3fffffc0 offset: 0000
>...

My code resumed:

```
FirebaseData firebaseDataStream;
FirebaseData firebaseDataSet;
FirebaseData firebaseDataGet;
```

Setup:

```
Firebase.begin(credentials.FIREBASE_HOST, credentials.FIREBASE_AUTH);
if (!Firebase.beginStream(firebaseDataStream, paths.PATH)){
        Serial.println("REASON: " + firebaseDataStream.errorReason());
 }

Firebase.setBool(firebaseDataSet, paths.CHECK_TEMPERATURE, false);
Firebase.getInt(firebaseDataGet, paths.SETTINGS_INTERVAL_TO_UPDATE);
```
```
void loop(){
    if (!Firebase.readStream(firebaseDataStream)){
        Serial.println("REASON: " + firebaseDataStream.errorReason());
        return;
    }

    if (firebaseDataStream.streamTimeout()){
        Serial.println("Stream timeout, resume streaming...");
        Serial.println();
    }

    if (firebaseDataStream.streamAvailable()){
        Serial.println("STREAM PATH: " + firebaseDataStream.streamPath());
        Serial.println("EVENT PATH: " + firebaseDataStream.dataPath());
        Serial.println("DATA TYPE: " + firebaseDataStream.dataType());
        Serial.println("EVENT TYPE: " + firebaseDataStream.eventType());
    }
}
```

to use **Firebase.getInt(firebaseDataGet, paths.SETTINGS_INTERVAL_TO_UPDATE);** the error is generated.

My code worked on an old version (I don't know which one), after downloading the newer version this problem is happening.

mobizt, 2021-06-20:
Check your free heap because the error involved the out of memory situation.

Reduce the Firebase Data object usage because it use large memory for data handling.

The new version of this library will reduce memory usage (approx. 5-7k lesser) which allows more space to run for your code.

closed 2021-06-20 by mobizt

mobizt, 2021-06-22:
The [latest library v3.3.0](https://github.com/mobizt/Firebase-ESP8266/releases/tag/v3.3.0) is available which improves the memory usage and 8-10 k less memory consumption as compare to the previous version. 

Please update the library.

cezar-pereira, 2021-06-24:
in version 3.3.1 ocurred the follow error:

------------------------------------
Can't begin stream connection...
REASON: not connected
------------------------------------

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: cont
sp: 3ffffa70 end: 3fffffc0 offset: 0000

Version 3.2.0 my code work.

Another question, I need send email with my arduino, but function does not work, in versio before 3.2.0 it work, i use: 
firebaseDataStream.pauseFirebase(true);
firebaseData.pauseFirebase(true);

send email....

firebaseDataStream.pauseFirebase(false);
firebaseData.pauseFirebase(false);

but, repeating, inversion before 3.2.0 it work 😢

mobizt, 2021-06-24:
I don't see your full code which you do something that causes out of memory.
It's not related to the library version.

mobizt, 2021-06-24:
Check this.
https://github.com/mobizt/Firebase-ESP8266#ide-configuaration-for-esp8266-mmu---adjust-the-ratio-of-icache-to-iram