mobizt/Firebase-ESP8266 #210: any thoughts why it happens

lexxwork, 2021-06-21:
First, thank you so much, @mobizt, for such awesome, powerful and reliable library.
I would like to know why my code is getting in **'Soft WDT reset'** when i compile it in debug mode. While in release mode it's working.

platform: Espressif 8266 3.0.0

only one library **Firebase-ESP8266@3.2.2**

<details>
<summary><b>platformio.ini:</b></summary>

```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags = -D PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED
monitor_filters = 
	default
	log2file
	esp8266_exception_decoder
upload_port = /dev/ttyUSB0
upload_speed = 115200
monitor_speed = 115200
build_type = debug
```
</details>

<details>
<summary><b>logs with exceptions stack:</b></summary>

```
IRAM free:  17320 bytes
DRAM free:  21352 bytes
register device test-device-id_5

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3ffff3e0 end: 3fffffb0 offset: 01a0
3ffff580:  0032e84c 00000000 3fff2c44 40220d36  
3ffff590:  00000000 401081bc 3fff2c44 40220d36  
3ffff5a0:  000579e6 00000000 3fff2c44 4022ce6d  
3ffff5b0:  00003a98 00002019 4ba8a200 00000109  
3ffff5c0:  00000001 00000008 3fff952c 4024f706  
3ffff5d0:  3fff9a2c 3fff791c 3fff952c 3fff952c  
3ffff5e0:  401081bc 00000001 3fff2c44 4022cfb8  
3ffff5f0:  401081bc 00000001 3fff2c44 4022d375  
3ffff600:  00000000 00000020 000000b8 4024817e  
3ffff610:  3fff1d28 00000000 00000000 40248cf3  
3ffff620:  00000000 40282090 00000000 00000000  
3ffff630:  00000000 00000000 00000000 00000000  
3ffff640:  40105fca 3ffee778 0026ca81 00000000  
3ffff650:  401061a1 0026ca81 3fff1a40 00000000  
3ffff660:  00000000 00000000 00000000 00000000  
3ffff670:  00000000 3ffff680 00000000 00000000  
3ffff680:  4022141d 00000001 00000000 40221412  
3ffff690:  3fff791c 3fff2c44 3fff7aec 40205896  
3ffff6a0:  3ffff6d0 dfe26b22 dfe26b22 3ffeec40  
3ffff6b0:  000001bb 3fff2c44 3fff791c 3ffeec40  
3ffff6c0:  000001bb 3fff2c44 3fff791c 4022d504  
3ffff6d0:  402359dc dfe26b22 402359dc dfe26b22  
3ffff6e0:  3fff8d2c 00000000 3ffeec40 4023406a  
3ffff6f0:  3fff234c 40204cdc 00000000 4021dfd3  
3ffff700:  3fff7999 3fff7964 00000035 4021e013  
3ffff710:  3ffef954 0000002f 3ffef9a8 00000000  
3ffff720:  3fff5ba4 3ffffa50 3ffeec40 40211953  
[... 223 lines cut ...]
  Firebase.setJSONAsync(fbdo, path, json);
}
```
**waitFbReady() in advance**
```cpp
void Device::waitFbReady()
{
  size_t tries = 10;
  while (tries-- > 0 && !Firebase.ready())
  {
    yield();
    unsigned long lastTime = millis();
    while (millis() - lastTime < 1000)
      yield();
  }
}
```
</details>

From first logs lines  you may see that massage
```register device test-device-id_5```
That is right before:
```Firebase.setJSONAsync(fbdo, path, json);```

I want to fix it somehow because it's hard to detect bugs in release mode.

mobizt, 2021-06-22:
Your device is running out of memory.

Please check your device free heap.

The [latest library v3.3.00](https://github.com/mobizt/Firebase-ESP8266/releases/tag/v3.3.0) is available which improves the memory usage and 8-10 k less memory consumption as compare to the previous version. 

Please update the library.

closed 2021-06-22 by mobizt

lexxwork, 2021-06-22:
confirm, now it's ok!
bravo 👏🏽!