mobizt/Firebase-ESP8266 #211: HELP

Pranaym09, 2021-06-23:
Hello I am new here and got stuck the exception is saying this

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: cont
sp: 3ffffa50 end: 3fffffc0 offset: 0000
3ffffa50:  3fff35d0 00000000 00000000 40239d61  
3ffffa60:  000000fe 00000000 00000000 00000000  
3ffffa70:  00000000 00000000 00000000 00000001  
3ffffa80:  00007fff 00000000 00000000 3fff7f84  
3ffffa90:  00000000 4021970f 00000d60 4021bd12  
3ffffaa0:  00000000 4021970f 00000020 4021bd31  
3ffffab0:  00000000 00000000 00000d60 4021b63c  
3ffffac0:  3ffee630 3fff7f84 3fff6f2c 4021970f  
3ffffad0:  0000002a 3fff3260 00000001 4021b78c  
3ffffae0:  00000000 00000001 000001bb 4021be36  
3ffffaf0:  00000000 3fff7fb4 3fff6f2c 40218852  
3ffffb00:  3fff7dfc 00000028 00000028 3fffaadc  
3ffffb10:  000001bb 3fff7f84 3fff6f2c 3fffaadc  
3ffffb20:  000001bb 3fff7f84 3fff6f2c 40219955  
3ffffb30:  40226e74 5561c923 40226e74 5561c923  
3ffffb40:  3ffffca8 00000000 3ffeeae8 402256ee  
3ffffb50:  3fff3bc4 40217d0c 3ffeeae8 402121ff  
3ffffb60:  3ffef8e8 3ffeeae8 00000000 40212246  
3ffffb70:  000001bb 3fff76e4 3ffeeae8 3ffeeae8  
3ffffb80:  3ffffd10 3ffffc88 00000000 40209df8  
3ffffb90:  00000000 3fff85ec 00000000 00000000  
3ffffba0:  00000000 00000000 00000000 01000001  
3ffffbb0:  00000000 3ffffbbc 00000000 3fff3800  
3ffffbc0:  3fff35d0 3fff3608 3fff38c8 3fff862c  
3ffffbd0:  00000020 00000020 3ffe8648 4010092f  
3ffffbe0:  3fff85ec 3ffffbec 00000005 736f6c63  
3ffffbf0:  3fff0065 00000020 3fffad34 3ffffc04  
3ffffc00:  00000001 3fff002f 3fff35d0 40237a90  
3ffffc10:  00000000 3ffffc1c 00000003 3fff2b18  
3ffffc20:  3fff773c 40230d2f 00000020 40100b38  
3ffffc30:  00000003 00223022 00000029 40230db0  
3ffffc40:  00000014 3fff7714 3ffffce0 40230e08  
3ffffc50:  000008ac 000008ac 3ffe8648 4010092f  
3ffffc60:  00000028 00000000 3ffecba8 3fff2b18  
3ffffc70:  3ffeeae8 00000020 3fff7fec 40100b0e  
3ffffc80:  00000002 3fff3088 3fffaadc 000000ea  
3ffffc90:  000000f0 0033532f 00000002 40230d4c  
3ffffca0:  3ffffca8 00000000 3ffeea00 4020dc85  
3ffffcb0:  4027018a 3ffeeae8 00000000 402310e4  
3ffffcc0:  00000000 00000001 3fff2b18 40230db0  
3ffffcd0:  3ffffd78 40217d0c 3fff8694 40230e72  
3ffffce0:  00000000 00000028 00000028 3fff2b18  
3ffffcf0:  3ffffeb0 3ffe8a17 3ffffd70 3fff2b18  
3ffffd00:  3ffffeb0 3ffef8e0 3ffeeae8 4020a2a6  
3ffffd10:  3ffffd18 00000003 0033532f 40100b0e  
3ffffd20:  3fff35d0 2c9f0300 00000005 00000006  
3ffffd30:  40102f00 00000000 00000000 00000000  
3ffffd40:  00000000 00000000 00000000 00000000  
3ffffd50:  00000000 00000000 3ffffd60 00000000 <
3ffffd60:  00004b00 00000000 00000000 fffffffe  
3ffffd70:  3ffffd78 00000000 00000000 3fff0af8  
3ffffd80:  3fff2b18 00000000 3ffffd90 00000000 <
3ffffd90:  00000100 3ffe8a17 3ffffe10 00001388  
3ffffda0:  3fff0000 3ffffdac 00000000 40104500  
3ffffdb0:  3ffed628 402349ca 3ffffdc0 00000000 <
3ffffdc0:  00000000 4010450e 3ffed628 402349ca  
3ffffdd0:  3fff0af8 3fff6214 0000002a 3fff2b18  
3ffffde0:  00000000 4bc6a7f0 53b645a1 004eef00  
3ffffdf0:  00000000 00000000 4bc6a7f0 40230db0  
3ffffe00:  00000000 3ffef6e8 401003c5 40230e08  
3ffffe10:  00004444 3ffef7e8 401003c5 4023157c  
3ffffe20:  00000003 3fff3088 00000000 3ffef8e0  
3ffffe30:  00000000 3ffef7e8 3ffeeae8 4020a2f0  
3ffffe40:  00000000 3ffef7e8 3ffeeae8 4020c8eb  
3ffffe50:  00000000 3fff8390 3fff838c 40100200  
3ffffe60:  00000000 3fff2b18 00000000 3fff838c  
3ffffe70:  3fff3178 3fff7f3c 3fff839c 402309f0  
3ffffe80:  00000813 00000813 3ffe8648 40230db0  
3ffffe90:  00000000 4bc6a7f0 26e978d4 004eeecc  
3ffffea0:  00000000 00000000 4bc6a7f0 00000000  
3ffffeb0:  3ffffeb8 00000003 0033532f 00000000  
3ffffec0:  00004444 3fff3088 00000000 40211e00  
3ffffed0:  00000001 3ffef8e8 3fff2b18 00000000  
3ffffee0:  3fff18f8 3fff0af8 3fff17f8 3fff2b18  
3ffffef0:  00000002 00000008 3fff3088 4020c944  
3fffff00:  00000001 00000004 3fff3088 4020c9a6  
3fffff10:  00000001 00000000 3fff2b18 3ffe88b7  
3fffff20:  00000000 00000002 3ffe8a9c 4021d4ac  
3fffff30:  3ffe8a9d 3ffeeae8 3fff2b18 00000001  
3fffff40:  3fff3184 3fff3180 00000000 40225490  
3fffff50:  4021a2ec 3ffe8a9b 3fff313c 4021a96d  
3fffff60:  007a1200 65c42021 3fff3100 4021a554  
3fffff70:  3fff838c 00000000 3fff313c 4021a638  
3fffff80:  3ffe88b4 3ffe88b1 feefeffe 3fff31fc  
3fffff90:  3fffdad0 00000000 3fff31e8 4021b875  
3fffffa0:  3fffdad0 00000000 3fff31e8 4021b8af  
3fffffb0:  feefeffe feefeffe 3ffe8644 40100dc9  
<<<stack<<<

last failed alloc call: 4021970F(3424)

mobizt, 2021-06-23:
Post your code here.

mobizt, 2021-06-23:
Out of memory, check your free heap.

The memory usage in latest version library was improved and try to update first.

closed 2021-06-23 by mobizt

Pranaym09, 2021-06-23:
`#include <FirebaseESP8266.h>
#include <FirebaseFS.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <ESP8266HTTPClient.h>
#define FIREBASE_HOST "dukh2-49941-default-rtdb.firebaseio.com"
#define FIREBASE_AUTH "jlgmWCQ0AYga3Dh3k9M0rVg5b1mPpNBzNigfJgoc"
 
  #define Relay1 12
  int val1;
  #define Relay2 15
  int val2;
  #define Relay3 13
  int val3;
  #define Relay4 14
  int val4;
  FirebaseData fbdo;
   FirebaseData kbdo;
    FirebaseData stream;
    FirebaseData pbdo;

  String WIFI_SSID = "oppo";
  String WIFI_PASSWORD = "pranaym";

    void setup()
      {
          Serial.begin(115200);
          pinMode(Relay1, OUTPUT);
          pinMode(Relay2, OUTPUT);
          pinMode(Relay3, OUTPUT);

  
          digitalWrite(Relay1, HIGH);
          digitalWrite(Relay2, HIGH);
          digitalWrite(Relay3, HIGH);

          WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
          Serial.print("connecting");
          delay(10000);
          while (WiFi.status() != WL_CONNECTED) 
            {
                Serial.print(".");
                delay(500);
             }
           Serial.println();
           Serial.print("connected:");
           Serial.println(WiFi.localIP());
   
               Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
               Firebase.setString(fbdo,"S1", "0");                   
               Firebase.setString(fbdo,"S2", "0");
               Firebase.setString(fbdo,"S3", "0");

            Firebase.setStreamCallback(fbdo, streamCallback);
            Firebase.beginStream(fbdo, "S1");
             Firebase.setStreamCallback(kbdo, streamCallback1);
             Firebase.beginStream(kbdo, "S2");
             Firebase.setStreamCallback(pbdo, streamCallback2);
             Firebase.beginStream(pbdo, "S3");
        }

        
    void firebasereconnect()
      {
         Serial.println("Trying to reconnect");
         Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
      }

void streamCallback(StreamData data)
{
 val1= data.stringData().toInt();
}

void streamCallback1(StreamData data1)
{
 val2= data1.stringData().toInt();
}
void streamCallback2(StreamData data2)
{
 val3= data2.stringData().toInt();
}

    void loop()
      {
          
          if (val1 == 1)
           {
               digitalWrite(Relay1, LOW);
               Serial.println("light ON");
           }
          else if (val1 == 0)
           {
              digitalWrite(Relay1, HIGH);
              Serial.println("light OFF");
           }

          if (val2 == 1)
            {
              digitalWrite(Relay2, LOW);
              Serial.println("Fan ON");
            }
          else if (val2 == 0)
            {
              digitalWrite(Relay2, HIGH);
              Serial.println("Fan OFF");
            }

            if (val3 == 1)
            {
              digitalWrite(Relay3, LOW);
              Serial.println("TV ON");
           }
          else if (val3 == 0)
           {
              digitalWrite(Relay3, HIGH);
              Serial.println("TV OFF");
            }

      }`

Pranaym09, 2021-06-23:
`
#include <FirebaseESP8266.h>
#include <FirebaseFS.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <ESP8266HTTPClient.h>
#define FIREBASE_HOST "dukh2-49941-default-rtdb.firebaseio.com"
#define FIREBASE_AUTH "jlgmWCQ0AYga3Dh3k9M0rVg5b1mPpNBzNigfJgoc"
 
  #define Relay1 12
  int val1;
  #define Relay2 15
  int val2;
  #define Relay3 13
  int val3;
  #define Relay4 14
  int val4;
  FirebaseData fbdo;
   FirebaseData kbdo;
    FirebaseData stream;
    FirebaseData pbdo;

  String WIFI_SSID = "oppo";
  String WIFI_PASSWORD = "pranaym";

    void setup()
      {
          Serial.begin(115200);
          pinMode(Relay1, OUTPUT);
          pinMode(Relay2, OUTPUT);
          pinMode(Relay3, OUTPUT);

  
          digitalWrite(Relay1, HIGH);
          digitalWrite(Relay2, HIGH);
          digitalWrite(Relay3, HIGH);

          WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
          Serial.print("connecting");
          delay(10000);
          while (WiFi.status() != WL_CONNECTED) 
            {
                Serial.print(".");
                delay(500);
             }
           Serial.println();
           Serial.print("connected:");
           Serial.println(WiFi.localIP());
   
               Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
               Firebase.setString(fbdo,"S1", "0");                   
               Firebase.setString(fbdo,"S2", "0");
               Firebase.setString(fbdo,"S3", "0");

            Firebase.setStreamCallback(fbdo, streamCallback);
            Firebase.beginStream(fbdo, "S1");
             Firebase.setStreamCallback(kbdo, streamCallback1);
             Firebase.beginStream(kbdo, "S2");
             Firebase.setStreamCallback(pbdo, streamCallback2);
             Firebase.beginStream(pbdo, "S3");
        }

        
    void firebasereconnect()
      {
         Serial.println("Trying to reconnect");
         Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
      }

void streamCallback(StreamData data)
{
 val1= data.stringData().toInt();
}

void streamCallback1(StreamData data1)
{
 val2= data1.stringData().toInt();
}
void streamCallback2(StreamData data2)
{
 val3= data2.stringData().toInt();
}

    void loop()
      {
          
          if (val1 == 1)
           {
               digitalWrite(Relay1, LOW);
               Serial.println("light ON");
           }
          else if (val1 == 0)
           {
              digitalWrite(Relay1, HIGH);
              Serial.println("light OFF");
           }

          if (val2 == 1)
            {
              digitalWrite(Relay2, LOW);
              Serial.println("Fan ON");
            }
          else if (val2 == 0)
            {
              digitalWrite(Relay2, HIGH);
              Serial.println("Fan OFF");
            }

            if (val3 == 1)
            {
              digitalWrite(Relay3, LOW);
              Serial.println("TV ON");
           }
          else if (val3 == 0)
           {
              digitalWrite(Relay3, HIGH);
              Serial.println("TV OFF");
            }

      }
`

mobizt, 2021-06-23:
Normally you can't use too many Firebase Data objects more than 2 because your device memory is not enough to use like this but in [v3.3.0](https://github.com/mobizt/Firebase-ESP8266/releases/tag/v3.3.0) which reduced the memory usage and you can try with 4 Firebase Data objects but it's  your own risk  and I'm **NOT RECOMMENDED**.

mobizt, 2021-06-23:
In v3.3.0, you can use 4 Firebase Data objects with 11k free memory which relatively low and not recommend, use 3 Firebase Data objects is enough.

mobizt, 2021-06-23:
I recommend to use MultipathStream instead for simple data with multiple path which will save your memory.
https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/DataChangesListener/MultiPath/MultiPath.ino

Pranaym09, 2021-06-23:
> I recommend to use MultipathStream instead for simple data with multiple path which will save your memory.
> https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/DataChangesListener/MultiPath/MultiPath.ino

I tried this example as it is but its giving the same exeception.

mobizt, 2021-06-23:
> > I recommend to use MultipathStream instead for simple data with multiple path which will save your memory.
> > https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/DataChangesListener/MultiPath/MultiPath.ino
> 
> I tried this example as it is but its giving the same exeception.

It's strange and impossible. 

Using to many Firebase Data objects is nonsense because it not a normal variable but it keeps open the session and SSL resource which consume large memory inside. 

Look like you're not yet update the library and never try what I said actually.