mobizt/Firebase-ESP8266 #212: Nested FirebaseJsonArray issue

lexxwork, 2021-06-24:
## Nested FirebaseJsonArray issue

**IDE and its version:**

- Arduino, PlatformIO
- Version Core 5.1.1 Home 3.3.4

**ESP8266 Arduino Core SDK version**

- Version 3.0.0

**library Firebase-ESP8266 v3.3.0**

**platformio.ini**

```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags = -D PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED
monitor_filters =
	default
	log2file
	esp8266_exception_decoder
upload_port = /dev/ttyUSB0
upload_speed = 115200
monitor_speed = 115200
build_type = debug; no matter
lib_deps = https://github.com/mobizt/Firebase-ESP8266.git#v3.3.0
```

### target problem code and logs are marked with '_<--this place_'

**expecting incoming FirebaseJson**

```jsonc
{
  "experimentId": "q34p0u8wer",
  //!!!nested array
  "scheduler": [
    [3600, 12900],
    [4500, 34200]
  ],
  "intervals": [10, 20, 5]
}
```

**full example code to reproduce**

```cpp
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

struct IntervalsSettings
{
  int initRate;
  int hours;
  int afterRate;
};

struct ScheduleSettings
{
  int start;
  int end;
};

struct Settings
{
  String experimentId = "";
  IntervalsSettings intervals;
  std::vector<ScheduleSettings> scheduleList = std::vector<ScheduleSettings>();
};

Settings settings;

void loadSettings(FirebaseJson &json)
{
  FirebaseJsonArray jsonArr;
  FirebaseJsonData jsData;

  if (json.get(jsData, "experimentId"))
  {
    settings.experimentId = jsData.stringValue;
    Serial.println(F("settings.experimentId: ") + settings.experimentId);
  }
  if (json.get(jsData, "intervals"))
  {
    if (jsData.getArray(jsonArr))
    {
      Serial.println(F("settings.intervals: "));
      IntervalsSettings &intervals = settings.intervals;
      for (size_t i = 0; i < jsonArr.size(); i++)
      {
        jsonArr.get(jsData, i);
        switch (i)
        {
        case 0:
          intervals.initRate = jsData.intValue;
          break;
        case 1:
          intervals.hours = jsData.intValue;
          break;
        case 2:
          intervals.afterRate = jsData.intValue;
          break;
        }
        yield();
      }
      Serial.printf("initRate: %d,  hours: %d, afterRate: %d\r\n",
                    intervals.initRate, intervals.hours, intervals.afterRate);
    }
    jsonArr.clear();
  }
  if (json.get(jsData, "scheduler"))
  {
    if (jsData.getArray(jsonArr))
    {
      Serial.println(F("settings.scheduler: "));
      for (size_t i = 0; i < jsonArr.size(); i++)
      {
        ScheduleSettings schedule = {start : 0, end : 0};
        FirebaseJsonData jsData2;
        FirebaseJsonArray jsonArr2;
        jsonArr.get(jsData2, i);
        Serial.printf("jsData2.success: %d\r\n", jsData2.success);
        jsData2.getArray(jsonArr2);
        String buf;
        jsonArr2.toString(buf);
        Serial.println("jsonArr2: " + buf); //<--this place
        jsonArr2.get(jsData2, 0);
        schedule.start = jsData2.intValue;
        jsonArr2.get(jsData2, 1);
        schedule.end = jsData2.intValue;
        Serial.printf("start: %d,  end: %d\r\n", schedule.start, schedule.end);
        settings.scheduleList.push_back(schedule);
        // jsonArr.clear();
        yield();
      }
    }
  }
}

void setup()
{
  Serial.begin(115200);

  // {
  //    "experimentId": "q34p0u8wer",
  //	  "scheduler" : [[3600, 12900 ], [4500, 34200]],
  //	  "intervals" : [10, 20, 5]
  // }
  FirebaseJson json;
  FirebaseJsonArray jsonarr;
  FirebaseJsonArray jsonarrNested;
  json.set("experimentId", "q34p0u8wer");
  jsonarr.add(10);
  jsonarr.add(20);
  jsonarr.add(25);
  json.add("intervals", jsonarr);
  jsonarr.clear();
  jsonarr.add(3600);
  jsonarr.add(12900);
  jsonarrNested.add(jsonarr);
  jsonarr.clear();
  jsonarr.add(4500);
  jsonarr.add(34200);
  jsonarrNested.add(jsonarr);
  json.add("scheduler", jsonarrNested);

  jsonarr.clear();
  jsonarrNested.clear();

  String buf;
  json.toString(buf);
  Serial.println("json: " + buf);

  loadSettings(json);
}

void loop()
{
}
```

**Serial logs from full example with Firebase-ESP8266 v3.3.0**

```
json: {"experimentId":"q34p0u8wer","intervals":[10,20,25],"scheduler":[[3600,12900],[4500,34200]]}
settings.experimentId: q34p0u8wer
settings.intervals:
initRate: 10,  hours: 20, afterRate: 25
settings.scheduler:
jsData2.success: 1 <--this place

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3460, room 16
....
```

**Serial logs from application with Firebase-ESP8266 v3.3.0**

```
register device test-device-id_5
IRAM free:  14280 bytes
DRAM free:  27872 bytes
command Stream:
settings.experimentId: -McpCoM0FDpe0CZr31jO
settings.intervals:
initRate: 60,  hours: 1, afterRate: 10
settings.scheduler:
jsData2.success: 1
jsonArr2: "root" <--this place
start: 0,  end: 0 <--not good
Schedule items:
start: 1624395600, end: 1624482000
```

**Serial logs from applictation with Firebase-ESP8266 v3.2.2**

```
register device test-device-id_5
IRAM free:  14704 bytes
DRAM free:  17024 bytes
command Stream:
settings.experimentId: -Mcto4TZXnfW74At6alb
settings.intervals:
initRate: 60,  hours: 1, afterRate: 10
settings.scheduler:
jsData2.success: 1
jsonArr2: [40260,82800] <--this place
start: 40260,  end: 82800 <--good
Schedule items:
start: 1624435860, end: 1624478400
```

---

## Extra question: is it safe to reuse the

```cpp
FirebaseJsonArray jsonArr;
FirebaseJsonData jsData;
```

instead

```cpp
FirebaseJsonData jsData2;
FirebaseJsonArray jsonArr2;
```

as I did it before with Firebase-ESP8266 v3.2.2 and it worked.

mobizt, 2021-06-24:
The issue was fixed in [v3.3.1](https://github.com/mobizt/Firebase-ESP8266/releases/tag/v3.3.1), please update.

For your last question, the jsonArr is still in used in the loop, it's nonsense to declare the new FirebaseJsonArray with the same name in that loop.

closed 2021-06-24 by mobizt

lexxwork, 2021-06-24:
i did not mean to declare with the same name, meant to use it loop nested loops, look:
i have removed
FirebaseJsonData jsData2;
FirebaseJsonArray jsonArr2;
```cpp
 if (json.get(jsData, "scheduler"))
  {
    if (jsData.getArray(jsonArr))
    {
      Serial.println(F("settings.scheduler: "));
      for (size_t i = 0; i < jsonArr.size(); i++)
      {
        ScheduleSettings schedule = {start : 0, end : 0};
        jsonArr.get(jsData, i);
        Serial.printf("jsData.success: %d\r\n", jsData.success);
        jsData.getArray(jsonArr);
        String buf;
        jsonArr.toString(buf);
        Serial.println("jsonArr: " + buf); //<--this place
        jsonArr.get(jsData, 0);
        schedule.start = jsData.intValue;
        jsonArr.get(jsData, 1);
        schedule.end = jsData.intValue;
        Serial.printf("start: %d,  end: %d\r\n", schedule.start, schedule.end);
        settings.scheduleList.push_back(schedule);
        jsonArr.clear(); //uncomented
        yield();
      }
    }
  }
```

mobizt, 2021-06-24:
Check your strange usage
```cpp
   for (size_t i = 0; i < jsonArr.size(); i++) //<---- this will change in the subsequence round due to you change its data in loop with jsData.getArray(jsonArr)
      {
        ScheduleSettings schedule = {start : 0, end : 0};
        jsonArr.get(jsData, i); //<- ok
        Serial.printf("jsData.success: %d\r\n", jsData.success);
        jsData.getArray(jsonArr); //<---- strange usage as it is already in used for iteration parsing in the loop
```

lexxwork, 2021-06-24:
ok, so it's not safe.
thank you for the fast answer.