mobizt/Firebase-ESP8266 #220: WDT RESET

Bingsitfx, 2021-07-02:
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

How to fix this?
Im using library version 3.3.5
Nodemcu V3 ESP8266 12E with board version 2.7.4
My code : 
https://pastebin.com/ZYHUrUUc

mobizt, 2021-07-02:
Don't post your code on that site, you should post the code and error in the Github post with proper mark down and also indent the code for easier reading.

The error is about bug in your code, power weakness, long running code (blocking code)

1. There is possible array index bug in your code.

```cpp
String arrData[22];
...

//Parsing data
int index = 0;
for(int i=0; i<= data.length(); i++)
{
    char delimiter = '#' ;
    if(data[i] != delimiter)
      arrData[index] += data[i] ; //<----------- dangerous use of array without index checking
     else
       index++; //variabel index bertambah 1
}
```

2. Your board power supply is weak. Change the usb cable for shorter one.

3. Use delay(0); some where in your long running code.

mobizt, 2021-07-02:
You **need to** debug your code first before asking the help.

closed 2021-07-02 by mobizt

Bingsitfx, 2021-07-03:
ok i will try to add delay(0); on my long code & trying to debug my code. If the problem still exist, i'ill let you know