mobizt/Firebase-ESP8266 #221: SSL Certificate broke Firebase with ESP8266?

kovikti, 2021-07-03:
Hi,

I've been using Firebase with ESP8266 for over half a year now. It has been working quite well (except when the https fingerprint had to be updated).
Since last night I can't get it working anymore. 
I've been using the FirebaseArduino library (with ESP Arduino lib 2.x.x, axTLS), now I switched to Firebase-ESP8266 (and Arduino 3.x.x, bearSSL) but the problem persists.
It seems like the certificate fingerprint changed again (RSA 4096 bits). I got a verification failure with the one I have been using. (With FirebaseArduino) I had similar problems way back, about 4-5 years ago that some https sites had large certificates the esp could not handle. I hoped this has changed since, especially with newer libraries.
"03:9E:4F:E6:83:FC:40:EF:FC:B2:C5:EF:36:0E:7C:3C:42:20:1B:8F"; // 2021-01
"50:89:50:57:90:1F:37:E3:B8:F3:5B:02:ED:3A:65:6E:6A:34:DB:93";    // 2021-07 changed now???
However when I change to the new one, I get exception. I suspect the certificate is too large and the ESP8266 cannot handle it.
When I moved to Firebase-ESP8266 I followed the guide to set up the project.
Any help is appreciated!

Thanks for your time. Let me know if this is a widespread issue or just me.

Here is a stack trace:

```
Exception Cause: Not found

0x40214a36: raise_exception at core_esp8266_postmortem.cpp:?
0x40214a48: __unhandled_exception at ??:?
0x40275058: ethernet_input at ??:?
0x40100c02: get_unpoisoned_check_neighbors at umm_malloc.cpp:?
0x401010bc: umm_poison_free_fl at ??:?
0x401010b9: umm_poison_free_fl at ??:?
0x4027505b: ethernet_input at ??:?
0x40100610: free at ??:?
0x40275058: ethernet_input at ??:?
0x40232658: operator delete(void*) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_op.cc:50
0x402319a0: operator delete[](void*) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_opv.cc:36
0x4020eb0e: _ZN10UtilsClass4delSEPc$constprop$0 at Signer.cpp:?
0x40275058: ethernet_input at ??:?
0x4020fe62: _ZN15Firebase_Signer19handleTokenResponseEv$part$0 at Signer.cpp:?
0x401004fd: millis at ??:?
0x40214470: esp_yield at ??:?
0x40214b00: micros_overflow_tick at ??:?
0x40100b65: check_poison_neighbors$part$0 at umm_malloc.cpp:?
0x40100a00: umm_get_ptr_context at umm_malloc.cpp:?
0x40100c71: umm_free_core at umm_malloc.cpp:?
0x4010109e: umm_free at ??:?
0x401010bc: umm_poison_free_fl at ??:?
0x402319a0: operator delete[](void*) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_opv.cc:36
0x4020f044: UtilsClass::reconnect(unsigned long) at ??:?
0x40203e30: ESP8266WiFiSTAClass::status() at ??:?
0x4021016b: Firebase_Signer::handleTokenResponse() at ??:?
0x40210b70: Firebase_Signer::getIdToken(bool, char const*, char const*) at ??:?
0x40212eb4: Print::write(char const*) at ??:?
0x401004fd: millis at ??:?
0x40211d96: _ZN15Firebase_Signer19tokenProcessingTaskEv$part$0 at Signer.cpp:?
0x401004fd: millis at ??:?
0x40231e12: std::char_traits<char>::assign(char&, char const&) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/char_traits.h:329
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_set_length(unsigned int) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:219
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:276
0x4021214a: Firebase_Signer::handleToken() at ??:?
0x402052f1: FirebaseESP8266::begin(fb_esp_cfg_t*, fb_esp_auth_signin_provider_t*) at ??:?
0x40231d50: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_S_copy(char*, char const*, unsigned int) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:354
0x4021624c: uart_write at ??:?
0x40232178: std::char_traits<char>::assign(char&, char const&) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/char_traits.h:329
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_set_length(unsigned int) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:219
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace(unsigned int, unsigned int, char const*, unsigned int) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:472
0x40212bd4: HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
0x40203619: setup at ??:?
0x40226c0c: sdfat::FatPartition::clusterStartSector(unsigned int) const at ??:?
0x40214588: loop_wrapper() at core_esp8266_main.cpp:?
0x401002b9: cont_wrapper at ??:?
```

Serial debug log:

```
[... 233 lines cut ...]
    //Or use legacy authenticate method
    //Firebase.begin(DATABASE_URL, "<database secret>");

    Firebase.reconnectWiFi(true);
}

void loop()
{
    if (Firebase.ready() && !taskCompleted)
    {
        taskCompleted = true;
        Serial.println("Ready");
        delay(100);

         if (Firebase.getInt(fbdo, "/device-3/OpenClose/openPercent")){
            Serial.println("Success:");
            Serial.println(fbdo.intData());
          }else{
            Serial.println("Failed to get data");
            Serial.println(fbdo.errorReason().c_str());
          }

    }
}
```

mobizt, 2021-07-03:
Nothing is related to library based on your code and not related thing you mentioned.

You need to update the esp8266 sdk in platformio.

mobizt, 2021-07-03:
You also need to change usb cable for shorter cable and use large capacitor connect between 3v3 and gnd on your board.

closed 2021-07-03 by mobizt

kovikti, 2021-07-03:
I'm using a beefy cable, never had any problems with that. Also the system has been running for over half a year now, practically untouched, I tried 2 other boards, other cables, but no luck now.
I'm on a nodemcu board, I'll add extra capacitory but I doubt that it would help.

I see the cert has changed on the 1st July.

Info of the certificate (using sslshopper.com)
Common name: firebaseio.com
SANs: firebaseio.com, *.firebaseio.com
**Valid from July 1, 2021 to September 29, 2021**
Serial Number: a5a1b9f178774a630a00000000edbc63
Signature Algorithm: sha256WithRSAEncryption
Issuer: GTS CA 1D4

It is strange that axSSL reported RSA 4096bit key length, SSL checker sites report 2048. BearSSL does not print this info.

Any ideas for workaround? What I see now is that the library just became unusable :(

mobizt, 2021-07-03:
It's because unstable power supply which inrush current occurred  when data begin transmission.

**No certificate issue here** and the **certificate verification was not assign in your code**. 

The SSL handshake process can be done normally with reserved memory enough for transmit and receive which can be configurable via

**fbdo.setBSSLBufferSize(1024, 512); //rx/tx BearSSL buffer size.**

1024 bytes for rx and 512 bytes for tx is enough for normal data transfer which set as default.

I currently test for weeks until now as  preparing for the next update and everything is ok.

#### You should try other boards instead or provide the strong supply for board.

kovikti, 2021-07-04:
Tried another beefy power supply, other cables, added 1000uF capactior, tried other boards, all the same.
It definitely was working great until a few days ago.

```
Exception Cause: Not found

0x40214a66: raise_exception at core_esp8266_postmortem.cpp:?
0x40214a78: __unhandled_exception at ??:?
0x40100c02: get_unpoisoned_check_neighbors at umm_malloc.cpp:?
0x401010bc: umm_poison_free_fl at ??:?
0x401010b9: umm_poison_free_fl at ??:?
0x40274dcd: ethernet_input at ??:?
0x40100610: free at ??:?
0x40232684: operator delete(void*) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_op.cc:50
0x402319cc: operator delete[](void*) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_opv.cc:36
0x4020eb3e: _ZN10UtilsClass4delSEPc$constprop$0 at Signer.cpp:?
0x40275084: ethernet_input at ??:?
0x4020fe92: _ZN15Firebase_Signer19handleTokenResponseEv$part$0 at Signer.cpp:?
0x401004fd: millis at ??:?
0x402144a0: esp_yield at ??:?
0x40214b00: delay_end at ??:?
0x40100b65: check_poison_neighbors$part$0 at umm_malloc.cpp:?
0x40100a00: umm_get_ptr_context at umm_malloc.cpp:?
0x40100c71: umm_free_core at umm_malloc.cpp:?
0x4010109e: umm_free at ??:?
0x401010bc: umm_poison_free_fl at ??:?
0x402319cc: operator delete[](void*) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_opv.cc:36
0x4020f074: UtilsClass::reconnect(unsigned long) at ??:?
0x40203e60: ESP8266WiFiSTAClass::status() at ??:?
0x4021019b: Firebase_Signer::handleTokenResponse() at ??:?
0x40210ba0: Firebase_Signer::getIdToken(bool, char const*, char const*) at ??:?
0x40212ee4: Print::write(char const*) at ??:?
0x401004fd: millis at ??:?
0x40211dc6: _ZN15Firebase_Signer19tokenProcessingTaskEv$part$0 at Signer.cpp:?
0x401004fd: millis at ??:?
0x40231e3e: std::char_traits<char>::assign(char&, char const&) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/char_traits.h:329
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_set_length(unsigned int) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:219
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:276
0x4021217a: Firebase_Signer::handleToken() at ??:?
0x40205321: FirebaseESP8266::begin(fb_esp_cfg_t*, fb_esp_auth_signin_provider_t*) at ??:?
0x40231d7c: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_S_copy(char*, char const*, unsigned int) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:354
0x4021627c: uart_write at ??:?
0x402321a4: std::char_traits<char>::assign(char&, char const&) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/char_traits.h:329
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_set_length(unsigned int) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:219
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace(unsigned int, unsigned int, char const*, unsigned int) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:472
0x40212c04: HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
0x40203639: setup at ??:?
0x40226c38: sdfat::FatPartition::clusterStartSector(unsigned int) const at ??:?
0x402145b8: loop_wrapper() at core_esp8266_main.cpp:?
0x401002b9: cont_wrapper at ??:?
```

I believe poison check errors relate to heap, dynamic memory allocation etc.

mobizt, 2021-07-04:
I still tell you the same, your device is defected (memory problem) or the supply problem.

The library works normally.

The dynamic allocation is not the problem as the code is trying to send the http request to the Google API endpoint as normal.

PlatformIO board settings are ok.

You should change the device.

Library is [updated](https://github.com/mobizt/Firebase-ESP8266/releases/tag/v3.3.6) today but nothing is related to your problem.

mobizt, 2021-07-04:
Update.

I think I found the issue on BearSSL reserved memory which may be low in token generation process which may lead to crash.

I will update the library and let you try.

mobizt, 2021-07-04:
You can update the library to [v3.3.7](https://github.com/mobizt/Firebase-ESP8266/releases/tag/v3.3.7) which will fix this issue.

kovikti, 2021-07-04:
I check the following:
-Simple HTTPS request with WifiClientSecure to my Firebase db using the API key works, returns the nodes as json.
-Firebase-ESP8266 works using the legacy begint method, failes using config+auth (crash).

kovikti, 2021-07-04:
The update didn't really seem to help, maybe I made a mistake updating, however the legacy authentication method works, I'm sticking to that for now. Hopefully it'll be available for some time.
Thanks for all your hard work!

mobizt, 2021-07-04:
Please reinstall the library.

The issue is really strange and the situation and your internet connection code are not known.

Any memory leaks in your code?

Change device is another option that will prove the problem but your situations are not known and not related to library if the latest update can't help and you should find solution yourself.

The bad clone boards may have the glitch problems about noise, clock and instability.

Use the nice clone board from official WeMos (Lolin) store is recommend. 

NodeMCU clone is not recommend.