mobizt/Firebase-ESP8266 #225: Soft WDT reset ESP8266

JeffrySan, 2021-07-09:
It says that 

BSSL:_connectSSL: start connection
BSSL:CERT: 30 82 05 6b 30 82 04 53 a0 03 02 01 02 02 11 00 a5 a1 b9 f1 78 77 4a 63 0a 00 00 00 00 ed bc 63 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 30 46 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 22 30 20 06 03 55 04 0a 13 19 47 6f 6f 67 6c 65 20 54 72 75 73 74 20 53 65 72 76 69 63 65 73 20 4c 4c 43 31 13 30 11 06 03 55 04 03 13 0a 47 54 53 20 43 41 20 31 44 34 30 1e 17 0d 32 31 30 37 30 31 32 30 31 36 35 32 5a 17 0d 32 31 30 39 32 39 32 30 31 36 35 31 5a 30 19 31 17 30 15 06 03 55 04 03 13 0e 66 69 72 65 62 61 73 65 69 6f 2e 63 6f 6d 30 82 01 22 30 0d 06 09 2a 86 48 86 f7 0d 01 01 01 05 00 03 82 01 0f 00 30 82 01 0a 02 82 01 01 00 ac a3 4a b6 4c 76 d5 55 c5 77 4d ef c1 5f 2a ad b1 cb 26 fc 80 82 ab 6a 2c 80 89 f4 04 12 1f 2b 83 56 6f 55 da 3a cf 51 a3 2b 02 45 19 
BSSL:CERT: a7 5c a6 9d 02 f9 bc 33 5f 4d 66 35 e2 d2 eb 8e b7 9b f7 1f c8 6e ea 9f 4b a6 75 9b a3 3a b7 e1 27 76 ac 76 42 e1 48 93 96 67 b9 17 be 09 44 64 df 63 e5 cf a4 e7 9c 2c 3d 9c 2b b9 5d 42 ba 4f 08 46 89 65 58 04 4c 02 97 97 97 2c 87 d6 a7 12 4d 08 06 7a 96 dc ed 2f 8e 47 5c e9 60 6a 2a ed c5 48 bd f7 b1 57 03 5f d7 4d ed 00 42 08 d3 12 87 72 0f b0 c2 8b 5b 50 da ed d8 9c c8 e9 94 d3 f3 16 6e 27 07 fd da f4 fb 14 96 ab 23 5a cd 4e fe a9 04 28 ff fd 97 b7 3b aa 6d 1f 79 9c 20 75 26 0a e9 09 f1 99 e7 f4 61 32 00 67 c6 dd 02 db e2 91 9f d8 5e 2a 1f de ed d8 0c 65 bb b3 07 d0 c5 cf ab 75 ce 38 3f 83 1d 6c 43 de fe f8 90 ac 91 47 65 02 03 01 00 01 a3 82 02 7f 30 82 02 7b 30 0e 06 03 55 1d 0f 01 01 ff 04 04 03 02 05 a0 30 13 06 03 55 1d 25 04 0c 30 0a 06 08 2b 06 01 
BSSL:CERT: 05 05 07 03 01 30 0c 06 03 55 1d 13 01 01 ff 04 02 30 00 30 1d 06 03 55 1d 0e 04 16 04 14 ed 1b 1a 23 ee 1a a2 5e 8a a9 68 9e 79 f5 be 83 06 b9 60 28 30 1f 06 03 55 1d 23 04 18 30 16 80 14 25 e2 18 0e b2 57 91 94 2a e5 d4 5d 86 90 83 de 53 b3 b8 92 30 6d 06 08 2b 06 01 05 05 07 01 01 04 61 30 5f 30 2a 06 08 2b 06 01 05 05 07 30 01 86 1e 68 74 74 70 3a 2f 2f 6f 63 73 70 2e 70 6b 69 2e 67 6f 6f 67 2f 67 74 73 31 64 34 69 6e 74 30 31 06 08 2b 06 01 05 05 07 30 02 86 25 68 74 74 70 3a 2f 2f 70 6b 6
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

and the stack decode result is down below

Decoding stack results
0x402354e4: vsnprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 42
0x402273c1: uart_write(uart_t*, char const*, size_t) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\cores\esp8266\uart.cpp line 509
0x402231a8: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\cores\esp8266/HardwareSerial.h line 165
0x402236ac: Print::printf_P(char const*, ...) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\cores\esp8266\Print.cpp line 97
0x40250ca0: br_x509_decoder_run at src/x509/x509_decoder.c line 447
0x4025103c: br_x509_decoder_run at src/x509/x509_decoder.c line 672
0x4022319c: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\cores\esp8266/HardwareSerial.h line 164
0x40247294: br_sha1_update at src/hash/sha1.c line 131
0x4021bfcd: BearSSL::insecure_append(br_x509_class const**, unsigned char const*, size_t) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 674
0x4024bddc: br_ssl_hs_client_run at src/ssl/ssl_hs_client.c line 1871
0x40249dc8: jump_handshake at src/ssl/ssl_engine.c line 1081
0x4024a2ad: br_ssl_engine_recvrec_ack at src/ssl/ssl_engine.c line 1206
0x4021cf6a: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 537
0x4024a3d2: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1320
0x4021d1c8: BearSSL::WiFiClientSecure::_wait_for_handshake() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 561
0x4021d42f: BearSSL::WiFiClientSecure::_connectSSL(char const*) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 1119
0x40100b79: check_poison_neighbors(uint16_t) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\cores\esp8266\umm_malloc/umm_local.c line 71
0x40244aa8: ip4_output_if_opt_src at core/ipv4/ip4.c line 1764
0x40100b79: check_poison_neighbors(uint16_t) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\cores\esp8266\umm_malloc/umm_local.c line 71
0x40100e98: umm_malloc(size_t) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x40100a64: get_poisoned(void*, size_t) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\cores\esp8266\umm_malloc/umm_poison.c line 117
0x40244af0: ip4_output_if_opt at core/ipv4/ip4.c line 1577
0x401004dc: malloc(size_t) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\cores\esp8266\heap.cpp line 232
0x40100a64: get_poisoned(void*, size_t) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\cores\esp8266\umm_malloc/umm_poison.c line 117
0x40244b16: ip4_output_if at core/ipv4/ip4.c line 1550
0x402456bf: ip_chksum_pseudo at core/inet_chksum.c line 395
0x40240c06: tcp_output at core/tcp_out.c line 1361
0x4023fef1: tcp_create_segment at core/tcp_out.c line 190
0x40225c86: __delay(unsigned long) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\cores\esp8266\core_esp8266_wiring.cpp line 54
0x4021b948: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\libraries\ESP8266WiFi\src/include/ClientContext.h line 144
0x4021d597: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 228
0x40214891: FB_HTTPClient::connect() at C:\Users\user\Documents\Arduino\libraries\Firebase-ESP8266-master\src\wcs\esp8266\FB_HTTPClient.cpp line 134

i didn't understand, i try to remove some line of code, and it works, but after i re-upload it, it start error again, can some one explain to me what happend please?

i use this library
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>
#include <WiFiManager.h>         // https://github.com/tzapu/WiFiManager
#include <NTPClient.h>
#include <WiFiUdp.h>

JeffrySan, 2021-07-09:
for now, it's working when i change this line of code

fbdo1.setBSSLBufferSize(2048, 2048);
fbdo1.setResponseSize(2048);

from 1024 to 2048, but 2 days ago, i try to change the size, it didn't change anything

mobizt, 2021-07-09:
This is due to incoming stream payload is larger than the reserved BearSSL buffer, only header can be read and SSL client stuck in somewhere when library is waiting for reading the unknown length stream payload and wdt overflown occurred caused by WiFiClientSecure class itself.

In Google RTDB http stream implementation, the content length header is not assigned and the client will not know the actual payload size to handle which is not possible for client to reserve the fix size memory for it which work for all possible situation.

Reserve the 8k memory of Rx can be handle all large payloads but it costs free memory available when Firebase Data object is in used.

I will add warning and error info for stream in the next update to let user know the risk situation of device crash of too large stream payload and defer the device crashed due to this situation.  We can only defer the crash due to WiFiClientSecure stuck in unknown payload length process.

The recommend is increase the Rx buffer of BearSSL or reduce the data being stream.

closed 2021-07-09 by mobizt

JeffrySan, 2021-07-09:
Oohh okay, it's make sense, cause when i change the buffer size, it's working normal
is there any side effect to use big size of buffer?

mobizt, 2021-07-09:
The direct effect is free heap reduced by the reserve mem amount as the stream connected and the memory free after disconnected.

The side effect is nothing except for the low memory situation depends on your code memory usage.

JeffrySan, 2021-07-09:
Ohh okayy.. thankyouu so much for the information sir..

JeffrySan, 2021-07-12:
Sorry sir, i want to ask again, i have 3 device, the first device is uploaded with the success code, but when i try to upload to other device, it get the same error, even i already increase the buffer rx

the same thing happen when i try to upload the first device, some times it work, some times it isn’t, is there are any advice that i could follow? Like how to check the current buffer size, my stream payload size, so i can solve it?

mobizt, 2021-07-12:
As I post above, the next update will add the stream error info and defer the device crashes.

The info included the stream payload size of streaming node of your database.

mobizt, 2021-07-12:
Anyway you need to minimize the size of your data under stream node and should not grow by using push function.