mobizt/Firebase-ESP8266 #226: HELP set bssl buffersize

ayyysh04, 2021-07-15:
First of all library is super cool.
I want to know how can i set buffer size for big data as running below lines says there is no function like this.
Could u please tell the correct fnc name/example

CODE:
//Setting buffer for big data -
  firebaseData1.setBSSlBufferSize(1024,1024);
  firebaseData1.setResponseSize(1024);

mobizt, 2021-07-16:
It's here
https://github.com/mobizt/Firebase-ESP8266/tree/master/src#firebase-data-object-functions

The problem is you choose incorrect board to compile, you need to select ESP8266 board.

closed 2021-07-16 by mobizt

mobizt, 2021-07-16:
Typo in your code (l letter), should be **setBSSLBufferSize** instead of **setBSSlBufferSize**.

ayyysh04, 2021-07-16:
Kk thanks it worked 😊