mobizt/Firebase-ESP8266 #227: HELP Supported ardiuno

ayyysh04, 2021-07-16:
Can u please tell if this aruino support ur firebase library with multipath stream?
[image]

If not can u plz tell a arduino (not nodemcu )which support ur library with multi stream support.

Love ur work bro.

mobizt, 2021-07-16:
You should use [Discussions](https://github.com/mobizt/Firebase-ESP8266/discussions) instead of open issue.

The issue should be related to code not application.

You should lean and start to use ESP8266 with module or board like Wemos or NodeMCU, Wemos D1 Mini or D1 mini pro is recommended. 

The serial bridge board that has multi mcu is not recommend even it can use with this library.

If you want more gpio, analog pin, choose ESP32 is better than use Arduino + ESP8266.

closed 2021-07-16 by mobizt

ayyysh04, 2021-07-16:
actually i want to connect my tft and want to retrieve data using multi path.I am using arduino not nodemcu because tft consume 12 pins and there will be less pin for my relays as i am making this for commercial use.

could u suggest me a better way or should i go with the arduino mentioned in the screenshot

mobizt, 2021-07-16:
use spi or i2c fft screen instead. There are parallel to serial interface modules available.

ayyysh04, 2021-07-16:
actually i want to use a touch display

mobizt, 2021-07-16:
If it fit your requirement it's ok.

mobizt, 2021-07-16:
For me I will use ESP32 instead.

ayyysh04, 2021-07-16:
can i connect my tft with arduino and communicate between ardino and nodemcu using multidata serial comunication 
And then sync data of firbase with nodemcu?

Also using multi stream, heats up my esp8266 genric ,so should i increase heap memory or what??

mobizt, 2021-07-16:
I don't know what you mean but for board application you should go to arduino forum.

Multipath stream is just normal stream with function to parse your  parent node data based on child path.

The memory reserved for BearSSL client can be configured and should larger than your data (whole data) under parent node which you should test it. The next library update will add some waring info in the error because of unmatched between data and reserved memory of BearSSL.