mobizt/Firebase-ESP8266 #232: Configuring IP Address doesn't allow Firebase to begin properly

NiteAx, 2021-07-22:
Hi, i'm having an issue where configuring a specific IP when starting Wifi causes firebase to never start.

```C++
      WiFi.mode(WIFI_STA);
      IPAddress local_ip(192, 168, 0, 100);
      IPAddress gateway(192, 168, 0, 1);
      IPAddress subnet(255, 255, 255, 0);
      //WiFi.config(local_ip, gateway, subnet);
      WiFi.begin(ssid, pass);
```
Firebase Client starts fine as long as the line of code is commented out in the snippet above. 

To be clear, all other network functionality works fine. (ESP8266 Successfully connects to Wifi at the specified IP and runs a webserver. This is not an issue of setting a used IP.)

mobizt, 2021-07-22:
This is not related to the library.

Library does nothing with WiFi and ip stack.

mobizt, 2021-07-22:
Connect to AP or router doesn't mean your device has the internet accessibility.

You should test with HTTPS or HTTP or Ping examples.

closed 2021-07-22 by mobizt

mobizt, 2021-07-22:
If no internet you should set your router properly.

mobizt, 2021-07-22:
If your internet is ok and you used external WiFi connection management library or your own routine code, post your minimal code here.

NiteAx, 2021-07-23:
I was writing the minimum reproducible code using your custom_token example and I figured out the problem. You were right, while the static IP was configured and accessible on the local network, it turns out the NodeMCU had no internet connectivity. I was missing one line of code, which is to also set a dns address along with local_ip, gateway and subnet as in the snippet above. Sorry for wasting your time.