mobizt/Firebase-ESP8266 #234: cant read a string having more then 447 chars!

linktoahmad, 2021-07-22:
this is the code

        if(Firebase.getString(fbdo, "/tokens")){
          Serial.println(fbdo.stringData().c_str());
          } 
          else{
            Serial.println(fbdo.errorReason().c_str());
            }

works great with string less than 447 chars  
the error displays nothing
on updating the data in firebase from less then 447 chars to more than 447 chars the fbdo.errorReason().c_str() keep on displaying the data before update!!

mobizt, 2021-07-22:
This is not a bug.

In ESP8266 BearSSL library, you can reserve the memory for it's transmit and receive buffer for transmission data handling.

The Firebase Data object creates a WiFiClientSecure that use core SSL's BearSSL library.

With Firebase library, this option can set from setBSSLBufferSize like this
```cpp
fbdo.setBSSLBufferSize(2048 /* rx buffer size in byte from 512 to 16384 */, 2048/* tx buffer size in byte from 512 to 16384 */);
```

mobizt, 2021-07-22:
You should set this option to match your payload size to receive or send.
You should try to increase the value until it is ok for your data.

mobizt, 2021-07-22:
You should use the Discussion first as I wrote in the issue template.
If it is the bug that you provide the reproduce code, then it can review and translate to the issue.

closed 2021-07-22 by mobizt