mobizt/Firebase-ESP8266 #246: HELP WANTED ! mutliple values wanted in multiStreampath

ayyysh04, 2021-08-07:
I am working on a project and wanted a help to this.

[image]

Here i am streaiming path switch 1 and switch 2 using multipathstreaming.
But i wanted to get both the values : pin and status when switch child gets triggered.
How can i archive this?
should i use json parsing ?or any other method.

My code of multiStreamingPath :
FirebaseData firebaseData1;

String parentPath = "/Devices";
String childPath[1] = {"/Room1"};

size_t childPathSize = 1;
void handlePin(String path,String value)
{
if(path == "/Board1/Switch1")
{
Serial.println("Switch 1 data recieved");
Serial.println(value);
}
else if(path=="/Board1/Switch2")
{
Serial.println("Switch 2 data recieved");
Serial.println(value);
}

}

void streamCallback(MultiPathStreamData stream)
{
Serial.println();
Serial.println("Stream Data1 available");

size_t numChild = sizeof(childPath) / sizeof(childPath[0]);

for (size_t i = 0; i < numChild; i++)
{
if (stream.get(childPath[i]))
{
Serial.println("stream ");
Serial.println(stream.get(childPath[i]));

  Serial.println("path: "+ stream.dataPath + ", type: " + stream.type + ",value: " + stream.value);

  //Custom function to make pin high and low
  handlePin(stream.dataPath,stream.value);
}
}

Serial.println();
}

void streamTimeoutCallback(bool timeout)
{
if (timeout)
Serial.println("stream timeout, resuming...\n");
}

void setup()
{

Serial.begin(115200);

WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
Serial.print("Connecting to Wi-Fi");
while (WiFi.status() != WL_CONNECTED)
{
Serial.print(".");
delay(300);
}
Serial.println();
Serial.print("Connected with IP: ");
Serial.println(WiFi.localIP());
Serial.println();

Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
Firebase.reconnectWiFi(true);

//Check stream is working or not
if (!Firebase.beginMultiPathStream(firebaseData1, parentPath,childPath,childPathSize))
{
Serial.println("cant begin stream connection\n");
Serial.println("Reason: "+firebaseData1.errorReason());
}

Firebase.setMultiPathStreamCallback(firebaseData1,streamCallback,streamTimeoutCallback);
}
void loop()
{
}

mobizt, 2021-08-07:
You should assign the child path values in childPath array as what you want to get the value e.g. **Room1/Switch1** and **Room1/Switch2**

You can also use normal stream and parse the data under **/Devices** manually.

Multistream is only parse the value for you which childPath used as the key the parse value from stream json payload.

closed 2021-08-07 by mobizt

ayyysh04, 2021-08-07:
Can stream.value return a json??

mobizt, 2021-08-07:
Why don't test yourself?

mobizt, 2021-08-07:
You should learn how Firebase works before using tool or sdk or library.

ayyysh04, 2021-08-07:
No, actually i over used my nodemcu and its not working right now

mobizt, 2021-08-07:
There are many examples and documentation.

This is the repository for code related issue not for getting started to Firebase.

If you ever use or program any Firebase client app before, you will understand how to use the library.

ayyysh04, 2021-08-07:
ok thanks,i will get back to this after my new nodemcu comes

mobizt, 2021-08-07:
You should use community forum for general help. I'm trying to support and spend the time but you should try it first and read the document related to what you're working.

ayyysh04, 2021-08-07:
ahh really sorry for that.

ayyysh04, 2021-08-09:
I am trying to get the json data from multipatahstream object in handlepin() function, but i cant find function for the same.
database:
[image]

Json getting from stream :
[image]

My code :
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

#define FIREBASE_HOST "prototypexreal-default-rtdb.firebaseio.com/"
#define FIREBASE_AUTH "FoUFEsd9oeh1iQfxswlxJwkCP5ybHEWKTp3i3SyP"

/*Define the WiFi credentials */
#define WIFI_SSID "WIFI_AP"
#define WIFI_PASSWORD "WIFI_PASSWORD"

FirebaseData firebaseData1;

String parentPath = "/Devices";
String childPath[1] = {"/Room 1"};

size_t childPathSize = 1;

void handlePin(String path,FirebaseJson &value)
{
 if(path == "/Board1/Switch1")
  {
    Serial.println("Switch 1 data recieved");  
    FirebaseJsonData result;
    value.get(result,"Board1/Switch1");
     Serial.println(result.stringValue);
  }
  else if(path=="/Board1/Switch2")  
  {
    Serial.println("Switch 2 data recieved");  
    FirebaseJsonData result;
    value.get(result,"Board1/Switch2");
    Serial.println(result.stringValue);
    }
  
}

void streamCallback(MultiPathStreamData stream)
{
 Serial.println();
 Serial.println("Stream Data1 available");
 
 size_t numChild = sizeof(childPath) / sizeof(childPath[0]);

 for (size_t i = 0; i < numChild; i++)
  {
    if (stream.get(childPath[i]))
    {
      Serial.println("stream ");
      Serial.println(stream.get(childPath[i]));
      
      Serial.println("path: "+ stream.dataPath + ", type: " + stream.type + ",value: " + stream.value);

      //Custom function to make pin high and low
      handlePin(stream.dataPath,stream);
    }
  }

  Serial.println();
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
    Serial.println("stream timeout, resuming...\n");   
}

void setup()
{
 
  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  //Check stream is working or not
  if (!Firebase.beginMultiPathStream(firebaseData1, parentPath,childPath,childPathSize))
   {
    Serial.println("cant begin stream connection\n");
    Serial.println("Reason: "+firebaseData1.errorReason());
    }

  Firebase.setMultiPathStreamCallback(firebaseData1,streamCallback,streamTimeoutCallback);
}
void loop()
{
}

mobizt, 2021-08-09:
Because you don't get what I mentioned above.

You should do this.

**String childPath[2] = {"/Board1/Switch1", "/Board1/Switch2"};**

The childPath array contains the children paths that you want to listen the value changes.

ayyysh04, 2021-08-09:
But i want to parse it as json ,because i will be adding more data under Switch .
I think its not possible ,as stream.value gives String.

I think single child streaming will work?

mobizt, 2021-08-09:
You can't do with multipath stream, use [normal stream](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/DataChangesListener/Callback/Callback.ino) instead.

The multipath stream internally parses the values from json payload and don't allow the json payload as this is the design of this function and you can use [normal stream](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/DataChangesListener/Callback/Callback.ino) and handle payload yourself.

mobizt, 2021-08-09:
stop using multipath stream and use normal stream i your solution.

ayyysh04, 2021-08-09:
yup i got that,thanks 
Working on it

ayyysh04, 2021-08-10:
database :
[image]

Code:
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

#define FIREBASE_HOST "x-x-x"
#define FIREBASE_AUTH "x-x-x"

/*Define the WiFi credentials */
#define WIFI_SSID "WIFI_AP"
#define WIFI_PASSWORD "WIFI_PASSWORD"

FirebaseData firebaseData1;

String Path = "/Devices/Room1/Board1";
String ChildPath[2] = {"Switch1","Switch2"};

void setup()
{
 
  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (!Firebase.beginStream(firebaseData1, Path))
  {
      Serial.println("Could not begin stream");
      Serial.println("REASON: " + firebaseData1.errorReason());
      Serial.println();
  }

  }
void loop()
{
  
  if (!Firebase.readStream(firebaseData1))
  {                                             Serial.println();
      Serial.println("Can't read stream data");
      Serial.println("REASON: " + firebaseData1.errorReason());
      Serial.println();
  }
  else
   Serial.println("Streaming....");

  if (firebaseData1.streamTimeout())
  {
      Serial.println();
      Serial.println("Stream timeout, resume streaming...");
      Serial.println();
  }

  if (firebaseData1.streamAvailable())
  {
       if (firebaseData1.dataType() == "json")
      {
        FirebaseJson &json = firebaseData1.jsonObject();
      
        for(int i=0;i<2;i++)
        {
       FirebaseJsonData jsondata;  
        json.get(jsondata,ChildPath[i]);
        Serial.println(jsondata.stringValue);
        }   
      }
  }
  else
   Serial.println("No stream available");
}

Output:
[image]

Ca you tell why its not getting stream data??