mobizt/Firebase-ESP8266 #251: HELP NEEDED!! Http post request not gettiing detect by firebase esp library

ayyysh04, 2021-09-05:
I am trying to on and off a led connected with Arduino using firebase and voiceflow.

I am giving a http post request in voiceflow to change the data like
POST request :https://x-x-x-x-x-x-x/Devices.json
Content type value : "application/json"
Body : {"Nodemcu 1":{"status":0}}

This command is successfully running and updated the value of the firebase rt database but data is not received by the Arduino (firebase library)

I think the issue is related to the api ,Can you please tell the reason for the same ,I tried using get function and stream to get the data from firebase but its not getting the data changed using http post/put/patch request.

Is there any alternative to do the same? Should getJson function be a good idea to fix this ??

mobizt, 2021-09-05:
You should use [Discussions](https://github.com/mobizt/Firebase-ESP8266/discussions) for question.

Make sure you update the library to latest version.

You must spend more time to try and making understand the [Firebase RTDB REST API](https://firebase.google.com/docs/reference/rest/database) when you're going to use it.

I can't help because you did not follow the issue template and post the part of your code that you have tried.

mobizt, 2021-09-05:
The examples provide enough use cases that you can test.

The debug information from **fbdo.errorReason** will tell you the error which you should always use it during test.

closed 2021-09-05 by mobizt

ayyysh04, 2021-09-05:
I tried using fbdo.errorReason to know whats the reason.
I updated the library and http request is working fine now,Thanks for that ,but i am still getting a bug.

CODE:
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>
#define FIREBASE_HOST "x-x-x-"
#define FIREBASE_AUTH "x-x-x"
#define WIFI_SSID "x-x-"
#define WIFI_PASSWORD "xx-x-"

IRrecv irrec(4);
decode_results results;
FirebaseData firebaseData;
bool bstate1;
void setup()
{
  irrec.enableIRIn();
  //pin modes
  pinMode(5,OUTPUT);

  //IR sensor
  pinMode(4,INPUT);

  //serial monitor config
  Serial.begin(115200);

  //Wifi config
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  //Firebase config
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.setInt(firebaseData,"/Devices/Nodemcu 1",0);
  //buttons state
  bstate1=false;
  Firebase.reconnectWiFi(true);
}
void loop()
{
 if(Firebase.getInt(firebaseData,"/Devices/Nodemcu 1/status"))
 {  
 if(firebaseData.intData()==1 && bstate1==false)
 {
  Serial.printf("Led is On\n");
  digitalWrite(5,HIGH);
  bstate1=true; 
 }
 else if(firebaseData.intData()==0 && bstate1==true)
 {
  
  Serial.printf("Led is Off\n");
 digitalWrite(5,LOW); 
bstate1==false;
 }
 }
 else
 Serial.printf(firebaseData.errorReason().c_str());
 Serial.printf("\n");
 IRcodeSensor();
}

void IRcodeSensor()
{
  if (irrec.decode(&results)) {
 const uint16_t ircode = results.value;

if(ircode==1)
{
  if(bstate1==false)
  {
   bstate1=true;
   if(Firebase.setInt(firebaseData,"/Devices/Nodemcu 1/status",1))
    Serial.printf("Done\n");
   else
    firebaseData.errorReason().c_str();
    digitalWrite(5,HIGH); 
  }
 else
 {
  digitalWrite(5,LOW);
  if(Firebase.setInt(firebaseData,"/Devices/Nodemcu 1/status",0))
    Serial.printf("Done\n");
   else
    firebaseData.errorReason().c_str();
  
  bstate1=false;
}

}
irrec.resume(); // Receive the next value 
}
}

OUTPUT:When i first boot up the arduino ,Data mismatch error comes for about 3-4 times and after that it work fine .
Then when i press the ir remote button,it takes very high time to set value in firebase,How i know that?well thats because the done command is print after some times.

mobizt, 2021-09-05:
That is inappropriate and inefficient to use library to 
1. Call get inside the loop function and run it repeatedly without provide the time
ING for execution.

2. The get function send http request (REST API) to server which the header size is approx. 40 bytes to 1 k depends on the authentication method used, you can read this in the Readme doc.

3. The TCP client needs to open and close connection too often as the http mode e.g. get, post, push, patch and delete, changed in a new session which will waste the time to establish new connection.

You should use stream instead to avoid those issues and more effective.

Any of your code for third party sensor lib used  in loop can be delayed as it need to wait hardware to be ready.

mobizt, 2021-09-06:
I also update the comment in basic examples to avoid these use cases
https://github.com/mobizt/Firebase-ESP8266/blob/abcf739e67484aecfb795397353386fed06f5b41/examples/Basic/Basic.ino#L151-L184

ayyysh04, 2021-09-06:
ok got that what you are saying but in this code i am using function without using a delay or millis but it still working with no lag at all whereas replacing stream with get function introduces delay and type mismatch eroor
CODE:
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>
#define FIREBASE_HOST "-x-x"
#define FIREBASE_AUTH "-x-xx-x-"
#define WIFI_SSID "-x-x-x-"
#define WIFI_PASSWORD "-x-x-x"

IRrecv irrec(4);
decode_results results;
FirebaseData firebaseData;
FirebaseData firebaseData2;
bool bstate1;
void setup()
{
  irrec.enableIRIn();
  //pin modes
  pinMode(5,OUTPUT);

  //IR sensor
  pinMode(4,INPUT);

  //serial monitor config
  Serial.begin(115200);

  //Wifi config
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  //Firebase config
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.setInt(firebaseData,"/Devices/Nodemcu 1",0);//Automatically adds room
  //buttons state
  bstate1=false;
  Firebase.reconnectWiFi(true);
   if (!Firebase.beginStream(firebaseData2,"/Devices/Nodemcu 1"))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

}
void loop()
{
  if (!Firebase.readStream(firebaseData2))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't read stream data...");
    Serial.println("REASON: " + firebaseData2.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  if (firebaseData2.streamTimeout())
  {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
  if (firebaseData2.streamAvailable())
  {
   if(firebaseData2.dataType() == "int")
   {
    if(firebaseData2.intData()==1 && bstate1==false)//switch on
    {
    digitalWrite(5,HIGH);
    bstate1=true;
    }
    else if(firebaseData2.intData()==0 && bstate1==true)
    {
    digitalWrite(5,LOW);
    bstate1=false;
    }
    Serial.println(firebaseData2.intData());
   }
    Serial.println();
  }
  IRcodeSensor();
}

void IRcodeSensor()
{
  if (irrec.decode(&results)) {
 const uint16_t ircode = results.value;
 Serial.println(ircode);

if(ircode==1)
{
  if(bstate1==false)
  {
   digitalWrite(5,HIGH); 
   bstate1=true;
   Firebase.setInt(firebaseData,"/Devices/Nodemcu 1/status",1);
  }
 else
 {
  digitalWrite(5,LOW);
  Firebase.setInt(firebaseData,"/Devices/Nodemcu 1/status",0);
  bstate1=false;
}

}
else
 Serial.println("Wrong ircode");
irrec.resume(); // Receive the next value 
}
}

OUTPUT :No error in recieving data and setting data ,works fine

mobizt, 2021-09-06:
Did you understand "close and reopen session" I mentioned and why this happens and the effect of this in SSL connection.

Are you sure the IRremote lib does not block? How you prove that.

mobizt, 2021-09-06:
Normally the TCP session was kept open (server is always connected or keep alive) for the next request to speed up data transfer, this is how the session was reuse. 

If the next request has different HTTP method e.g. the previous request is GET (from get function) while the next request is POST  request (from push function) which its method is different from the previous one, the TCP session need to be closed (server disconnected) and establish the new connection.

Every time secure TCP session was new established,  the SSL handshake is need to be performed and it takes time about 1-2 seconds to be done.