mobizt/Firebase-ESP8266 #256: Multiple libraries were found for "SD.h" when using both FirebaseJson and ESP Firebase Client

phakhee, 2021-09-23:
I installed both Firebase Arduino Client Library for ESP8266 and ESP32 version 2.5.1 and FirebaseJSON version 2.5.0. As I try to upload the script to my WeMos, I get the error meesage:

`collect2: error: ld returned 1 exit status
Multiple libraries were found for "SD.h"
 Used: /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/SD
 Not used: /Users/user/Desktop/Arduino.app/Contents/Java/libraries/SD`

`exit status 1
Error compiling for board LOLIN(WEMOS) D1 mini Pro.`

When removing the Firebase Arduino Client Library for ESP8266 and ESP32 imports, the script uploads fine without problem to my WeMos.

I also get a repeating red text on my Arduino IDE console:
`/Users/phakheeman/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /var/folders/nm/1d0_95tn0876rwqns3f33c300000gn/T/arduino_build_344249/libraries/FirebaseJson/cJSON/cJSON_Utils.c.o: in function `cJSONUtils_FindPointerFromObjectTo':
/Users/phakheeman/Documents/Arduino/libraries/FirebaseJson/src/cJSON/cJSON_Utils.c:199: multiple definition of `cJSONUtils_FindPointerFromObjectTo'; /var/folders/nm/1d0_95tn0876rwqns3f33c300000gn/T/arduino_build_344249/libraries/Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32/json/cJSON/cJSON_Utils.c.o:/Users/phakheeman/Documents/Arduino/libraries/Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32/src/json/cJSON/cJSON_Utils.c:199: first defined here
/Users/phakheeman/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /var/folders/nm/1d0_95tn0876rwqns3f33c300000gn/T/arduino_build_344249/libraries/FirebaseJson/cJSON/cJSON_Utils.c.o: in function `cJSONUtils_GetPointer':`

I'm using a Macbook Pro 2020 with an M1 chip.

I tried installing the Firebase client that's only for ESP8267 but that didn't fix it either.

mobizt, 2021-09-23:
The error showed the problem already.

Because you install other third party libraries named SD.h which is already installed in ESP8266 Arduino Core library

Remove this folder.
/Users/user/Desktop/Arduino.app/Contents/Java/libraries/SD

mobizt, 2021-09-23:
Are you new to this Firebase library?

The FirebaseJson library was included in the Firebase library already and not need to install.

closed 2021-09-23 by mobizt

phakhee, 2021-09-23:
Im indeed new, sorry :))