mobizt/Firebase-ESP8266 #276: BUG firebase access stops after ~59 minutes in version 3.6.2 but continues to work in 3.6.0

amitrao17, 2021-11-19:
I am interacting with a firebase project using email/password authentication. I was able to interact indefinitely (days) with no issue in version 3.6.0. However, when I updated the library to 3.6.2, I was no longer able to interact with the firebase project after about ~59 minutes.

Function calls such as Firebase.setString were failing (returning 0), and calling firebaseData.errorReason() returned MISSING_REFRESH_TOKEN 

There was more in the errorReason text but I didn't save the output. I can generate it again if that is helpful.

Resetting the device allowed interaction to resume, since Firebase.begin was called again.

mobizt, 2021-11-19:
Thanks again for reporting this bug.

Now the issue was fixed and the library [v3.6.3](https://github.com/mobizt/Firebase-ESP8266/releases/tag/v3.6.3) is now available.

For testing, to refresh token earlier before 60 minutes, e.g. 1 minute, you can use
```cpp
config.signer.preRefreshSeconds = 3600 - 60; 
```

closed 2021-11-19 by mobizt

amitrao17, 2021-11-19:
Thank *you*! Your library is incredibly helpful and well implemented