mobizt/Firebase-ESP8266 #277: BUG Firebase.pathExisted returns true on non-existed path

lexxwork, 2021-11-19:
**Firebase.pathExisted() returns true on non-existed path**

**IDE and its version:**

- Arduino, PlatformIO
- Version 5.2.4a3

**ESP8266 Arduino Core SDK version**

- Version 3.2.0

**library Firebase-ESP8266 v3.6.3**

**platformio.ini**

```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags = -D PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED
monitor_filters =
	default
	log2file
	esp8266_exception_decoder
upload_port = /dev/ttyUSB0
upload_speed = 115200
monitor_speed = 115200
build_type = debug
lib_deps = https://github.com/mobizt/Firebase-ESP8266.git#v3.6.3
```

**my code**
```cpp
//"device_task/arduino-home"
String path = FPSTR(_str_device_task) + this->Id; 

//should be true, but false
if (!Firebase.pathExisted(this->fbdo, path.c_str()))
{
  log_println(F("No task assigned"));
  return;
}

//it's ok also
if (!Firebase.getJSON(this->fbdo, path, &json) || !this->fbdo.dataAvailable())
{
  log_println(F("Error: Firebase.getJSON"));
  return;
}

//json.toString() produces "{}";

//it fails as expected
if (!json.get(jsData, F("state")))
{
  log_println(F("Error: 'state' not found"));
  return;
}
```

mobizt, 2021-11-19:
PathExisted is not available from Firebase APIs but only implement by the library based on null response.

You should post your data structure here instead of few line of code.

I can't test anything for you.

Create your minimal and complete working code to reproduce the issue for me to test.

mobizt, 2021-11-19:
To reproduce and confirm the issue, you need to add the node, check for existent, delete the node and check for existent.

closed 2021-11-19 by mobizt

mobizt, 2021-11-19:
Due to recently [bug on Google server side](/https://github.com/mobizt/Firebase-ESP-Client/discussions/165#discussioncomment-1561941), the temporary workaround was applied but I forgot to apply to the pathExisted then it is still not work.

And I check again and found that the Google server bug as gone now, you can comment  or remove this line to make pathExisted to work as normal.
https://github.com/mobizt/Firebase-ESP8266/blob/61a07ef35864583cc3fe15e0e5ff9a053e048b30/src/FirebaseFS.h#L42-L44

I will remove this Google bug workaround in the next library update soon.