mobizt/Firebase-ESP8266 #280: Wdt reset in FirebaseData::StreamEventCallback function

lexxwork, 2021-11-24:
**Wdt reset when**
```
FirebaseJson json;
FirebaseJsonData jsData;
```
**defined in  FirebaseData::StreamEventCallback function**

**IDE and its version:**

- Arduino, PlatformIO
- Version 5.2.4a3

**ESP8266 Arduino Core SDK version**

- Version 3.2.0

**library Firebase-ESP8266 v3.6.3**

Project to reproduce [esp8266-firebase-test](https://github.com/lexxwork/esp8266-firebase-test)
May cause at first time or after few loops
_Wi-Fi access parameters defined in **secrets.ini**_

mobizt, 2021-11-24:
Post you minimum code that reproduce the error here.

closed 2021-11-24 by mobizt

mobizt, 2021-11-24:
Follow the example and try to understand it first.
https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/DataChangesListener/Callback/Callback.ino