mobizt/Firebase-ESP8266 #282: wdt reset in FirebaseData::StreamTimeoutCallback

lexxwork, 2021-11-24:
### Wdt reset in  FirebaseData::StreamTimeoutCallback when define  FirebaseJson json and FirebaseJsonData jsData inside a function

**IDE and its version:**

- Arduino, PlatformIO
- Version 5.2.4a3

**ESP8266 Arduino Core SDK version**

- Version 3.2.0

**library Firebase-ESP8266 v3.6.3**

**platformio.ini**

### **full TEST project to reproduce** 
### https://github.com/lexxwork/esp8266-firebase-test
### Please, just clone it and fill your wifi settings in _secrets.ini_. Set up your upload_port and monitor_port in _platformio.ini_.

This test project includs all needed code 
```cpp
 FirebaseJson json;
 FirebaseJsonData jsData;
```
is needed to load json data with ```Firebase.getJson``` in full project, also ```Firebase.getString``` is used for simple test.

<details>
<summary>platformio.ini</summary>

```ini
[platformio]
extra_configs =
  secrets.ini

[env]
platform = espressif8266@3.2.0
board = nodemcuv2
framework = arduino
build_flags = 
	${secrets.common}
  -D PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED
monitor_filters = 
	default
	log2file
	esp8266_exception_decoder
upload_speed = 115200
monitor_speed = 115200
lib_deps = https://github.com/mobizt/Firebase-ESP8266.git#v3.6.3
test_ignore = test_native

[env:nodemcuv2-linux]
; extends = env:nodemcuv2
build_type = debug
build_flags = 
  ${env.build_flags}
  ${secrets.nodemcuv2-linux}
upload_port = /dev/ttyUSB-arduino2.4
monitor_port = /dev/ttyUSB-arduino2.4
```
</details>

**main.cpp**
```cpp
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>
#include <secrets.h>
#include <millistimer.h>

FirebaseAuth auth;
FirebaseConfig config;
[... 144 lines cut ...]
0x4023fe14 in operator delete(void*) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_op.cc:50
0x40213e6c in FB_RTDB::readStream(FirebaseData*) at .pio/libdeps/nodemcuv2-linux/Firebase ESP8266 Client/src/rtdb/FB_RTDB.cpp:537
0x40213efe in FB_RTDB::runStreamTask() at .pio/libdeps/nodemcuv2-linux/Firebase ESP8266 Client/src/rtdb/FB_RTDB.cpp:838 (discriminator 5)
0x40211448 in std::_Function_handler<void (), UtilsClass::set_scheduled_callback(std::function<void ()>)::{lambda()#1}>::_M_manager(std::_Any_data&, std::_Function_handler<void (), UtilsClass::set_scheduled_callback(std::function<void ()>)::{lambda()#1}> const&, std::_Manager_operation) at /home/lexx/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:268
0x4020cbe0 in std::_Function_handler<void (), UtilsClass::set_scheduled_callback(std::function<void ()>)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /home/lexx/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:289
0x40233b50 in std::_Function_handler<void (), std::_Bind<void (FB_RTDB::*(FB_RTDB*))()> >::_M_invoke(std::_Any_data const&) at /home/lexx/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:289
0x40233b50 in std::_Function_handler<void (), std::_Bind<void (FB_RTDB::*(FB_RTDB*))()> >::_M_invoke(std::_Any_data const&) at /home/lexx/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:289
0x401005fd in millis at /home/lexx/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_wiring.cpp:188
0x40233b71 in std::_Function_handler<void (), std::_Bind<void (FB_RTDB::*(FB_RTDB*))()> >::_M_invoke(std::_Any_data const&) at /home/lexx/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:293
0x40100105 in std::function<void ()>::operator()() const at /home/lexx/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:623
0x4021f3ec in esp8266::InterruptLock::InterruptLock() at /home/lexx/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/interrupts.h:25
 (inlined by) run_scheduled_functions() at /home/lexx/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Schedule.cpp:151
0x4022067d in loop_end at /home/lexx/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:188
0x402206b5 in loop_wrapper() at /home/lexx/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:202
0x401003ad in cont_wrapper at /home/lexx/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/cont.S:81
0x401000c4 in app_entry at /home/lexx/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:332
0x40240005 in dhcp_stop at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:431
0x40240055 in glue2esp_ifupdown at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:781
0x40100000 in _stext at ??:?
0x40232200 in sdfat::SdSpiCard::writeSector(unsigned int, unsigned char const*) at /home/lexx/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266SdFat/src/SdCard/SdSpiCard.h:229
 (inlined by) sdfat::FatFormatter::initFatDir(unsigned char, unsigned int) at /home/lexx/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266SdFat/src/FatLib/FatFormatter.cpp:94
0x40220b22 in __wrap_system_restart_local at /home/lexx/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_postmortem.cpp:188

```
</details>

lexxwork, 2021-11-24:
I want to make a request to database inside a callback - why this is improper used?

mobizt, 2021-11-24:
Don't need to open the issue as I convert this into discussion already.

The error is because stack overflow as ESP8266 has only 4K and this is not bug but user related memory management

This is the improper usage of async/non-async with the same FirebaseData object which used under the callback function.

You should aware that in the callback or ISR function, the memory and time consumed tasks to run in that function should be avoided.

The 8 - 10 k memory required for SSL data + additional reservation by the WiFiClientSecure (reserved for RX and TX).

If the session of FirebaseData object, fbdo for this case, was not closed (release memory) prior to it reused  in the callback function, the stack memory will required to push (keep) all memory reserved and used by fbdo in previous call (Firebase.setTimestampAsync), then close the session and re-open the session again for Firebase.getString in the callback function which this cause the stack to glow.

The proper usage to prevent stack overflow
1. use different FirebaseData object between sync and async or
2. use non-aync function.

closed 2021-11-24 by mobizt