mobizt/Firebase-ESP8266 #302: StramAvailabe in readStream also writing on path other than that used in the beginStream

pelizz, 2022-02-13:
Using ESP8266 with platformio in a Tasmota user driver:
1) I beginStream on a certain path for example /room/config : Firebase.beginStream(firebaseData, "/room/config")
2) I use readStream in loop:
if (!Firebase.readStream(firebaseData))
        {...}
        if (firebaseData.streamAvailable())
        {..}
3) StreamAvailable is trigged when i write on this path -> OK
**But Stream available is trigged ALSO when i write on another path; for example /boilerConfig**
 for example using Firebase.setJSON(firebaseData, "/boilerConfig", json1)
In this case firebaseData.streamAvailable() reports "/" for firebaseData.dataPath().c_str() 

Using    https://github.com/mobizt/Firebase-ESP8266#v3.8.2

mobizt, 2022-02-14:
It's impossible.

closed 2022-02-14 by mobizt

mobizt, 2022-02-14:
Your code may do something incorrectly or not properly.

You should separate to use FirebaseData object for stream from other functions.

Please see the examples for how to use it properly.
https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/DataChangesListener/Callback/Callback.ino
https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/DataChangesListener/NoCallback/NoCallback.ino

mobizt, 2022-02-14:
You should post your code and database structure snapshot.

mobizt, 2022-02-14:
Even you use the same FirebaseData objeect sharing between stream and other functions.
https://github.com/mobizt/Firebase-ESP8266/tree/master/examples/DataChangesListener/SingleDataObject

Everything works and ok.

The library was intensively tested for many years and does not have this kind of issue.

pelizz, 2022-02-14:
> Your code may do something incorrectly or not properly.
>
> You should separate to use FirebaseData object for stream from other functions.
>
> Please see the examples for how to use it properly. [...]

Sorry ...i used the same object for read and write. I will try  with different objicts. Thank you.

pelizz, 2022-02-14:
Sorry...i used the same Firebase object for readStream and for setJSON.
I will try .

Thank you.
Maurizio