mobizt/Firebase-ESP8266 #307: Token is not ready Wemos D1 mini pro

floatAsNeeded, 2022-03-01:
Hi!

I had recently a problem after updating. I didn't get any error this time (as per the previous discussion), but I tried my code and I got an error. Then later on I tried the example code and I still get this error:

```
19:55:24.860 -> ............
19:55:29.216 -> Connected with IP: 10.0.0.6
19:55:29.216 -> 
19:55:29.216 -> Firebase Client v3.9.3
19:55:29.216 -> 
19:55:29.216 -> Set int... token is not ready
19:55:34.232 -> Set int... token is not ready
```

That's the code I'm using to test:

```
/**
 * Created by K. Suwatchai (Mobizt)
 * 
 * Email: k_suwatchai@hotmail.com
 * 
 * Github: https://github.com/mobizt/Firebase-ESP8266
 * 
 * Copyright (c) 2022 mobizt
 *
*/

/** This example will show how to authenticate using 
 * the legacy token or database secret with the new APIs (using config and auth data).
*/

#if defined(ESP32)
#include <WiFi.h>
#include <FirebaseESP32.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>
#endif

//Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID ""
#define WIFI_PASSWORD ""

/* 2. If work with RTDB, define the RTDB URL and database secret */
#define DATABASE_URL "weather-station-95094.firebaseio.com" //<databaseName>.firebaseio.com or <databaseName>.<region>.firebasedatabase.app
#define DATABASE_SECRET ""

/* 3. Define the Firebase Data object */
FirebaseData fbdo;

/* 4, Define the FirebaseAuth data for authentication data */
FirebaseAuth auth;

/* Define the FirebaseConfig data for config data */
FirebaseConfig config;

unsigned long dataMillis = 0;
int count = 0;

void setup()
{

    Serial.begin(115200);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.print("Connecting to Wi-Fi");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

    /* Assign the certificate file (optional) */
    //config.cert.file = "/cert.cer";
    //config.cert.file_storage = StorageType::FLASH;

    /* Assign the database URL and database secret(required) */
    config.database_url = DATABASE_URL;
    config.signer.tokens.legacy_token = DATABASE_SECRET;

    Firebase.reconnectWiFi(true);

    /* Initialize the library with the Firebase authen and config */
    Firebase.begin(&config, &auth);

    //Or use legacy authenticate method
    //Firebase.begin(DATABASE_URL, DATABASE_SECRET);
}

void loop()
{
    if (millis() - dataMillis > 5000)
    {
        dataMillis = millis();
        Serial.printf("Set int... %s\n", Firebase.setInt(fbdo, "/test/int", count++) ? "ok" : fbdo.errorReason().c_str());
    }
}
```

I deleted the secret to post here and my WiFi credentials. Any ideas what could be the problem? I'm using Arduino IDE 1.8.19 and SDK 3.0.2 for the ESP module core

Thank you in advance

mobizt, 2022-03-01:
Thanks for reporting this issue.

The library update [v3.9.4](https://github.com/mobizt/Firebase-ESP8266/releases/tag/v3.9.4) is available with this bug fixed, please update.

closed 2022-03-01 by mobizt

floatAsNeeded, 2022-03-01:
No problems. Thank you for that! I will try it later on