mobizt/Firebase-ESP8266 #310: BUG

jgftech, 2022-03-06:
Hello!!

My program is getting into exception (WDT) , exception decoder give me this:

Decoding stack results
0x40247814: verify_SKE_sig at src/ssl/ssl_hs_client.c line 246
0x40247864: br_ssl_hs_client_init_main at /home/earle/src/esp-quick-toolchain/arduino/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/pgmspace.h line 108
0x40246e52: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1312
0x402367a9: __ssvfscanf_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfscanf.c line 332

This is the code:
 
After some executions ( about six times ) , the ESP gets into exception..

void update_firebase_last(){
  
    if (Firebase.ready() && signupOK){
      Firebase.RTDB.setInt(&fbdo, path + "last/" + "V1N", V1N);
      Firebase.RTDB.setInt(&fbdo, path + "last/" + "V2N", V2N);
      Firebase.RTDB.setInt(&fbdo, path + "last/" + "V3N", V3N);
      Firebase.RTDB.setInt(&fbdo, path + "last/" + "V12", V12);
      Firebase.RTDB.setInt(&fbdo, path + "last/" + "V23", V23);
      Firebase.RTDB.setInt(&fbdo, path + "last/" + "EMW", EMW);
      Firebase.RTDB.setInt(&fbdo, path + "last/" + "EKW", EKW);
      Firebase.RTDB.setInt(&fbdo, path + "last/" + "EW", EW);

        if (Firebase.RTDB.setTimestamp(&fbdo, path + "last/" + "date"))  
  lastdate= fbdo.payload();
  lastdateInt= fbdo.intData();
   }

mobizt, 2022-03-06:
It's not bug unless wdt reset.

You should use delay(0); at some place in your code to allow wdt feed.

closed 2022-03-06 by mobizt

jgftech, 2022-03-06:
Thanks for the awesome work on this library.

I tryed the delay(0) but without success. I was using the library about a year ago, and worked fine, but know I'm getting this issue... It's very weird. The exception is followed sometimes by Soft WDT Reset, and sometimes by HW WDT Reset

mobizt, 2022-03-06:
It depends on your code design.

You should understand what is soft wdt reset in esp MCU and how to solve it.

The library was intensive tested and active for bugs fix and improvements.

You should check the free heap, even you use the same code.

When you update the ESP8266 Arduino core sdk e.g. via boards manager, to v3.0.x, you need to set the MMU option that is suitable for large memory used by SSL engine which will set to low er heap as default.

mobizt, 2022-03-06:
With some blocking code as calling Firebase to read and store value, you need to add delay(0) to feed the wdt when you calling many functions continuously as this.

Do not run the code e.g. calling the Firebase function to read and store data that required large memory in the callback functions or ISR as it can cause the stack overflow and then wdt reset crashed.

mobizt, 2022-03-06:
The stack memory reserved in esp8266 is only 4k which can edit via the source code of core sdk to increase.

mobizt, 2022-03-07:
The **Firebase.ready()** should call **repeatedly in loop**.

Don't use **Firebase.ready()** as for status checking.

The function **Firebase.ready()** hides the complexities operation inside which is not check for some flag but process the authentication and token generation, refreshment and checking the token expiration.

This is the test code that runs 1000 times without problem.

```cpp

FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;

int count = 0;

void test()
{

  Firebase.setInt(fbdo, "test/1", count + 1);
  Firebase.setInt(fbdo, "test/2", count + 2);
  Firebase.setInt(fbdo, "test/3", count + 3);
  Firebase.setInt(fbdo, "test/4", count + 4);
  Firebase.setInt(fbdo, "test/5", count + 5);
  Firebase.setInt(fbdo, "test/6", count + 6);
  Firebase.setInt(fbdo, "test/7", count + 7);
  Firebase.setInt(fbdo, "test/8", count + 8);

  String lastdate;
  int lastdateInt = 0;

  if (Firebase.setTimestamp(fbdo, "test/9"))
    lastdate = fbdo.payload();

  lastdateInt = fbdo.intData();

  Serial.println(count);
  Serial.println(lastdate);
  Serial.println(lastdateInt);

  count++;
}

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  config.api_key = API_KEY;
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;
  config.database_url = DATABASE_URL;
  config.token_status_callback = tokenStatusCallback; 

  Firebase.begin(&config, &auth);

  Firebase.reconnectWiFi(true);

}

void loop()
{

  if (Firebase.ready() && (millis() - sendDataPrevMillis > 5000 || sendDataPrevMillis == 0))
  {
    sendDataPrevMillis = millis();
    test();
  }
}
```