mobizt/Firebase-ESP8266 #313: Cloud Messaging FCM unauthorized Error

garymansted, 2022-03-25:
Not to sure if this is a bug or not but I’ve been using the FirebaseESP8266 library to push notifications which has been working great.I have recently updated to the latest version - v3.9.5 and now I am getting an ‘’unauthorized’ error when sending a push notification. I haven’t changed anything. Ive checked the server key and is still current, the registration token - for the device I’m sending the notification to - which is also current. I am unsure what is going on. I may be missing something. Any help would be greatly appreciated.

` fbdo.fcm.begin(FIREBASE_FCM_SERVER_KEY);
    fbdo.fcm.addDeviceToken(DEVICE_REGISTRATION_ID_TOKEN);
    fbdo.fcm.setPriority("high");
    fbdo.fcm.setTimeToLive(100000);
    fbdo.fcm.setNotifyMessage(“Notification, "Hello World");
    Serial.printf("Send message... %s\n", Firebase.sendMessage(fbdo, 0) ? "ok" : fbdo.errorReason().c_str());
`

mobizt, 2022-03-25:
I test it and everything is ok.

mobizt, 2022-03-25:
Your FCM server key may not valid.

mobizt, 2022-03-25:
The test result from the example

```yml
Connected with IP: 192.168.137.219

Firebase Client v3.9.5

Send message... ok
{"multicast_id":857059186714013249,"success":1,"failure":0,"canonical_ids":0,"results":[{"message_id":"87bd9450-756c-467b-8760-23ad127fdeca"}]}   

Send message... ok
{"multicast_id":6849660580849654691,"success":1,"failure":0,"canonical_ids":0,"results":[{"message_id":"0aff9f09-29a0-4acc-a2a8-93d5dbf5384a"}]}  

Send message... ok
{"multicast_id":2832913015446131274,"success":1,"failure":0,"canonical_ids":0,"results":[{"message_id":"8c756a12-0b0f-4924-8993-d88f956440b6"}]}  

Send message... ok
{"multicast_id":4032599787453686828,"success":1,"failure":0,"canonical_ids":0,"results":[{"message_id":"d55620f1-dd91-47e3-9da1-15e466ecacf6"}]} 
```

garymansted, 2022-03-25:
I just created a new server key, removed the old one, did a test from firebase itself to my app (just to make sure its working) and it's still giving the unauthorised error. Im at a loss of what is going on because it was working fine

mobizt, 2022-03-25:
I don't know, the above test result is from the example and I just test it when I see this issue.

You should try to create the new project and test it.

garymansted, 2022-03-25:
Hi, I just created a new project and tested it with the example and it works fine. Don't know why its not working in y main project. Thanks for your help!

closed 2022-03-25 by mobizt