mobizt/Firebase-ESP8266 #316: BUG

Kesavan-Hex, 2022-03-26:
In file included from C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.cpp:42:0:
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:595:22: error: 'Client' has not been declared
     bool mReadClient(Client *client);
                      ^
In file included from C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.cpp:42:0:
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1151:18: error: 'Client' has not been declared
     int readLine(Client *stream, char *buf, int bufLen)
                  ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1175:18: error: 'Client' has not been declared
     int readLine(Client *stream, MB_String &buf)
                  ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1199:25: error: 'Client' has not been declared
     int readChunkedData(Client *stream, char *out, int &chunkState, int &chunkedSize, int &dataLen, int bufLen)
                         ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1278:25: error: 'Client' has not been declared
     int readChunkedData(Client *stream, MB_String &out, int &chunkState, int &chunkedSize, int &dataLen)
                         ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1352:20: error: 'Client' has not been declared
     int readClient(Client *client, MB_String &buf)
                    ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h: In member function 'int FirebaseJsonBase::readLine(int*, char*, int)':
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1158:24: error: request for member 'available' in '* stream', which is of non-class type 'int'
         while (stream->available() && idx <= bufLen)
                        ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1162:27: error: request for member 'read' in '* stream', which is of non-class type 'int'
             res = stream->read();
                           ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h: In member function 'int FirebaseJsonBase::readLine(int*, MB_String&)':
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1182:24: error: request for member 'available' in '* stream', which is of non-class type 'int'
         while (stream->available())
                        ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1186:27: error: request for member 'read' in '* stream', which is of non-class type 'int'
             res = stream->read();
                           ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h: In member function 'int FirebaseJsonBase::readClient(int*, MB_String&)':
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1366:36: error: request for member 'available' in '* client', which is of non-class type 'int'
         int chunkBufSize = client->available();
                                    ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1376:24: error: request for member 'connected' in '* client', which is of non-class type 'int'
         while (client->connected() && chunkBufSize == 0 && millis() - dataTime < 5000)
                        ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1378:36: error: request for member 'available' in '* client', which is of non-class type 'int'
             chunkBufSize = client->available();
                                    ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1382:21: error: request for member 'connected' in '* client', which is of non-class type 'int'
         if (client->connected())
                     ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1393:44: error: request for member 'available' in '* client', which is of non-class type 'int'
                     chunkBufSize = client->available();
                                            ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1495:56: error: request for member 'available' in '* client', which is of non-class type 'int'
                                         while (client->available() > 0)
                                                        ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1496:53: error: request for member 'read' in '* client', which is of non-class type 'int'
                                             client->read();
                                                     ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1503:52: error: request for member 'available' in '* client', which is of non-class type 'int'
                                     while (client->available() > 0)
                                                    ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1504:49: error: request for member 'read' in '* client', which is of non-class type 'int'
                                         client->read();
                                                 ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1523:32: error: request for member 'available' in '* client', which is of non-class type 'int'
                 while (client->available() > 0)
                                ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1524:29: error: request for member 'read' in '* client', which is of non-class type 'int'
                     client->read();
                             ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h: At global scope:
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1832:19: error: 'Client' has not been declared
     bool readFrom(Client &client) { return mReadClient(&client); }
                   ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:1834:19: error: 'Client' has not been declared
     bool readFrom(Client *client) { return mReadClient(client); }
                   ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.h:2285:19: error: 'Client' has not been declared
     bool readFrom(Client &client) { return mReadClient(&client); }
                   ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.cpp:532:36: error: 'bool FirebaseJsonBase::mReadClient' is not a static member of 'class FirebaseJsonBase'
 bool FirebaseJsonBase::mReadClient(Client *client)
                                    ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.cpp:532:36: error: 'Client' was not declared in this scope
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.cpp:532:44: error: 'client' was not declared in this scope
 bool FirebaseJsonBase::mReadClient(Client *client)
                                            ^
C:\Users\kesav\Documents\Arduino\libraries\Firebase-ESP8266\src\json\FirebaseJson.cpp:533:1: error: expected ',' or ';' before '{' token
 {
 ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

mobizt, 2022-03-26:
You should install ESP8266 Arduino Core SDK from Arduino IDE's Boards Manager or update to latest version.

You should select one of ESP8266 board in Arduino IDE.

closed 2022-03-26 by mobizt

Kesavan-Hex, 2022-03-26:
thank you updating esp8266 version in board manager fixed the problem